h2zero/NimBLE-Arduino #203: Question: scanning and connecting

btsimonh, 2021-03-02:
Hi all,

Question about scanning types and connecting.

Currently by default, I start a 'new' scan (i.e. pass 0 into continuation) every time.
I start a new scan after connect() and every time a scan ends (20s scans by default).

I then store MACs of things I hear adverts from, and discard those I don't want.

**So, the question:**
If I try to connect to a device, is there any advantage to it being in the current scan results?  i.e. is me restarting the scan from scratch every time making it harder for the subsystem to connect to devices (longer connection times, etc...)?

If I did do a continuation every time, I imaging the memory building over time - especially with lots of random addresses from phones, etc.  
Should I modify the code to do continuation scans, but to explicitly delete the macs that I am ignoring?  Or maybe periodically go through all the scan results and remove ones I'm not interested in?
So basically, is there an advantage to using continuation, and if so how to control memory usage?

best,

Simon

h2zero, 2021-03-02:
>If I try to connect to a device, is there any advantage to it being in the current scan results? i.e. is me restarting the scan from scratch every time making it harder for the subsystem to connect to devices (longer connection times, etc...)?

No advantage to keeping it, the erase method is there for this. There is no effect on connections when scanning.

>If I did do a continuation every time, I imaging the memory building over time - especially with lots of random addresses from phones, etc.

You can limit this with `pScan->setMaxResults(xx)`. Also you can add macs to the ignore list so you won't see them.

btsimonh, 2021-03-02:
thankyou;  I just wondered if it kept anything useful like rf frequencies for connection, etc.  But if not, then I'm not doing anything wrong :).

closed 2021-03-02 by btsimonh

h2zero, 2021-03-03:
You're welcome, hope the project is going well over there.

btsimonh, 2021-03-03:
we still have restarts - uptimes between 1 hour and 2 days.  Has anyone produced a simple reproduction of the stall issue?
I also still see 'lld_pdu_get_tx_flush_nb HCI packet count mismatch (1, 2)' in serial logs regularly.  I feel this is related to the device I'm connecting to - like BLE receiving two of something during connection rather than one, or it sending a notify packet at connect.