h2zero/NimBLE-Arduino #204: onConnect() not getting called?

doudar, 2021-03-09:
OnDisconnect seems to be getting called, any idea why onConnect may not be working? Below is the relevant code from https://github.com/doudar/SmartSpin2k/tree/develop-release-candidate/

-------------------------
pClient->setClientCallbacks(new SpinBLEClient::MyClientCallback(), true);
-------------------------
private:
    class MyAdvertisedDeviceCallback : public NimBLEAdvertisedDeviceCallbacks
    {
       public:
       void onResult(NimBLEAdvertisedDevice *);
    };

    class MyClientCallback : public NimBLEClientCallbacks
    {
        public:
        void        onConnect(BLEClient *,  ble_gap_conn_desc* desc);
        void        onDisconnect(BLEClient *);
        uint32_t    onPassKeyRequest();
        bool        onConfirmPIN(uint32_t);
        void        onAuthenticationComplete(ble_gap_conn_desc);
    };
};
-------------------------
void SpinBLEClient::MyClientCallback::onConnect(BLEClient *pclient, ble_gap_conn_desc* desc)
{

    auto addr = pclient->getPeerAddress();
    for (size_t i = 0; i < NUM_BLE_DEVICES; i++)
    {
        if (addr == spinBLEClient.myBLEDevices[i].peerAddress)
        {
            spinBLEClient.myBLEDevices[i].doConnect = true;
            if ((spinBLEClient.myBLEDevices[i].charUUID == CYCLINGPOWERMEASUREMENT_UUID) || (spinBLEClient.myBLEDevices[i].charUUID == FITNESSMACHINEINDOORBIKEDATA_UUID) || (spinBLEClient.myBLEDevices[i].charUUID == FLYWHEEL_UART_RX_UUID))
            {
                spinBLEClient.connectedPM = true;
                debugDirector("Registered PM on Connect");
                //spinBLEClient.removeDuplicates(pclient);
                return;
            }
            if ((spinBLEClient.myBLEDevices[i].charUUID == HEARTCHARACTERISTIC_UUID))
            {
                spinBLEClient.connectedHR = true;
                debugDirector("Registered HRM on Connect");
                return;
            }
        }
        debugDirector("On Connect: ", String(i));
    }
}

void SpinBLEClient::MyClientCallback::onDisconnect(BLEClient *pclient)
{

    if (spinBLEClient.intentionalDisconnect)
    {
        debugDirector("Intentional Disconnect");
        spinBLEClient.intentionalDisconnect = false;
        return;
    }
    if (!pclient->isConnected())
    {
        auto addr = pclient->getPeerAddress();
        //auto addr = BLEDevice::getDisconnectedClient()->getPeerAddress();
        debugDirector("This disconnected client Address " + String(addr.toString().c_str()));
        for (size_t i = 0; i < NUM_BLE_DEVICES; i++)
        {

            if (addr == spinBLEClient.myBLEDevices[i].peerAddress)
            {
                //spinBLEClient.myBLEDevices[i].connectedClientID = BLE_HS_CONN_HANDLE_NONE;
                debugDirector("Detected " + String(spinBLEClient.myBLEDevices[i].serviceUUID.toString().c_str()) + " Disconnect");
                spinBLEClient.myBLEDevices[i].doConnect = true;
                if ((spinBLEClient.myBLEDevices[i].charUUID == CYCLINGPOWERMEASUREMENT_UUID) || (spinBLEClient.myBLEDevices[i].charUUID == FITNESSMACHINEINDOORBIKEDATA_UUID) || (spinBLEClient.myBLEDevices[i].charUUID == FLYWHEEL_UART_RX_UUID))
                {
                    spinBLEClient.connectedPM = false;
                    break;
                }
                if ((spinBLEClient.myBLEDevices[i].charUUID == HEARTCHARACTERISTIC_UUID))
                {
                    spinBLEClient.connectedHR = false;
                    break;
                }
            }
        }
        return;
    }
}

kadaan, 2021-03-09:
Hi.  Let me see what I can find.  I'll have to dig up how to turn on your logs and get back to you.

kadaan, 2021-03-09:
I assume I should see this logged when -DCORE_DEBUG_LEVEL=5 is set, right?
`NIMBLE_LOGD("NimBLEClientCallbacks", "onConnect: default");`

kadaan, 2021-03-09:
There we go, with the default callback: 
```
D NimBLEClient: "Got Client event "
I NimBLEClient: "Connected event"
D NimBLEClient: "Got Client event "
I NimBLEClient: "mtu update event; conn_handle=0 mtu=153"
I NimBLEClient: "Connection established"
D NimBLEClient: ">> deleteServices"
D NimBLEClient: "<< deleteServices"
D NimBLEClientCallbacks: "onConnect: default"
D NimBLEClient: "<< connect()"
 - Connected to server
```

kadaan, 2021-03-09:
I can see the custom callback happening too.  Thanks!

kadaan, 2021-03-09:
User error, probably :P

TY

doudar, 2021-03-11:
Yah, not certain.....

It seems something unrelated may have fixed it. Closing.

closed 2021-03-11 by doudar

doudar, 2021-03-20:
@h2zero, Ultimately I think our issue was that we errantly expected the callback to happen later in our connection sequence (after notification subscription) not immediately when the client is created. Definitely not an issue with Nimble.

h2zero, 2021-03-20:
Ahh the wonders of wireless communication (still amazes me that this stuff works so well). 
Thanks for the update/info, glad to hear all is well 😄