h2zero/NimBLE-Arduino #205: How to combine continuous BLE scan and the IotWebConf lib?

thomassandberg, 2021-03-13:
I have set up NimBLE to act as a client up against a density meter from Anton Paar.
The BLE part works great but the continuous  BLE scan is giving me some headache.

I want to use the IotWebConf lib to configure wifi and mqtt server parameters etc.
The problem is that this lib requires a loop to be functioning and that do not pair up well with the scan.

The ESP32 connects to the Wifi but I am not able to access the internal webserver in IotWebConf while the scan is running. If I turn on the density meter and the BLE client connect (scan is stopped) I can access the webserver.
But as soon as the scan start again (I turn of the density meter) the webserver stops responding.

Everything else with the BLE client works as it should - just not scan part.

Is there a way to make keep the continuous scan and still be able to access the webserver ? 

I "delay" the BLE part that should have been in setup() by not running it until the Wifi is connected.
I use a executeonce inside loop() for this.

`void loop() {
  
  iotWebConf.doLoop();

  if ((executeonce == false) && (WiFi.status() == WL_CONNECTED)) {
    executeonce = true;
    setClock();  // set the clock using NTP
    startBLE();  // starts the BLE part that should have been in setup()
  }

  if(millis() - last_time > 1000) {
    
///// The rest of the BLE client code goes here and this part works well
  
  last_time = millis();
  
  }
}
`

`void startBLE()
{
  Serial.println("Starting Arduino BLE Client application...");
  BLEDevice::init("");
  BLEScan* pBLEScan = BLEDevice::getScan();
  pBLEScan->setAdvertisedDeviceCallbacks(new MyAdvertisedDeviceCallbacks());
  pBLEScan->setInterval(1349);
  pBLEScan->setWindow(449);
  pBLEScan->setActiveScan(true);
  pBLEScan->start(0, false);
}`

h2zero, 2021-03-13:
Hi @thomassandberg, what I have done before to make this work is put the iotwebconf.loop in a separate task with a delay.
```
void webConfTask(void *params){
  uint32_t notifiedValue;
    for(;;){
      xTaskNotifyWait(0x00,ULONG_MAX,&notifiedValue, (1 / portTICK_PERIOD_MS));
      iotWebConf.doLoop();
    }
}
```

Created in setup like this `xTaskCreatePinnedToCore(webConfTask, "webConfTask", 4500, NULL, 1, &hWebConfTask, 1);`

thomassandberg, 2021-03-13:
Dude! That worked like a charm! Thank you very much :)

h2zero, 2021-03-14:
Great, probably not the most applicable code for your use but I'm glad it works. 
`xTaskNotifyWait(0x00,ULONG_MAX,&notifiedValue, (1 / portTICK_PERIOD_MS));`
could simply be replaced with `delay(1);` for you usage unless you need to send task notifications to it later.

h2zero, 2022-04-05:
Closing as resolved.

closed 2022-04-05 by h2zero