h2zero/NimBLE-Arduino #206: Connection failure type

btsimonh, 2021-03-14:
Hi H2zero,
some time ago, I had a modification in NimBLEClient which detected the rc from ble_gap_connect() and allowed me to read it in the event of connection failure.
After moving to 1.20, I removed this.

However, have just got a report of a device getting into a state where connect() is failing immediately, and never again succeeds.

this was my code outside the client, run after a connect failure:
```
#ifdef NIMBLE_CLIENT_HAS_GETRESULT
    int rc = pClient->getResult();

    switch (rc){
      case (0x0200+BLE_ERR_CONN_LIMIT ):
#ifdef BLE_ESP32_DEBUG
        AddLog(LOG_LEVEL_ERROR,PSTR("BLE: Hit connection limit? - restarting NimBLE"));
#endif
        BLERestartNimBLE = 1;
        BLERestartBLEReason = BLE_RESTART_BLE_REASON_CONN_LIMIT;
        break;
      case (0x0200+BLE_ERR_ACL_CONN_EXISTS):
#ifdef BLE_ESP32_DEBUG
        AddLog(LOG_LEVEL_ERROR,PSTR("BLE: Connection exists? - restarting NimBLE"));
#endif
        BLERestartNimBLE = 1;
        BLERestartBLEReason = BLE_RESTART_BLE_REASON_CONN_EXISTS;
        break;
    }
#endif
```

This type of failure was not uncommon in 1.00; do you think I should re-instate the code, at least temporarily, to see if we have the same issue in 1.20?

It seems that it may be good to have something in the client which would enable a user to find out why the connect failed - the two above in 1.00 were effectively un-recoverable without restarting the BLE stack....

your advice appreciated..

br,
simon

p.s. these are those pesky EQ3 thermostats which have given both me and @fightforlife such grief.

btsimonh, 2021-03-14:
my new mod:
https://github.com/btsimonh/Tasmota/commit/31afd59175b96bd6d49869eb4df0a7e52fa70021

will let you know if we see it.

h2zero, 2021-03-14:
The changes made to the client in 1.2.0 should handle all of the possible error codes and print a debug message with the error message and code.

btsimonh, 2021-03-14:
that's what I thought...  in this case we only have syslog messages, not serial.
What could result in never being able to make another connection and immediate 'cannot connect'?

h2zero, 2021-03-14:
There are a few possibilities depending on the peer device:
- It could be in a connection with something else already
- It could be in a non-connectable state (check flags)
- It could be advertising with a white list and only allowing connections to certain peers
- You could already be connected but that should not be possible any longer.
- You or the device could be out of memory.
- Some bug that we don't know of yet.
- It doesn't like you anymore and doesn't want to talk to you 😄 

Without log info it's pretty hard to say, I would be inclined to use a sniffer to trace this.

Ysbrand, 2021-03-14:
Some observations:

-  It's not device specific, it simply refuses to send commands to any device
- It's still able to process messages from other devices which are broadcasting messages (pvvx sensors)
- A reboot of the ESP32 clears the situation immediately

So I'm inclined to say that it's not something on the target device, but more on 'our' side

btsimonh, 2021-03-14:
@h2zero - so the of two specific rc values I saw previously - we should see neither, as we've always asked it to disconnect previously...  That was my impression of your work on 1.20, hence why I removed the test :).
I'll add a log to display the rc for all failure to connect(), as it looks like my code will only log those two.

h2zero, 2021-03-14:
Ok, so if the problem is on "our side", which seems to be the case, then there are a few possibilities. 

The reasons for this could be:
 - the scanner is active and not being stopped
 - the previous connection is still active somewhere in the stack
 -  A previous attempt to connect is still ongoing (not yet completed or timed out)
 - not enough ram

btsimonh, 2021-03-14:
The scanner stops automagically?
Previous connections SHOULD have been closed, or failed.
(but hence the two things I previously looked for - already connected and too many connections).
'timed out' - the failure occurred ~2:30 am, and every connection attempt after that failed immediately - up until reboot at 9:30am.
ram is probably ok, between 120-140k free heap reported.

h2zero, 2021-03-14:
>The scanner stops automagically?

It does if the call succeeds, however it can fail to stop and will result in an immediate return from `connect()` - impossible to know without a log.

> Previous connections SHOULD have been closed, or failed.

They should if all went well, however there could be a bug dangling a connection somewhere.

>'timed out' - the failure occurred ~2:30 am, and every connection attempt after that failed immediately - up until reboot at 9:30am.
ram is probably ok, between 120-140k free heap reported.

Probably not the issue, and narrows down the list.

btsimonh, 2021-03-14:
@ysbrand now has a version which logs the rc on every connect failure; so assuming that may tell us something, we will know when it happens again.  However, we've been running this for some months, and this is the first time we've noticed it doing this specific thing - so we may be in for a long wait! :(

h2zero, 2021-03-14:
I guess we'll have to be patient, maybe some early logs will give a clue though.

h2zero, 2022-04-05:
Closing as stale, please reopen if the issue still persists.

closed 2022-04-05 by h2zero