h2zero/NimBLE-Arduino #207: Sending text to Client?

PDE-Paul, 2021-03-15:
Hi h2zero,

Firstly I want to thank you for a fantastic library, it has solved my heap issue. Now I need to know how do I send text to my client at run time. 

I am experimenting with your Server example and it only ever sends "Burger" once, I need to be able to send text in the loop() routine. This may be a stupid question :- How is this done?

Regards
Paul

h2zero, 2021-03-15:
Hi @PDE-Paul, you're welcome.
You would do that by sending a notification to the client after updating the characteristic value. 

The client would need to subscribe to notifications for that as well.

PDE-Paul, 2021-03-15:
I have tried this..

                      pCharacteristic->setValue((BLE_Tx_MSG).c_str());
                      pCharacteristic->notify();

With no luck

h2zero, 2021-03-15:
Was there a client subscribed? 

Could you turn on logging and post any log messages you get?

PDE-Paul, 2021-03-15:
I just fixed this, thank you for the assistance.

closed 2021-03-15 by PDE-Paul