h2zero/NimBLE-Arduino #208: STA_DISCONNECTED reason 8 always and reboot

noukee, 2021-03-15:
Using NimBLE-Arduino and PubSubClient with WiFiClientSecure with TLS (and We are using PSRAM)

We can connect to the MTT broker but after a few seconds we always receive a SYSTEM_EVENT_STA_DISCONNECTED reason 8 that always reboot the system. 

If we do not use bluetooth the problem does not occur or when we use a very small bluetooth service. 

Can someone help us? 

platformio.ini

```
platform = espressif32
framework = arduino
board =  esp-wrover-kit
```
Customized SDKconfig to use PSRAM:

```
# PSRAM
CONFIG_SPIRAM_USE_CAPS_ALLOC=y
CONFIG_SPIRAM_CACHE_WORKAROUND=y

# SW COEXIST WIFI AND BLE
CONFIG_SW_COEXIST_ENABLE=y

# SINGLE CORE
CONFIG_FREERTOS_UNICORE=y

# TLS USES PSRAM
CONFIG_MBEDTLS_EXTERNAL_MEM_ALLOC=y
CONFIG_MBEDTLS_TLS_CLIENT_ONLY=y
```

[sdkconfig.TXT](https://github.com/h2zero/NimBLE-Arduino/files/6143750/sdkconfig.TXT)

Log:

```
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:391] _eventCallback(): Reason: 8 - ASSOC_LEAVE
[E][ssl_client.cpp:36] _handle_error(): [data_to_read():287]: (-76) UNKNOWN ERROR CODE (004C)
[D][mymqtt.cpp:102] task(): === MQTT NOT CONNECTED ===
[E] Received event:
[D][mywifi.cpp:85] reconnect(): [!] WIFI Disconnected
[E] Received event:
[!] MQTT Disconnected
[!] WIFI Disconnected
[D][mymqtt.cpp:108] task(): === MQTT RECONNECTING ===
[I][mymqtt.cpp:125] reconnect(): Attempting MQTT connection...
[I][mymqtt.cpp:126] reconnect(): Connecting to cloud.noukee.com
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:391] _eventCallback(): Reason: 8 - ASSOC_LEAVE
[D][mywifi.cpp:85] reconnect(): [!] WIFI Disconnected
[E][ssl_client.cpp:98] start_ssl_client(): Connect to Server failed!
[D][mymqtt.cpp:223] sendData(): Sending message to MQTT topic gateway/setup/40f52055dd80
{
  "v
[E][WiFiClientSecure.cpp:127] connect(): start_ssl_client: -1
[!] WIFI Disconnected
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x40094014  PS      : 0x00060230  A0      : 0x80122a30  A1      : 0x3ffe2670  
A2      : 0xfffffffe  A3      : 0xfffffffc  A4      : 0x000000ff  A5      : 0x0000ff00  
A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x00000000  A9      : 0x00000000  
A10     : 0x00000000  A11     : 0x3f402c5c  A12     : 0x00000000  A13     : 0x00000000  
A14     : 0x00000000  A15     : 0xffffffc0  SAR     : 0x00000004  EXCCAUSE: 0x0000001c  
EXCVADDR: 0xfffffffc  LBEG    : 0x40094029  LEND    : 0x40094039  LCOUNT  : 0xfffffffd  
ELF file SHA256: 0000000000000000
Backtrace: 0x40094014:0x3ffe2670 0x40122a2d:0x3ffe2680 0x4012b84e:0x3ffe2990 0x4012b88a:0x3ffe2a20 0x400ddab7:0x3ffe2a60 0x400db111:0x3ffe2ac0 0x40095cae:0x3ffe2af0
Rebooting...

```

h2zero, 2021-03-15:
Simultaneous use of BLE and TLS data transfer is a known issue. The TLS stuff seems to take too much time and BLE events interrupt it which stuffs up the stacks.

The only fix is to do BLE operations, stop, then do MQTT then BLE scan advertise etc. and repeat.

noukee, 2021-03-16:
@h2zero We are using NimBLE_Server with Advertisement (without BLE scan), should we stop advertisement to do MQTT? Because that happens without any BLE clients 

Thanks for your support

h2zero, 2021-03-16:
I haven't seen this happen while advertising before but it's possibly the same issue. Advertising generates interrupts to switch channels on the interval time and I believe that this problem occurs due to these interrupts.

It would be worth it to try stopping the advertising while running the mqtt task. I'd be very curious to know if that resolves the issue.

h2zero, 2022-04-05:
Closing as stale. Please reopen if the issue still persists.

closed 2022-04-05 by h2zero