h2zero/NimBLE-Arduino #209: Memory allocation while not using NimBLE.

rvbc1, 2021-03-19:
Hello, 
I am making device which can use WiFi or BT. At boot I want to run BT or WiFi (not both) depends config file stored in SPIFFS, but when if bluetoothSettings.active is false or true ESP32 shows 42kb of free heap.
`
if(bluetoothSettings.active == true){
  BLEDevice::init(bluetoothSettings.name);
}
`
When I delete line BLEDevice::init(bluetoothSettings.name); ESP shows 82kb of free heap, that means BT driver is started?
Is possible to not reserve RAM for unused BT? How to do that, because now unused BT takes ~40kb of RAM.
I only need at boot read config file and decide which protocol (BT or WiFi) run.
Config file can be changed if user push button but after that i run ESP.restart();.

h2zero, 2021-03-20:
This looks to be due to the compiler/linker, when that line is commented out the arduino core will release the memory when it is initialize as it detects bt is not in use. 

The fix for this would be changing your code to something like this:
```
if(bluetoothSettings.active == true) { 
    BLEDevice::init(bluetoothSettings.name);
} else {
    esp_bt_controller_mem_release(ESP_BT_MODE_BTDM);
}
```

rvbc1, 2021-03-22:
It's working, thank you for your help, you are awesome!

closed 2021-03-22 by rvbc1