h2zero/NimBLE-Arduino #210: Connect to a device with a static pin

jhoydich, 2021-03-20:
First off I wanted to thank you for the work on this library. I'm am trying to connect to a ble device that has a static 6 digit pin and I've been having some trouble connecting to it. I tried the secure client example, replacing all the necessary uuids / pincode to the appropriate values for my device. onPasskeyRequest() gets called but the connection fails just after. Any advice?

h2zero, 2021-03-20:
Hard to say what is going on based on your description, would you be able to provide any logs?

The good news is `onPasskeyRequest()` is being called so at least we know it's getting to that point. After that, unless you are not providing the correct passkey I'm unsure of why it would fail (other than a poor connection).

jhoydich, 2021-03-20:
Unfortunately no logs currently, would I get logs by setting the `CORE_DEBUG_LEVEL` higher?

I have quadruple checked the passcode, and I don't thing the connection is an issue, the devices are right next to each other.

h2zero, 2021-03-20:
Yes, if you set `CORE_DEBUG_LEVEL 5` you will get the library logs. Also if you define `CONFIG_BT_NIMBLE_DEBUG` you will get additional logs from the NimBLE core.

jhoydich, 2021-03-20:
Wonderful! Thank you! I will get some logs early next week and link here if I can't figure it out.

jhoydich, 2021-03-22:
Turns out the specific service that I wanted was not able to be found without connecting to the device. So instead of `if (device.isAdvertisingService(serviceUuid))` I found the device based off of the device name `if (strcmp(device.getName().c_str(), "My device name") == 0)`.  Everything (almost) worked nicely then. 

The almost refers to the value returned from the characteristic. Unfortunately it returns as garbage runes instead of the expected output. Any advice? I find it odd that it prints out the device name perfectly fine but messes this up. I have attached a log showing my output and a picture of the expected output. Again, thank you for your help on this!
[image]
[nimble_logs.txt](https://github.com/h2zero/NimBLE-Arduino/files/6185465/nimble_logs.txt)

h2zero, 2021-03-22:
Glad you've got it working this far. The string isn't printing due to the value shown in the picture not having ascii characters. 

To solve that you will need to print the bytes in a individually in a loop.

jhoydich, 2021-03-22:
Yep you're right! Once more, thanks!

closed 2021-03-22 by jhoydich