h2zero/NimBLE-Arduino #211: pClient->isConnected() sometimes returns the wrong value?

doudar, 2021-03-20:
@h2zero , Any ideas why this block is needed in our code? It' routinely getting triggered on reconnects but also occasionally on new connections. 
   
https://github.com/doudar/SmartSpin2k/blob/2389e8dd89541eec837142d48150b6e238a3a31b/src/BLE_Common.cpp#L101

What seems to be happening is that the client gets "stuck" immediately after a connection, the subscribed UUID value never updates and when I've tried to do a pClient->disconnect() I get a NimBLE error saying it's connected. So I've had to resort to deleting the client in this case. It's a new gremlin since we've worked our way out of using Callbacks to process our UUID data. 

It happens when pClient->isConnected() in this line returns false:
https://github.com/doudar/SmartSpin2k/blob/2389e8dd89541eec837142d48150b6e238a3a31b/src/BLE_Common.cpp#L40

h2zero, 2021-03-20:
I think this stems from the `BLEDevice::getClientByPeerAddress` check. If you were connected to that address before and have not deleted the client instance it will be retained in that list for more efficient reconnecting.

doudar, 2021-03-20:
To help me further understand, you're thinking we occasionally get two instances of the same (address) client in BLEdevices ?

Seems plausible

h2zero, 2021-03-20:
>To help me further understand, you're thinking we occasionally get two instances of the same (address) client in BLEdevices ?
(One connected and one not)

This is entirely possible as the current code does not handle this situation. What goes on is the NimBLEDevice stores a list of clients that have been created and when you call `getClientByPeerAddress` it returns a pointer to the first client instance that has that peer address. If more than one client was created and connected to the same address you will see the issue you have described. This could possibly be handled better in the library and has been noted for improvement.
 
>The funny thing is that I think the reconnect code is about the only thing untouched in that PR you saw 🤣🤣🤣

That's amazing with 700+ files changed lol

h2zero, 2022-04-05:
Closing as resolved.

closed 2022-04-05 by h2zero