h2zero/NimBLE-Arduino #212: Client code fails to connect to Android phone

balk77, 2021-03-21:
I have loaded `NimBLE_Client` on an ESP32, slightly modified. I want to connect an ESP32 to my Android Pocophone F1 and show Komoot directions on the screen attached to the ESP. 

```
        static BLEUUID serviceUUID("71C1E128-D92F-4FA8-A2B2-0F171DB3436C");
        if (advertisedDevice->haveServiceUUID() && advertisedDevice->isAdvertisingService(serviceUUID)) 
        {
            Serial.println("Found Our Service");
            /** stop scan before connecting */
            NimBLEDevice::getScan()->stop();
            /** Save the device reference in a global for the client to use*/
            advDevice = advertisedDevice;
            /** Ready to connect now */
            doConnect = true;
        }
```
I have enabled debugging in `nimconfig.h` and I am getting this; disconnect code at the end. Do you know why this connection fails?

```
LE advertising report. len=32 num=1 evtype=0 addrtype=1 addr=4c.76.32.db.a7.65 advlen=21 rssi=-56
02 01 02 11 07 6c 43 b3 
1d 17 0f b2 a2 a8 4f 2f 
d9 28 e1 c1 71 
LE advertising report. len=25 num=1 evtype=4 addrtype=1 addr=4c.76.32.db.a7.65 advlen=14 rssi=-56
0d 09 50 4f 43 4f 50 48 
4f 4e 45 20 46 31 
Advertised Device found: Name: POCOPHONE F1, Address: 4c:76:32:db:a7:65, serviceUUID: 71c1e128-d92f-4fa8-a2b2-0f171db3436c
Found Our Service
ble_hs_hci_cmd_send: ogf=0x08 ocf=0x000c len=2
0x0c 0x20 0x02 0x00 0x00 
Command complete: cmd_pkts=5 ogf=0x8 ocf=0xc status=0 
Scan Ended
New client created
GAP procedure initiated: connect; peer_addr_type=1 peer_addr=4c:76:32:db:a7:65 scan_itvl=16 scan_window=16 itvl_min=12 itvl_max=12 latency=0 supervision_timeout=51 min_ce_len=0 max_ce_len=0 own_addr_type=0
ble_hs_hci_cmd_send: ogf=0x08 ocf=0x000d len=25
0x0d 0x20 0x19 0x10 0x00 0x10 0x00 0x00 0x01 0x65 0xa7 0xdb 0x32 0x76 0x4c 0x00 0x0c 0x00 0x0c 0x00 0x00 0x00 0x33 0x00 0x00 0x00 0x00 0x00 
Command Status: status=0 cmd_pkts=5 ocf=0xd ogf=0x8
LE connection complete. handle=1 role=0 paddrtype=1 addr=4c.76.32.db.a7.65 local_rpa=0.0.0.0.0.0 peer_rpa=0.0.0.0.0.0 itvl=12 latency=0 spvn_tmo=51 mca=0
0x65 0xa7 0xdb 0x32 0x76 0x4c 
GATT procedure initiated: exchange mtu
host tx hci data; handle=1 length=7
ble_hs_hci_acl_tx(): 0x01 0x00 0x07 0x00 0x03 0x00 0x04 0x00 0x02 0xff 0x00 
txed att command: mtu req; conn=1 mtu=255
ble_hs_hci_cmd_send: ogf=0x08 ocf=0x0016 len=2
0x16 0x20 0x02 0x01 0x00 
Command Status: status=2 cmd_pkts=5 ocf=0x16 ogf=0x8
Number of Completed Packets: num_handles=1
handle:1 pkts:1
Disconnection Complete: status=0 handle=1 reason=11
E NimBLEClient: "Connection failed; status=523 "
Failed to connect, deleted client
Failed to connect, starting scan

```

h2zero, 2021-03-21:
Very strange error in that situation. The error code is:

0x020b | 0x0b | BLE_ERR_ACL_CONN_EXISTS | ACL Connection Already Exists
-- | -- | -- | --

So, for some reason it looks like the esp controller says you were already connected and are attempting to connect again. Not sure why or how but if you have different tasks doing BLE stuff this is possible. There is also a bug in the esp controller in the latest arduino idf version which could also be a possible cause.

h2zero, 2022-04-05:
Closing this as old with no further data. Assuming resolved. Please reopen if the issue still persists.

closed 2022-04-05 by h2zero