h2zero/NimBLE-Arduino #214: Server: add Client connection to white list and save to nvc

kirus1997, 2021-04-01:
Hi.

My idea is to set up a white list for a Server, which will store the BLE addresses of the allowed Clients. The Server will then advertise with white list and only connect to the BLE addresses in the list (using setScanFilter(bool scanRequestWhiteListOnly, bool connectWhitelistOnly). 

I am planning to use bonding, which will call onAuthenticationComplete(ble_gap_conn_desc) callback. If Connection has been established successfully, I will add the BLE address of the Client to white list, and save it to NVC. Next time the Server is powered up, it will be using white list information and only connect to the Client on the list. If white list needs to be deleted, I will press a button, which will call a function that deletes the white list.

I am not sure where I can find information on white list addition/deletion and how to implement it in NimBLE.

Would appreciate any help and/or pointers to the right direction.

Thanks!

K

h2zero, 2021-04-01:
Hi @kirus1997, while the use case you describe is somewhat available already at low levels an API has not been implemented yet. This is due to issues regarding random addresses, the ESP32 controller does not resolve them, and so if a random address (such as a phone) was used as the whitelist address it would not be able to connect when the address changes. So to work around that issue there would need to be a good amount of workarounds written into the server code. 

If you are not concerned with whitelisting random addresses then the solution would revolve around a call to :
`ble_gap_wl_set(const ble_addr_t *addrs, uint8_t white_list_count)`

Which could be done like so:
```
NimBLEAddress address(xxxxxxxx);

ble_addr_t peerAddr;
memcpy(&peerAddr.val, address.getNative(), 6);
peerAddr.type = address.getType();

ble_gap_wl_set(peerAddr, 1);
```
Or any other derived method of getting the address (keep in mind the address needs to be little endian).

kirus1997, 2021-04-01:
OK, thank you.

Would you be able to let me know how I can check the contents of the white list?
I think if I know this, I will be able to figure the rest out. :)

Thanks,

K

h2zero, 2021-04-01:
There is no function to check the contents that I know of. Each call to `ble_gap_wl_set()` clears the current list and updates it with the provided addresses.

closed 2021-04-06 by kirus1997