h2zero/NimBLE-Arduino #216: 'class NimBLEServer' has no member named 'getConnId'

societyofrobots, 2021-04-04:
I'm converting from the older BLE to NimBLE, but got this one compile error I can't figure out.

This works in the old version:
`pServer->disconnect(pServer->getConnId());`

But in NimBLE gives me this error:
`'class NimBLEServer' has no member named 'getConnId'`

h2zero, 2021-04-04:
Sorry for the trouble. This was not implemented due to multi-connection support. Instead you should store the connection ID from the `onConnect` callback and use it directly with this function. I hope to update this in the future to support other methods.

h2zero, 2021-05-08:
Just pushed a commit yesterday that addresses this 206cb3d4af100265c92dd568dafbc23736f94e5e.

societyofrobots, 2021-05-13:
I ended up using this and it solved my issue:
`NimBLEDevice::getClientByID(desc->conn_handle)->disconnect();`

closed 2021-05-13 by societyofrobots