h2zero/NimBLE-Arduino #217: FreeRTOS compile errors

societyofrobots, 2021-04-04:
I'm finishing up converting my code from the old BLE to NimBLE.

I'm getting these compile errors about FreeRTOS. What am I doing wrong?

```
Arduino: 1.8.13 (Windows 10), Board: "Adafruit ESP32 Feather, 80MHz, 921600, None, Minimal SPIFFS (Large APPS with OTA)"

libraries\NimBLE-Arduino\FreeRTOS.cpp.o: In function `FreeRTOS::sleep(unsigned int)':

c:\users\X\appdata\local\arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-97-gc752ad5-5.2.0\xtensa-esp32-elf\include\c++\5.2.0\bits/basic_string.tcc:1163: multiple definition of `FreeRTOS::sleep(unsigned int)'

libraries\BLE\FreeRTOS.cpp.o:C:\Users\X\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\libraries\BLE\src/FreeRTOS.cpp:21: first defined here

libraries\NimBLE-Arduino\FreeRTOS.cpp.o: In function `FreeRTOS::startTask(void (*)(void*), std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, void*, unsigned int)':

FreeRTOS.cpp:(.text._ZN8FreeRTOS9startTaskEPFvPvENSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEES0_j+0x0): multiple definition of `FreeRTOS::startTask(void (*)(void*), std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, void*, unsigned int)'

libraries\BLE\FreeRTOS.cpp.o:FreeRTOS.cpp:(.text._ZN8FreeRTOS9startTaskEPFvPvENSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEES0_j+0x0): first defined here

libraries\NimBLE-Arduino\FreeRTOS.cpp.o: In function `FreeRTOS::deleteTask(void*)':

FreeRTOS.cpp:(.text._ZN8FreeRTOS10deleteTaskEPv+0x0): multiple definition of `FreeRTOS::deleteTask(void*)'

libraries\BLE\FreeRTOS.cpp.o:FreeRTOS.cpp:(.text._ZN8FreeRTOS10deleteTaskEPv+0x0): first defined here

libraries\NimBLE-Arduino\FreeRTOS.cpp.o: In function `FreeRTOS::getTimeSinceStart()':

FreeRTOS.cpp:(.text._ZN8FreeRTOS17getTimeSinceStartEv+0x0): multiple definition of `FreeRTOS::getTimeSinceStart()'

libraries\BLE\FreeRTOS.cpp.o:FreeRTOS.cpp:(.text._ZN8FreeRTOS17getTimeSinceStartEv+0x0): first defined here

libraries\NimBLE-Arduino\FreeRTOS.cpp.o: In function `FreeRTOS::Semaphore::wait(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >)':

FreeRTOS.cpp:(.text._ZN8FreeRTOS9Semaphore4waitENSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE+0x0): multiple definition of `FreeRTOS::Semaphore::wait(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >)'

libraries\BLE\FreeRTOS.cpp.o:FreeRTOS.cpp:(.text._ZN8FreeRTOS9Semaphore4waitENSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE+0x0): first defined here

libraries\NimBLE-Arduino\FreeRTOS.cpp.o: In function `FreeRTOS::Semaphore::timedWait(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, unsigned int)':

FreeRTOS.cpp:(.text._ZN8FreeRTOS9Semaphore9timedWaitENSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEEj+0x0): multiple definition of `FreeRTOS::Semaphore::timedWait(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, unsigned int)'

libraries\BLE\FreeRTOS.cpp.o:FreeRTOS.cpp:(.text._ZN8FreeRTOS9Semaphore9timedWaitENSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEEj+0x0): first defined here

libraries\NimBLE-Arduino\FreeRTOS.cpp.o: In function `FreeRTOS::Semaphore::Semaphore(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >)':

FreeRTOS.cpp:(.text._ZN8FreeRTOS9SemaphoreC2ENSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE+0x0): multiple definition of `FreeRTOS::Semaphore::Semaphore(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >)'

libraries\BLE\FreeRTOS.cpp.o:FreeRTOS.cpp:(.text._ZN8FreeRTOS9SemaphoreC2ENSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE+0x0): first defined here

libraries\NimBLE-Arduino\FreeRTOS.cpp.o: In function `FreeRTOS::Semaphore::Semaphore(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >)':

FreeRTOS.cpp:(.text._ZN8FreeRTOS9SemaphoreC2ENSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE+0x0): multiple definition of `FreeRTOS::Semaphore::Semaphore(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >)'

libraries\BLE\FreeRTOS.cpp.o:FreeRTOS.cpp:(.text._ZN8FreeRTOS9SemaphoreC2ENSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE+0x0): first defined here

libraries\NimBLE-Arduino\FreeRTOS.cpp.o: In function `FreeRTOS::Semaphore::~Semaphore()':

FreeRTOS.cpp:(.text._ZN8FreeRTOS9SemaphoreD2Ev+0x0): multiple definition of `FreeRTOS::Semaphore::~Semaphore()'

libraries\BLE\FreeRTOS.cpp.o:FreeRTOS.cpp:(.text._ZN8FreeRTOS9SemaphoreD2Ev+0x0): first defined here

libraries\NimBLE-Arduino\FreeRTOS.cpp.o: In function `FreeRTOS::Semaphore::~Semaphore()':

FreeRTOS.cpp:(.text._ZN8FreeRTOS9SemaphoreD2Ev+0x0): multiple definition of `FreeRTOS::Semaphore::~Semaphore()'

libraries\BLE\FreeRTOS.cpp.o:FreeRTOS.cpp:(.text._ZN8FreeRTOS9SemaphoreD2Ev+0x0): first defined here

libraries\NimBLE-Arduino\FreeRTOS.cpp.o: In function `FreeRTOS::Semaphore::give()':

FreeRTOS.cpp:(.text._ZN8FreeRTOS9Semaphore4giveEv+0x0): multiple definition of `FreeRTOS::Semaphore::give()'
[... 68 lines cut ...]

libraries\NimBLE-Arduino\FreeRTOS.cpp.o: In function `Ringbuffer::returnItem(void*)':

FreeRTOS.cpp:(.text._ZN10Ringbuffer10returnItemEPv+0x0): multiple definition of `Ringbuffer::returnItem(void*)'

libraries\BLE\FreeRTOS.cpp.o:FreeRTOS.cpp:(.text._ZN10Ringbuffer10returnItemEPv+0x0): first defined here

libraries\NimBLE-Arduino\FreeRTOS.cpp.o: In function `Ringbuffer::send(void*, unsigned int, unsigned int)':

FreeRTOS.cpp:(.text._ZN10Ringbuffer4sendEPvjj+0x0): multiple definition of `Ringbuffer::send(void*, unsigned int, unsigned int)'

libraries\BLE\FreeRTOS.cpp.o:FreeRTOS.cpp:(.text._ZN10Ringbuffer4sendEPvjj+0x0): first defined here

libraries\NimBLE-Arduino\FreeRTOS.cpp.o: In function `FreeRTOS::Semaphore::toString[abi:cxx11]()':

FreeRTOS.cpp:(.text._ZN8FreeRTOS9Semaphore8toStringB5cxx11Ev+0x0): multiple definition of `FreeRTOS::Semaphore::toString[abi:cxx11]()'

libraries\BLE\FreeRTOS.cpp.o:FreeRTOS.cpp:(.text._ZN8FreeRTOS9Semaphore8toStringB5cxx11Ev+0x0): first defined here

collect2.exe: error: ld returned 1 exit status

exit status 1

Error compiling for board Adafruit ESP32 Feather.
```

h2zero, 2021-04-04:
>libraries\BLE\FreeRTOS.cpp.o:C:\Users\X\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\libraries\BLE\src/FreeRTOS.cpp:21: first defined here

libraries\NimBLE-Arduino\FreeRTOS.cpp.o: In function `FreeRTOS::startTask(void (*)(void*), std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, void*, unsigned int)':

looks like the original library is still being linked somewhere in your code. Check the included headers in your project.

societyofrobots, 2021-04-05:
Yeap that was the problem. I had this still in my code:
`#include <BLESecurity.h>`

closed 2021-04-05 by societyofrobots