h2zero/NimBLE-Arduino #218: Device name being trimmed when using the multiconnect example

ketan, 2021-04-04:
Hi,

Thanks for this awesome library and the great documentation and very useful practical examples. I was able to do very much a lot of thing with this that were not possible with the standard arduino bluetooth library for ESP32.

I'm now trying to see if is possible to pair and have my ESP broadcast data to multiple devices at once and I'm using the example from and [here](https://github.com/h2zero/NimBLE-Arduino/blob/master/examples/Refactored_original_examples/BLE_server_multiconnect/BLE_server_multiconnect.ino) I have tweaked the `BLEDevice::init("ESP32");` line to use a longer name (`BLEDevice::init("12345678901234567890")`). It turns out that the BT device name being broadcast is being truncated to 8 characters. See attached screenshots:

[image]

[image]

Is this to be expected? Does anyone have a snippet that allows me to pair and broadcast from ESP to multiple devices?

Thanks

ketan, 2021-04-04:
Funnily enough changing the min preferred interval `pAdvertising->setMinPreferred(0x10)` reduces the device name down to 2 characters. I'm not sure I understand what's happen. Could someone help shine some light on this?

[image]

h2zero, 2021-04-04:
Hi @ketan, the name is being shortened because there is not enough room in the advertising packet for the complete name. By using the max/minPreferred advertisment option it reduces the available advertising space by a further 6 bytes. There are only 31 bytes available in an advertisement, If you have not already done so I suggest turning on scan response data which will provide another 31 bytes.

h2zero, 2021-04-04:
>Is this to be expected? Does anyone have a snippet that allows me to pair and broadcast from ESP to multiple devices?

You can see how this works here:https://github.com/h2zero/NimBLE-Arduino/blob/master/examples/NimBLE_Server/NimBLE_Server.ino

Simply start advertising after being connected, or in a loop/task check if advertising is active and if not start it there.

ketan, 2021-04-04:
Thank you for the valuable insight. I'm closing this issue out since my problem got resolved.

closed 2021-04-04 by ketan