h2zero/NimBLE-Arduino #219: setValue() only sends 252 bytes of data

ln-12, 2021-04-08:
While playing around with the [uart example](https://github.com/h2zero/NimBLE-Arduino/blob/76ce18a10f29cfbda991e00467f442d142e7384b/examples/Refactored_original_examples/BLE_uart/BLE_uart.ino) in the library, I noticed that [`setValue(const std::string &value)`](https://github.com/h2zero/NimBLE-Arduino/blob/76ce18a10f29cfbda991e00467f442d142e7384b/src/NimBLEDescriptor.h#L57) only transmits 252 bytes of data. Is there any reason why it's not 512? How would I send 512 bytes at once?

h2zero, 2021-04-08:
Hi, you will need to change the MTU size. `NimBLEDevice::setMTU(517)` (note that you need to use 517 for 512 bytes due to header info). Both devices will need to support this MTU as well.

ln-12, 2021-04-08:
It works now. Thank you very much for your fast response! :)

closed 2021-04-08 by ln-12

joseitor, 2023-05-19:
I can't sent more than 20 bytes by notification. Can you share your code? Even using `NimBLEDevice::setMTU(517)`, nothing change.