h2zero/NimBLE-Arduino #221: secure BLE appends mysterious '^@' to every xmitted message

societyofrobots, 2021-04-10:
I'm using the refactored BLE_server_multiconnect example, and I changed it to be secure (code below).

It's bonding and receiving data great, however it appends '^@' to every transmitted message and I can't figure out why.

What I should be seeing in BLE serial data:
ccccccccccccc

What I'm actually seeing:
c^@c^@c^@c^@c^@c^@c^@c^@c^@c^@

What am I doing wrong?

```
#include <NimBLEDevice.h>

BLEServer *pServer = NULL;
BLECharacteristic * pTxCharacteristic;
bool deviceConnected = false;
bool oldDeviceConnected = false;
uint8_t txValue = 0;

// See the following for generating UUIDs:
// https://www.uuidgenerator.net/

#define SERVICE_UUID           "6E400001-B5A3-F393-E0A9-E50E24DCCA9E" // UART service UUID
#define CHARACTERISTIC_UUID_RX "6E400002-B5A3-F393-E0A9-E50E24DCCA9E"
#define CHARACTERISTIC_UUID_TX "6E400003-B5A3-F393-E0A9-E50E24DCCA9E"

/**  None of these are required as they will be handled by the library with defaults. **
 **                       Remove as you see fit for your needs                        */  
class MyServerCallbacks: public BLEServerCallbacks {
    void onConnect(BLEServer* pServer) {
      deviceConnected = true;
      BLEDevice::startAdvertising();//adding this line allows for multiple simultaneous BLE connections
    };

    void onDisconnect(BLEServer* pServer) {
      deviceConnected = false;
    }
  /***************** New - Security handled here ********************
  ****** Note: these are the same return values as defaults ********/
    uint32_t onPassKeyRequest(){
      Serial.println("Server PassKeyRequest");
      return 123456; 
    }

    
    void onPassKeyNotify(uint32_t passkey) {
      Serial.println("Passkey Notify");
    }
    
    bool onConfirmPIN(uint32_t pass_key){
      Serial.print("The passkey YES/NO number: ");Serial.println(pass_key);
      return true; 
    }

    void onAuthenticationComplete(ble_gap_conn_desc* desc) {
     if(desc->sec_state.encrypted){
        if(!desc->sec_state.encrypted) {
            Serial.println("Encrypt connection failed - disconnecting");
            /** Find the client with the connection handle provided in desc */
            NimBLEDevice::getClientByID(desc->conn_handle)->disconnect();
            vTaskDelay(5000);//bruteforce protection, 5s
            return;
          }
       }
    }
  /*******************************************************************/
};

class MyCallbacks: public BLECharacteristicCallbacks {
    void onWrite(BLECharacteristic *pCharacteristic) {
      std::string rxValue = pCharacteristic->getValue();

      if (rxValue.length() > 0) {
        Serial.print("Received Value: ");
        for (int i = 0; i < rxValue.length(); i++)
          Serial.print(rxValue[i]);

        if(rxValue.find("F") != -1)
          Serial.println("do stuff");
        }
    }
};

void setup() {
  Serial.begin(115200);

  // Create the BLE Device
  NimBLEDevice::init("NimBLE");
  NimBLEDevice::setPower(ESP_PWR_LVL_P9);

  NimBLEDevice::setSecurityAuth(true, true, true);
  NimBLEDevice::setSecurityPasskey(123456);
  NimBLEDevice::setSecurityIOCap(BLE_HS_IO_DISPLAY_ONLY);
  
  // Create the BLE Server
  pServer = BLEDevice::createServer();
  pServer->setCallbacks(new MyServerCallbacks());

  // Create the BLE Service
  BLEService *pService = pServer->createService(SERVICE_UUID);

  // Create a BLE Characteristic
  pTxCharacteristic = pService->createCharacteristic(
                                        CHARACTERISTIC_UUID_TX,
                                        NIMBLE_PROPERTY::NOTIFY
                                       );

  BLECharacteristic * pRxCharacteristic = pService->createCharacteristic(
                                            CHARACTERISTIC_UUID_RX,
                                            NIMBLE_PROPERTY::WRITE
                                            );

  pRxCharacteristic->setCallbacks(new MyCallbacks());

  // Start the service
  pService->start();

  // Start advertising
  pServer->getAdvertising()->start();
  Serial.println("Waiting a client connection to notify...");
}

void loop() {

    if (deviceConnected) {
      pTxCharacteristic->setValue("c"); // Sending bluetooth message    
      pTxCharacteristic->notify();
      delay(500);
   }
   
    // disconnecting
    if (!deviceConnected && oldDeviceConnected) {
        delay(500); // give the bluetooth stack the chance to get things ready
        pServer->startAdvertising(); // restart advertising
        Serial.println("start advertising");
        oldDeviceConnected = deviceConnected;
    }
    // connecting
    if (deviceConnected && !oldDeviceConnected) {
        // do stuff here on connecting
        oldDeviceConnected = deviceConnected;
    }
}
```

h2zero, 2021-04-10:
Not sure why that happens but I suspect it has to do with setting the value as a string. 
Try this: `pTxCharacteristic->setValue('c', 1);`

societyofrobots, 2021-04-10:
I changed to that line, but it throws the below error.

"candidate expects 1 argument, 2 provided"

Is there a null char getting passed that shouldn't be?

```
C:\Users\x\Documents\Arduino\BLE_uart_test\BLE_uart_test.ino: In function 'void loop()':
BLE_uart_test:149:41: error: no matching function for call to 'NimBLECharacteristic::setValue(char, int)'
       pTxCharacteristic->setValue('c', 1);
                                         ^
In file included from C:\Users\x\Documents\Arduino\libraries\NimBLE-Arduino\src/NimBLEService.h:24:0,
                 from C:\Users\x\Documents\Arduino\libraries\NimBLE-Arduino\src/NimBLEServer.h:26,
                 from C:\Users\x\Documents\Arduino\libraries\NimBLE-Arduino\src/NimBLEDevice.h:35,
                 from C:\Users\x\Documents\Arduino\BLE_uart_test\BLE_uart_test.ino:23:
C:\Users\x\Documents\Arduino\libraries\NimBLE-Arduino\src/NimBLECharacteristic.h:107:23: note: candidate: void NimBLECharacteristic::setValue(const uint8_t*, size_t) <near match>
     void              setValue(const uint8_t* data, size_t size);
                       ^
C:\Users\x\Documents\Arduino\libraries\NimBLE-Arduino\src/NimBLECharacteristic.h:107:23: note:   conversion of argument 1 would be ill-formed:
BLE_uart_test:149:41: error: invalid conversion from 'char' to 'const uint8_t* {aka const unsigned char*}' [-fpermissive]
       pTxCharacteristic->setValue('c', 1);
                                         ^
In file included from C:\Users\x\Documents\Arduino\libraries\NimBLE-Arduino\src/NimBLEService.h:24:0,
                 from C:\Users\x\Documents\Arduino\libraries\NimBLE-Arduino\src/NimBLEServer.h:26,
                 from C:\Users\x\Documents\Arduino\libraries\NimBLE-Arduino\src/NimBLEDevice.h:35,
                 from C:\Users\x\Documents\Arduino\BLE_uart_test\BLE_uart_test.ino:23:
C:\Users\x\Documents\Arduino\libraries\NimBLE-Arduino\src/NimBLECharacteristic.h:108:23: note: candidate: void NimBLECharacteristic::setValue(const string&)
     void              setValue(const std::string &value);
                       ^
C:\Users\x\Documents\Arduino\libraries\NimBLE-Arduino\src/NimBLECharacteristic.h:108:23: note:   candidate expects 1 argument, 2 provided
C:\Users\x\Documents\Arduino\libraries\NimBLE-Arduino\src/NimBLECharacteristic.h:114:23: note: candidate: template<class T> void NimBLECharacteristic::setValue(const T&)
     void              setValue(const T &s) {
                       ^
C:\Users\x\Documents\Arduino\libraries\NimBLE-Arduino\src/NimBLECharacteristic.h:114:23: note:   template argument deduction/substitution failed:
C:\Users\x\Documents\Arduino\BLE_uart_test\BLE_uart_test.ino:149:41: note:   candidate expects 1 argument, 2 provided
       pTxCharacteristic->setValue('c', 1);
                                         ^
exit status 1
no matching function for call to 'NimBLECharacteristic::setValue(char, int)'

```

h2zero, 2021-04-10:
Oops, yeah I suspect that is what's happening. I was thinking about the other setvalue method. It should work with just removing the extra parameter "1" from the call.

societyofrobots, 2021-04-10:
I figured it out.

changed this:
pTxCharacteristic->setValue("c");

to this:
pTxCharacteristic->setValue('c');

societyofrobots, 2021-04-22:
Re-opening this issue, as I'm now transmitting sentences and not just single characters.

This code below transmits what I think is a null character, ^@, when it shouldn't be.
```
pCharacteristic->setValue("Pikachu is yellow.");
pCharacteristic->notify();
```

What I receive in BLE serial:
`Pikachu is yellow.^@`

h2zero, 2021-04-22:
I think you missed the re-open button :).

What application are you using that is showing this? I do not see it in the serial console.

chegewara, 2021-04-22:
Did you check notification length?
The thing is you may send not null terminated string, and using `Serial.println()` will end up like this, just may have different garbage appended.

societyofrobots, 2021-04-22:
> What application are you using that is showing this? I do not see it in the serial console.

I'm using a basic bluetooth serial app on both an Android (Serial Bluetooth) and an iOS phone, same thing happens on both. You can try my code in the OP and verify the issue.

The same code didn't have this issue using the older bluedroid library. It only happens when strings are transmitted:

`pCharacteristic->setValue("c");` <- has ^@

`pCharacteristic->setValue('c');` <- does not have issue

h2zero, 2021-04-22:
I think @chegewara is pointing in the right direction here. If you add `\0` to the end of the string or `\r\n` does this correct the issue?  
The app seems to be printing the data after the proper data so you need to terminate it.

societyofrobots, 2021-05-13:
Currently using NimBLE 1.3 and 2.0.0-alpha1 core, but still no luck.

I tried a bunch of combinations:

```
    pCharacteristic->setValue("\0\n\0a\0");
    pCharacteristic->notify();
    pCharacteristic->setValue("\n\0a\0");
    pCharacteristic->notify();
    pCharacteristic->setValue("\na\0");
    pCharacteristic->notify();
    pCharacteristic->setValue("\na");
    pCharacteristic->notify();
    pCharacteristic->setValue("\na\r\n");
    pCharacteristic->notify();
    pCharacteristic->setValue("\na\r");
    pCharacteristic->notify();
```

This was the output:
```
^@
^@a^@^@
^@a^@^@
a^@^@
a^@
a
^@
a^M^@
```

ketan, 2021-05-13:
Try converting it to a `std::string` or use `setValue(foo, strlen(foo)`

- Ketan

h2zero, 2021-05-13:
Just did a quick test with the code pasted above and monitoring with nRF Toolbox. I was able to reproduce somewhat (didn't get ^@ but instead the message was cut off)

Changing `pCharacteristic->setValue("Pikachu is yellow.");` to `pCharacteristic->setValue(std::string("Pikachu is yellow."));` resolved it.

societyofrobots, 2021-05-13:
> Changing pCharacteristic->setValue("Pikachu is yellow."); to pCharacteristic->setValue(std::string("Pikachu is yellow.")); resolved it.

This fixed it. Maybe should be default for the library? Anyway, thanks!

closed 2021-05-13 by societyofrobots

h2zero, 2021-05-13:
It sort of is the default, in this case the compiler chose to use the template/uint8_t overload which I did not expect.

societyofrobots, 2021-09-08:
I had the same issue again when constructing char arrays and strings to send floats. A lot of junk ^@ was included.

After a lot of hair pulling, this is what I got to work. Hopefully someone finds this useful.

```
    float variable=124.1;
    String txString="\nThe float or int variable is: ";
    txString+= variable;
    pCharacteristic->setValue(std::string(txString.c_str()));
    pCharacteristic->notify();
```