h2zero/NimBLE-Arduino #222: BLE working in Android, but not showing up in

societyofrobots, 2021-04-10:
The below code (refactored BLE_server_multiconnect example with added security) works great on both Android 8.1 and Android 11 phones. No problems.

However, iPhone 10 and iPhone 11 phones aren't even detecting the existence of NimBLE.

What am I doing wrong here?

```
#include <NimBLEDevice.h>

BLEServer *pServer = NULL;
BLECharacteristic * pTxCharacteristic;
bool deviceConnected = false;
bool oldDeviceConnected = false;
uint8_t txValue = 0;

// See the following for generating UUIDs:
// https://www.uuidgenerator.net/

#define SERVICE_UUID           "6E400001-B5A3-F393-E0A9-E50E24DCCA9E" // UART service UUID
#define CHARACTERISTIC_UUID_RX "6E400002-B5A3-F393-E0A9-E50E24DCCA9E"
#define CHARACTERISTIC_UUID_TX "6E400003-B5A3-F393-E0A9-E50E24DCCA9E"

/**  None of these are required as they will be handled by the library with defaults. **
 **                       Remove as you see fit for your needs                        */  
class MyServerCallbacks: public BLEServerCallbacks {
    void onConnect(BLEServer* pServer) {
      deviceConnected = true;
      BLEDevice::startAdvertising();//adding this line allows for multiple simultaneous BLE connections
    };

    void onDisconnect(BLEServer* pServer) {
      deviceConnected = false;
    }
  /***************** New - Security handled here ********************
  ****** Note: these are the same return values as defaults ********/
    uint32_t onPassKeyRequest(){
      Serial.println("Server PassKeyRequest");
      return 123456; 
    }

    
    void onPassKeyNotify(uint32_t passkey) {
      Serial.println("Passkey Notify");
    }
    
    bool onConfirmPIN(uint32_t pass_key){
      Serial.print("The passkey YES/NO number: ");Serial.println(pass_key);
      return true; 
    }

    void onAuthenticationComplete(ble_gap_conn_desc* desc) {
     if(desc->sec_state.encrypted){
        if(!desc->sec_state.encrypted) {
            Serial.println("Encrypt connection failed - disconnecting");
            /** Find the client with the connection handle provided in desc */
            NimBLEDevice::getClientByID(desc->conn_handle)->disconnect();
            vTaskDelay(5000);//bruteforce protection, 5s
            return;
          }
       }
    }
  /*******************************************************************/
};

class MyCallbacks: public BLECharacteristicCallbacks {
    void onWrite(BLECharacteristic *pCharacteristic) {
      std::string rxValue = pCharacteristic->getValue();

      if (rxValue.length() > 0) {
        Serial.print("Received Value: ");
        for (int i = 0; i < rxValue.length(); i++)
          Serial.print(rxValue[i]);

        if(rxValue.find("F") != -1)
          Serial.println("do stuff");
        }
    }
};

void setup() {
  Serial.begin(115200);

  // Create the BLE Device
  NimBLEDevice::init("NimBLE");
  NimBLEDevice::setPower(ESP_PWR_LVL_P9);

  NimBLEDevice::setSecurityAuth(true, true, true);
  NimBLEDevice::setSecurityPasskey(123456);
  NimBLEDevice::setSecurityIOCap(BLE_HS_IO_DISPLAY_ONLY);
  
  // Create the BLE Server
  pServer = BLEDevice::createServer();
  pServer->setCallbacks(new MyServerCallbacks());

  // Create the BLE Service
  BLEService *pService = pServer->createService(SERVICE_UUID);

  // Create a BLE Characteristic
  pTxCharacteristic = pService->createCharacteristic(
                                        CHARACTERISTIC_UUID_TX,
                                        NIMBLE_PROPERTY::NOTIFY
                                       );

  BLECharacteristic * pRxCharacteristic = pService->createCharacteristic(
                                            CHARACTERISTIC_UUID_RX,
                                            NIMBLE_PROPERTY::WRITE
                                            );

  pRxCharacteristic->setCallbacks(new MyCallbacks());

  // Start the service
  pService->start();

  // Start advertising
  pServer->getAdvertising()->start();
  Serial.println("Waiting a client connection to notify...");
}

void loop() {

    if (deviceConnected) {
      pTxCharacteristic->setValue('c'); // Sending bluetooth message    
      pTxCharacteristic->notify();
      delay(500);
   }
   
    // disconnecting
    if (!deviceConnected && oldDeviceConnected) {
        delay(500); // give the bluetooth stack the chance to get things ready
        pServer->startAdvertising(); // restart advertising
        Serial.println("start advertising");
        oldDeviceConnected = deviceConnected;
    }
    // connecting
    if (deviceConnected && !oldDeviceConnected) {
        // do stuff here on connecting
        oldDeviceConnected = deviceConnected;
    }
}
```

h2zero, 2021-04-11:
I don't see anything in the code that would cause this issue. What app are you using to scan for the device?

societyofrobots, 2021-04-11:
iOS -> Settings -> Bluetooth

It just spins and spins but never finds it.

h2zero, 2021-04-11:
yep, that's iOS. install nRFConnect app and you will see it.

societyofrobots, 2021-04-11:
It is showing up with nRF Connect.

Any idea why it's not showing up in the regular iOS scanned list, like it does with the older BLE? I'm trying to get an app to connect with the device.

h2zero, 2021-04-11:
iOS seems to only show classic bluetooth capable devices. NimBLE is BLE only so iOS does not show it natively. The original library has both functionalities which is why it would show up then.

societyofrobots, 2021-04-11:
Interesting, I think that's what's happening. I'll look deeper into this and report back anything useful.

wakwak-koba, 2021-04-13:
I'm not sure because I don't own an IOS device, but maybe the device requires encryption.
hen connecting a HID device to iOS, it didn't work without explicit cryptography.

https://github.com/h2zero/NimBLE-Arduino/pull/182

```c
   BLECharacteristic * pRxCharacteristic = pService->createCharacteristic(
                                             CHARACTERISTIC_UUID_RX,
                                             NIMBLE_PROPERTY::WRITE
                                             );
```

Would you like to improve this part by doing the following?

```c
   BLECharacteristic * pRxCharacteristic = pService->createCharacteristic(
                                             CHARACTERISTIC_UUID_RX,
                                             NIMBLE_PROPERTY::WRITE | NIMBLE_PROPERTY::WRITE_ENC
                                             );
```

societyofrobots, 2021-04-13:
> maybe the device requires encryption.
hen connecting a HID device to iOS, it didn't work without explicit cryptography.
...NIMBLE_PROPERTY::WRITE | NIMBLE_PROPERTY::WRITE_ENC

That was one of the first things I tried. Nope, doesn't work.

I've learned that iOS apps need to be programmed differently for BLE as compared to Bluetooth classic. I don't suppose NimBLE can be upgraded to also use classic? (probably easier said than done?)

wakwak-koba, 2021-04-13:
I have received reports that my modified `NimBLE-Keyboard` (https://github.com/wakwak-koba/ESP32-NimBLE-Keyboard) works on iOS.
Is there any change when the device type is specified using `setAppearance()` ?

h2zero, 2021-04-13:
From my understanding it's only the iOS built in BT scanner that ignores BLE devices and that was intentional as to connect to BLE devices it is expected that there would be a separate app to perform that function. Which makes sense because many BLE devices only advertise data and the iOS scanner will not report this, instead it only shows it in a list. So that leaves it up to the app developers to deal with BLE at the app level, which makes sense because every BLE device is different and could be non-standard while classic bluetooth services are more defined and easily supported by the OS.

As the name NimBLE implies, there is not and probably should not be any support for classic bluetooth, however you might be able to play with the advertised flags to fake BR/EDR capability.

h2zero, 2021-04-13:
>I have received reports that my modified NimBLE-Keyboard (https://github.com/wakwak-koba/ESP32-NimBLE-Keyboard) works on iOS.

Interesting, thanks for that. So it seems iOS will discover devices if they have a bluetooth defined type as well, good to know.

societyofrobots, 2021-04-13:
> Is there any change when the device type is specified using setAppearance() ?

I tried all the below values. All it did was change the BT icon on my Android Pixel phone, but still none of them showed up on my iOS phone.

```
  //pAdvertising->setAppearance(0x19);
  //pAdvertising->setAppearance(0x8000);
  //pAdvertising->setAppearance(0xC103);
  //pAdvertising->setAppearance(0x961);
  //pAdvertising->setAppearance(HID_KEYBOARD);
  //pAdvertising->setAppearance(GENERIC_HID);
  pAdvertising->setAppearance(384);
  //pAdvertising->setAppearance(1153);
```

Should I try messing with `NimBLEHIDDevice()`?

fyi, This link in the comments is now a dead link:
https://github.com/h2zero/NimBLE-Arduino/blob/3990ec2912bdfcc332a2e8635d44abd6113d1eef/src/NimBLEAdvertising.cpp#L781

Use this instead:
https://www.bluetooth.com/wp-content/uploads/Sitecore-Media-Library/Gatt/Xml/Characteristics/org.bluetooth.characteristic.gap.appearance.xml

societyofrobots, 2021-04-14:
> I have received reports that my modified NimBLE-Keyboard (https://github.com/wakwak-koba/ESP32-NimBLE-Keyboard) works on iOS.

I am confirming your code shows up in iOS. Now the question is, why doesn't mine?

h2zero, 2021-04-14:
It likely needs more than just the appearance to show in iOS, I think you may also need to advertise the HID service UUID as well.

societyofrobots, 2021-04-14:
> I have received reports that my modified NimBLE-Keyboard (https://github.com/wakwak-koba/ESP32-NimBLE-Keyboard) works on iOS.

I confirmed it shows up in iOS. However, it's not secure BLE and doesn't require a pin. This tells me it's something about the security that iOS doesn't like.

I discovered that Samsung (Galaxy J3 Orbit android 9, and a Galaxy a10s Android 10) phones are having the same problem. My Pixel and Pixel 3 phones work great, though.

h2zero, 2021-04-14:
If advertising as an HID device security is required, probably with a passkey.

societyofrobots, 2021-04-14:
> I think you may also need to advertise the HID service UUID as well.

I added `addServiceUUID` to my above code, but it's still not showing in iOS. Is this what you meant?

```
  pService->start();

  BLEAdvertising *pAdvertising = BLEDevice::getAdvertising();
  
  pAdvertising->setAppearance(384);
  
  pAdvertising->addServiceUUID(SERVICE_UUID);
  pAdvertising->setScanResponse(true); // must be true or else BLE name gets truncated
  pAdvertising->setMinPreferred(0x0);  // set value to 0x00 to not advertise this parameter

  pServer->getAdvertising()->start();
  Serial.println("Waiting a client connection to notify...");
```

societyofrobots, 2021-04-14:
I tried the default examples, same problem.

Both the NimBLE_Secure_Server and NimBLE_Server examples don't show up in iOS.

So... why does the ESP32-NimBLE-Keyboard display in iOS, but not these server examples?

h2zero, 2021-04-14:
Did you define the service uuid to be the same as in the keyboard hid? That probably needs to be set to the Bluetooth defined uuid for a keyboard (don't remember it right now).

h2zero, 2021-04-14:
I should also add that by doing this you may get it to show in the list but will likely not connect or stay connected without implementing the full keyboard services. 

When iOS connects to something saying it has a service it supports it surely would try to communicate/configure that service and when the device doesn't respond to that I suspect it would disconnect. 

Just thinking that you may be going down a rabbit hole here and could lead to further issues. 

Why do you need the OS to do this anyway if you're writing an app? I think it Would make more sense to add the custom BLE code to your app so you don't need to fake an HID service.

societyofrobots, 2021-04-15:
> Did you define the service uuid to be the same as in the keyboard hid? That probably needs to be set to the Bluetooth defined uuid for a keyboard

How would I do that? For example, in the NimBLE_Secure_Server example, what would I add?

> Why do you need the OS to do this anyway if you're writing an app? I think it Would make more sense to add the custom BLE code to your app so you don't need to fake an HID service.

Both the Android and iOS apps are already written. I can either fix the firmware, or rewrite the apps. Fixing the firmware seemed a lot easier... it worked great with the old bluetooth library, figured I'm just missing something simple.

I'm not that strong of a programmer, but I feel like I'm supposed to be using [NimBLEHIDDevice()](https://github.com/h2zero/NimBLE-Arduino/blob/eefcc32809e35db774a04673bdae9e29243210f0/src/NimBLEHIDDevice.cpp#L27). The [NimBLE Keyboard has an example](https://github.com/wakwak-koba/ESP32-NimBLE-Keyboard/blob/0bcecb6218e28495abb644dfea1783c0a5a8dd15/BleKeyboard.cpp#L118) but it's way over my head. I'm just trying to implement a BLE controlled lock and light, so struggling to understand how that maps over. Or maybe easier to try and mimic the old bluetooth classic?

h2zero, 2021-04-15:
Understandable. I may have a simpler solution for you but you would need to alter the library for your use case.
I have not tested this but if you change [this line](https://github.com/h2zero/NimBLE-Arduino/blob/3990ec2912bdfcc332a2e8635d44abd6113d1eef/src/NimBLEAdvertising.cpp#L428) to just ` m_advData.flags = BLE_HS_ADV_F_DISC_GEN;` and remove [this line](https://github.com/h2zero/NimBLE-Arduino/blob/3990ec2912bdfcc332a2e8635d44abd6113d1eef/src/NimBLEAdvertising.cpp#L432) it might show up in the iOS bluetooth scanner.

societyofrobots, 2021-04-15:
I made the changes, but no effect.

h2zero, 2021-04-15:
That's disappointing. You will probably need to change the service UUID that is advertised then. Here are the HID UUID's
https://github.com/h2zero/NimBLE-Arduino/blob/3990ec2912bdfcc332a2e8635d44abd6113d1eef/src/NimBLEHIDDevice.cpp#L32

You will need to call `addServiceUUID()` with that UUID and probably set the appearance as well, hopefully that will work.

societyofrobots, 2021-04-15:
My above code already has these lines. Is that what you mean?
```
#define SERVICE_UUID           "6E400001-B5A3-F393-E0A9-E50E24DCCA9E"
...
BLEService *pService = pServer->createService(SERVICE_UUID);
...
pAdvertising->setAppearance(GENERIC_HID);
pAdvertising->addServiceUUID(SERVICE_UUID);
```

Or do you mean I should do this?
```
pAdvertising->addServiceUUID(0x180a);
pAdvertising->addServiceUUID(0x1812);
pAdvertising->addServiceUUID(0x180f);
```

h2zero, 2021-04-15:
What I mean is remove `pAdvertising->addServiceUUID(SERVICE_UUID);` and instead use `pAdvertising->addServiceUUID(0x1812);`

societyofrobots, 2021-04-17:
I changed this:
`pAdvertising->addServiceUUID(SERVICE_UUID);`
to this:
`pAdvertising->addServiceUUID((uint16_t)0x1812);`
and now it shows up in iOS and Samsung Galaxy. Progress!

However, it's still not working with my apps on iOS or Samsung Galaxy (Android). It *does* work with Pixel 1 and Pixel 3 (Android). I feel like I'm still missing something in NimBLE that'll fix it.

Here I'm using the WiFiman app, scanning for bluetooth. With the old Bluetooth library, my code shows:
```
Manufacturer: Espressif Inc.
Device Type: Peripheral
Model: Device_Name
Bluetooth Type: Low Energy
```

With NimBLE, both my code and NimBLE-Keyboard shows:
```
Manufacturer: Espressif Inc.
Device Type: Peripheral
Bluetooth Type: Low Energy
```

...the `Model` is missing, although probably not relevant?

At this point it may be easier to fix it in the apps. I'll post back if I figure anything out.

societyofrobots, 2021-04-18:
I now changed:
`pAdvertising->addServiceUUID((uint16_t)0x1812);`
to
`pAdvertising->addServiceUUID((uint16_t)0x1101);`

1101 is for serial data, which I suspected Bluedroid was set to default.

It now works with all Android phones. Yay! It also now shows `Model` in the WiFiman scanner like with Bluedroid.

I'm still having iOS issues but it may be the phone. I'm going to try another one soon.

societyofrobots, 2021-04-19:
Turns out the iOS app was caching BLE improperly. So, problem solved.

The solution was to add this single line:
`pAdvertising->addServiceUUID((uint16_t)0x1101);`

closed 2021-04-19 by societyofrobots

h2zero, 2021-04-19:
Awesome, glad you solved this!

societyofrobots, 2021-05-24:
It stopped working again after a lot of unrelated code changes.

So I changed:
`pAdvertising->addServiceUUID((uint16_t)0x1101);`
to this:
`pAdvertising->addServiceUUID((uint16_t)0x1812);`

And then it began working in iOS again. Go figure!

h2zero, 2021-05-24:
Likely a cache issue again, seems strange. Hopefully it stays working now.

mr-deamon, 2023-06-11:
Hi @societyofrobots 
Have you been able to make it work on android and ios? I used your code and the things you've described here and it's the best result i had so far. But: on my ipad it connects without passkey and does NOT reconnect. On android, it seems to be working.
Do you have any tips or maybe the code of your sketch?
Chris

societyofrobots, 2023-06-19:
@mr-deamon I have not done any additional work on this since my last update. Works fine now on Android and on iOS. However, I haven't tested the latest iOS versions in the last year, so maybe it broke again?

ksiahpar, 2023-08-14:
for BLE visibility in iOS you must add at least one service with 16-bit (short) UUIDs for well-known BLE services. UUID to advertise:

NimBLEService *pService_1 = pServer->getServiceByUUID(BLE_Service_1_UUID);
pAdvertising->addServiceUUID("xxxxxxx");

see below link for some defined service UUID:
https://developer.nordicsemi.com/nRF5_SDK/nRF51_SDK_v4.x.x/doc/html/group___u_u_i_d___s_e_r_v_i_c_e_s.html#gaafd551131de655ea5987ed148170bf2d