h2zero/NimBLE-Arduino #225: No recieve from Callbacks

podaen, 2021-04-22:
Dear,

I want to create a secure server to receive data from an app I created to an esp32. The pin-code was accepted in my android. But something strange happens when the code is accepted.  It doesn't confirm the code in the callbacks. I tryed too make a SecurityCallback class but there ware too many arguments...?? Also the CharacteristicCallbacks doesn't appear when I send data. Does anyone have suggestions on that?

```
#include <NimBLEDevice.h>
#define SERVICE_UUID        "4fafc201-1fb5-459e-8fcc-c5c9c331914b"
#define CHARACTERISTIC_UUID "beb5483e-36e1-4688-b7f5-ea07361b26a8"
#define PASSKEY 123456
NimBLEServer* pServer = NULL;
NimBLECharacteristic* pCharacteristic = NULL;
bool deviceConnected = false;
bool oldDeviceConnected = false;
String StrRecieved;
/**  None of these are required as they will be handled by the library with defaults. **
 **                       Remove as you see fit for your needs                        */
class MyServerCallbacks : public BLEServerCallbacks {
	void onConnect(BLEServer* pServer) {
		deviceConnected = true;
		BLEDevice::startAdvertising();
	};
	void onDisconnect(BLEServer* pServer) {
		deviceConnected = false;
	}
	/***************** New - Security handled here ********************
	****** Note: these are the same return values as defaults ********/
	uint32_t onPassKeyRequest() {
		Serial.println("Server PassKeyRequest");
		return PASSKEY;
		//return 000000;
	}
	bool onConfirmPIN(uint32_t pass_key) {
		Serial.print("The passkey YES/NO number: "); Serial.println(pass_key);
		return true;
	}
	void onAuthenticationComplete(ble_gap_conn_desc desc) {
		Serial.println("Starting BLE work!");
	}
	/*******************************************************************/
};
//class SecurityCallback : public BLESecurityCallbacks {
//
//	uint32_t onPassKeyRequest() {
//		return 000000;
//	}
//
//	void onPassKeyNotify(uint32_t pass_key) {}
//
//	bool onConfirmPIN(uint32_t pass_key) {
//		vTaskDelay(5000);
//		return true;
//	}
//
//	bool onSecurityRequest() {
//		return true;
//	}
//
//	void onAuthenticationComplete(esp_ble_auth_cmpl_t cmpl) {
//		if (cmpl.success) {
//			Serial.println("   - SecurityCallback - Authentication Success");
//		}
//		else {
//			Serial.println("   - SecurityCallback - Authentication Failure*");
//			pServer->removePeerDevice(pServer->getConnId(), true);
//		}
//		BLEDevice::startAdvertising();
//	}
//};

/** Handler class for characteristic actions */
class CharacteristicCallbacks : public NimBLECharacteristicCallbacks {
	void onWrite(NimBLECharacteristic* pCharacteristic) {
		std::string value = pCharacteristic->getValue();
		Serial.println("here");
		if (value.length() > 0) {
			StrRecieved = "";
			for (int i = 0; i < value.length(); i++) {
				// Serial.print(value[i]); 
				StrRecieved = StrRecieved + value[i];
			}
			Serial.println("*********");
			Serial.print("Recieved = ");
			Serial.println(StrRecieved); 
		}
	}
};
void setup() {
	Serial.begin(115200);
	Serial.println("Starting NimBLE Server");
	/** sets device name */
	NimBLEDevice::init("NimBLE-Arduino");
	/** Optional: set the transmit power, default is 3db */
	NimBLEDevice::setPower(ESP_PWR_LVL_P9); /** +9db */
	NimBLEDevice::setSecurityAuth(true, true, true);
	//NimBLEDevice::setSecurityAuth(false, false, true);
	NimBLEDevice::setSecurityPasskey(PASSKEY);
	//NimBLEDevice::setSecurityPasskey(PASSKEY);
	NimBLEDevice::setSecurityIOCap(BLE_HS_IO_DISPLAY_ONLY);
	//NimBLEServer* pServer = NimBLEDevice::createServer();
	//BLEDevice::setSecurityCallbacks(new SecurityCallback());
	pServer = NimBLEDevice::createServer();
	pServer->setCallbacks(new MyServerCallbacks());
	NimBLEService* pService = pServer->createService(SERVICE_UUID);
	NimBLECharacteristic* pCharacteristic = pService->createCharacteristic(CHARACTERISTIC_UUID, NIMBLE_PROPERTY::READ | NIMBLE_PROPERTY::READ_ENC | NIMBLE_PROPERTY::READ_AUTHEN);
	pCharacteristic->setCallbacks(new CharacteristicCallbacks());
        pService->start();
	NimBLEAdvertising* pAdvertising = NimBLEDevice::getAdvertising();
	pAdvertising->addServiceUUID(SERVICE_UUID);
	//pAdvertising->setScanResponse(true);
	pAdvertising->start();
	Serial.println("Advertising Started");
	////NimBLEDevice::setMTU(128);//128bytes
	//Serial.println("recieving size MTU");
	//Serial.println(NimBLEDevice::getMTU());
}
void loop() {}
```

h2zero, 2021-04-22:
Hi @podaen the issue with the authentication complete callback is that the parameters are not correct in your code;
`void onAuthenticationComplete(ble_gap_conn_desc desc) ` should be
`void onAuthenticationComplete(ble_gap_conn_desc *desc) ` as the callback takes a pointer.

Also your characteristic callback does not match the properties of the characteristic, if you are wanting to write values to this you should change all the `READ_X` properties to `WRITE_X`

podaen, 2021-04-22:
It works. I adapted it like you specified

```
void onAuthenticationComplete(ble_gap_conn_desc* desc) {
		Serial.println("Starting BLE work!");
	}
```

and 

`NimBLECharacteristic* pCharacteristic = pService->createCharacteristic(CHARACTERISTIC_UUID, NIMBLE_PROPERTY::WRITE | NIMBLE_PROPERTY::WRITE_ENC | NIMBLE_PROPERTY::WRITE_AUTHEN);`

I'm now recieving data, but only 20bytes... I checked the recieving size MTU and it gives me 255 like config in the nimconfig.h. How come?

h2zero, 2021-04-22:
The app you are connecting with needs to request "exchange MTU"  in order to increase the size of the data packet.

podaen, 2021-04-22:
hmmm. I have the same app working on the hug BLE library by setting the MTU. I have looked in what I can do in the app and I just can make an MTU request. I have tried it but with the same results.

h2zero, 2021-04-22:
Hard to say without seeing a log from either device. I know it works fine with nRFConnect and other esp32 clients. The flow should be connect > exchange mtu > read/write characteristics. If you are doing this in your app already please post a log from the server.

podaen, 2021-04-22:
You are right! I forgot... By doing so I did some changes in the app. I found out something went wrong after an update of the bluetoothLE extension 2.0.

closed 2021-04-22 by podaen