h2zero/NimBLE-Arduino #226: How to encoding onWrite Characteristic value

podaen, 2021-04-25:
Dear,

How can decode a recieved value (String) from UTF-16 little endian encoding or UTF-8 encoding and print it correctly in the serial monitor? 

```
void onWrite(NimBLECharacteristic* pCharacteristic) {
		std::string value = pCharacteristic->getValue();
```

podaen, 2021-04-25:
when I change

`NimBLECharacteristic* pCharacteristic = pService->createCharacteristic(CHARACTERISTIC_UUID, NIMBLE_PROPERTY::WRITE | NIMBLE_PROPERTY::WRITE_ENC | NIMBLE_PROPERTY::WRITE_AUTHEN | NIMBLE_PROPERTY::READ | NIMBLE_PROPERTY::READ_ENC | NIMBLE_PROPERTY::READ_AUTHEN);`

too

NimBLECharacteristic* pCharacteristic = pService->createCharacteristic(CHARACTERISTIC_UUID, NIMBLE_PROPERTY::WRITE_ENC | NIMBLE_PROPERTY::READ_ENC );

I don't recieve anything any more. Is this because I do a pinkey request or has the WRITE_ENC nothing to do with UTF-8?

h2zero, 2021-04-25:
>How can decode a recieved value (String) from UTF-16 little endian encoding or UTF-8 encoding and print it correctly in the serial monitor?

You'll need to use a loop and convert/print each byte from the `.data()` method of std::string.

>NimBLECharacteristic* pCharacteristic = pService->createCharacteristic(CHARACTERISTIC_UUID, NIMBLE_PROPERTY::WRITE_ENC | NIMBLE_PROPERTY::READ_ENC );

You need to have `NIMBLE_PROPERTY::WRITE` and `NIMBLE_PROPERTY::READ` as well the the `_ENC` suffixed properties.

podaen, 2021-04-26:
It doesn't sound difficult like you are telling this... but is very hard to do.

h2zero, 2021-04-26:
If all you want to see is the hex data you can use this:
```
char* pHex = NimBLEUtils::buildHexData(nullptr, YOURSTRING.data(), YOURSTRING.length());
Serial.println(pHex);
free(pHex);
```

podaen, 2021-04-26:
If I could just go I step further with 

std::string value = pCharacteristic->getValue();

and recieve evrything I have send.

podaen, 2021-04-26:
Is there an other way to recieve my string?

podaen, 2021-04-26:
I don't have the option too split it in too Characteristics soo I am stuck on this.

ketan, 2021-04-26:
BLE fundamentally lets you write packets of 20 bytes (configurable upto a maxium of around 500 bytes by changing the MTU). There is no mechanism to transfer "strings" or custom types. You'll need to convert all your data (utf-16 LE, in your case) to whatever MTU size you're using (20, if you're unsure what this is).

For an example. I'm sending a long string like this:

```
void send_ble_uart(const uint8_t* str, int size) {
  int offset          = 0;
  int remainingLength = size;

  while (remainingLength > 0) {
    int bytesToSend = min(20, remainingLength);
    // send  20 bytes until all the bytes are sent
    pTxCharacteristic->setValue((str + offset), bytesToSend);
    pTxCharacteristic->notify();
    remainingLength -= bytesToSend;
    offset += bytesToSend;
  }
}
```

On the receiver side, you'd subscribe to the characteristic and read the byte chunks and wait until you've read all the bytes.

```

pRxCharacteristic->setCallbacks(receiveCallback);

```

ketan, 2021-04-26:
Hit send too quickly.... On the receiver side, you'd do this:

```
class ReceiveCallback : public NimBLECharacteristicCallbacks {
 public:
  ReceiveCallback(bool* messageAvailable) {
    this->messageAvailable = messageAvailable;
  }

  std::string rxValue;

 private:
  bool* messageAvailable;

  void onWrite(NimBLECharacteristic* pCharacteristic) {
    rxValue = pCharacteristic->getValue();
    *messageAvailable = true;
  }
};

size_t readBytes(char* buf, size_t maxBufLength) {
  size_t bytesRead = 0;
  if (messageAvailable) {
    bytesRead = receiveCallback->rxValue.size();
    debugI("Message !%s! size is: %d", receiveCallback->rxValue.c_str(), bytesRead);
    memcpy(buf, receiveCallback->rxValue.c_str(), bytesRead);
    buf[bytesRead] = '\0';
    messageAvailable = false;
  }
  return bytesRead;
}

bool messageAvailable;
char inputBuffer[64];

void setup() {
  ...
  receiveCallback = new ReceiveCallback(&messageAvailable)
  pRxCharacteristic->setCallbacks(receiveCallback);
  ...
}

void loop() {
  size_t count = readBytes(inputBuffer, sizeof(inputBuffer)); 

  if (count) {
    // you got a packet of data. Most likely 20 bytes, but could be smaller. Collect all your packets and decode it.
  }
}

```

h2zero, 2021-04-26:
Sorry I misunderstood and thought you already had the data and wanted to print it on the serial monitor. @ketan has the right solution to getting the data.

Also there are templates to get the data in a different format instead of std::string; `getValue<DATA_TYPE>()`

podaen, 2021-04-26:
Maybe I have explaned myself clear. I m try too understand what Ketan is telling me. What I have tried before... it make sens too me.

ketan, 2021-04-26:
There's an obvious timing error in my code, if you send too many packets before you read them in the loop, it's possible you skip packets. You can handle this better in your `ReceiveCallback` and have it collect all the bytes in an array until you think it is done receiving the entire payload.

podaen, 2021-04-26:
So you have made your own Characteristic/callbacks... that's nice. But that's far away from my bed. And I did see there were some timings involved in the handling of properties where I did not have a clou what was happening. I am now start to understand it.

podaen, 2021-07-03:
@ketan I simply solved it by sending decimal bytes in stead of a string. Now I can recieve strings longer dan 20bytes.

closed 2021-07-19 by podaen