h2zero/NimBLE-Arduino #227: Devices not being reported in scan callback but added to scan results list

wesee-dev, 2021-04-30:
I created a custom scan callback and if the device name does not match the one from the device type that i want to connect, I erase it from scanned devices list, during the onResult callback.
But I noticied that some devices were being added to the list even though they shouldn't be there.
After investigating I found that the callback is not called during this section of code in file NimBLEScan.cpp: 

```cpp
            if (pScan->m_pAdvertisedDeviceCallbacks) {
                // If not active scanning or scan response is not available
                // report the result to the callback now.
                if(pScan->m_scan_params.passive ||
                  (advertisedDevice->getAdvType() != BLE_HCI_ADV_TYPE_ADV_IND &&
                   advertisedDevice->getAdvType() != BLE_HCI_ADV_TYPE_ADV_SCAN_IND))
                {
                	NIMBLE_LOGI(LOG_TAG, "Passive result!");
                    advertisedDevice->m_callbackSent = true;
                    pScan->m_pAdvertisedDeviceCallbacks->onResult(advertisedDevice);

                // Otherwise, wait for the scan response so we can report the complete data.
                } else if (event->disc.event_type == BLE_HCI_ADV_RPT_EVTYPE_SCAN_RSP) {
                	NIMBLE_LOGI(LOG_TAG, "Wait for scan response result!");
                    advertisedDevice->m_callbackSent = true;
                    pScan->m_pAdvertisedDeviceCallbacks->onResult(advertisedDevice);
                }
                else{
/* !Prevents onResult from running! PS: This can be called while waiting for scan response */
                	NIMBLE_LOGI(LOG_TAG, "Won't report callback, event if device was added to list!");
                	NIMBLE_LOGI(LOG_TAG, "getAdversitingType() = %d", advertisedDevice->getAdvType());
                	NIMBLE_LOGI(LOG_TAG, "event->disc.event_type = %d", event->disc.event_type);
                }
                // If not storing results and we have invoked the callback, delete the device. 
                if(pScan->m_maxResults == 0 && advertisedDevice->m_callbackSent) {
                    pScan->erase(advertisedAddress);
                }
            }
            else{
            	 NIMBLE_LOGI(LOG_TAG, "No custom callback defined!");
            }
```

The custom onResult is not called when:
advertisedDevice->getAdvType() and event->disc.event_type values  do not satisfy the if->else if block as comment in the code.
I don't know if this a bug, or it's the it supposed to work. But still this is preventing some scan results to be reported in the callback, which is not great.
The logs are included below.
Thank you, I really appreciate NimBLE! :)

# 
Logs:

```
D NimBLEScan: ">> start(duration=30)"
D NimBLEScan: "<< start()"
I NimBLEScan: "New advertiser: 21:04:e1:fb:85:23"
I NimBLEScan: "Passive result!"
[Info] Advertised device found!
[Info] Device address: 21:04:e1:fb:85:23
D NimBLEScan: "erase device: 21:04:e1:fb:85:23"
I NimBLEScan: "New advertiser: 10:98:c3:cf:4a:fb"
I NimBLEScan: "Won't report callback, event if device was added to list!"
I NimBLEScan: "getAdversitingType() = 0"
I NimBLEScan: "event->disc.event_type = 0"
I NimBLEScan: "New advertiser: a4:c1:38:9b:24:8e"
I NimBLEScan: "Won't report callback, event if device was added to list!"
I NimBLEScan: "getAdversitingType() = 0"
I NimBLEScan: "event->disc.event_type = 0"
I NimBLEScan: "Updated advertiser: a4:c1:38:9b:24:8e"
I NimBLEScan: "Wait for scan response result!"
[Info] Advertised device found!
[Info] Device name: LYWSD03MMC
[Info] Device address: a4:c1:38:9b:24:8e
[Info] Found a M3 thermometer!
[Info] Found devices: 1
I NimBLEScan: "New advertiser: 7c:88:72:44:a8:59"
I NimBLEScan: "Won't report callback, event if device was added to list!"
I NimBLEScan: "getAdversitingType() = 2"
I NimBLEScan: "event->disc.event_type = 2"
I NimBLEScan: "New advertiser: 34:f1:50:a2:6f:bd"
I NimBLEScan: "Won't report callback, event if device was added to list!"
I NimBLEScan: "getAdversitingType() = 0"
I NimBLEScan: "event->disc.event_type = 0"
I NimBLEScan: "Updated advertiser: 34:f1:50:a2:6f:bd"
I NimBLEScan: "Wait for scan response result!"
[Info] Advertised device found!
[Info] Device address: 34:f1:50:a2:6f:bd
D NimBLEScan: "erase device: 34:f1:50:a2:6f:bd"
I NimBLEScan: "Updated advertiser: 10:98:c3:cf:4a:fb"
I NimBLEScan: "Wait for scan response result!"
[Info] Advertised device found!
[Info] Device name: EssenzaTwo_1098C3CF4AFB
[Info] Device address: 10:98:c3:cf:4a:fb
[Info] Found a M3 thermometer!
[Info] Found devices: 2
[Info] (No whitelist) Maximum number of devices found!
D NimBLEScan: ">> stop()"
D NimBLEScan: "<< stop()"
[Info] Scan finished!
[Info] Devices found in scan: 
[Info] 10:98:c3:cf:4a:fb
[Info] a4:c1:38:9b:24:8e
[Info] 7c:88:72:44:a8:59
```

h2zero, 2021-04-30:
Thanks for reporting, definitely a bug. 

Please try changing: 
```
if(pScan->m_scan_params.passive ||
   (advertisedDevice->getAdvType() != BLE_HCI_ADV_TYPE_ADV_IND &&
    advertisedDevice->getAdvType() != BLE_HCI_ADV_TYPE_ADV_SCAN_IND))
{
```

to :
```
if(pScan->m_scan_params.passive ||
   advertisedDevice->getAdvType() != BLE_HCI_ADV_TYPE_ADV_SCAN_IND)
{
```

Let me know if that works for you.

wesee-dev, 2021-04-30:
Unfortunately did not work. 
What changed:
Some devices are no longer reporting name in the response anymore. 
Were reporting before with active scan enabled. (Probably the library is no longer waiting for scan response, due to removing BLE_HCI_ADV_TYPE_ADV_IND).

Some devices still are not triggering the callback. Adversiting type and event do not meet conditions to call onResult.
Log:
´´´
I NimBLEScan: "New advertiser: 67:03:31:d5:22:94"
I NimBLEScan: "Won't report callback, event if device was added to list!"
I NimBLEScan: "getAdversitingType() = 2" # Equals to BLE_HCI_ADV_TYPE_ADV_SCAN_IND
I NimBLEScan: "event->disc.event_type = 2" #Different from BLE_HCI_ADV_RPT_EVTYPE_SCAN_RSP
``

h2zero, 2021-05-01:
Thanks for testing, I'll look into this more when I get home tonight.

h2zero, 2021-05-01:
I think the explanation for the device you see in the logs not calling the callback is that it is not sending a scan response.
```
I NimBLEScan: "New advertiser: 7c:88:72:44:a8:59"
I NimBLEScan: "Won't report callback, event if device was added to list!"
I NimBLEScan: "getAdversitingType() = 2"
I NimBLEScan: "event->disc.event_type = 2"
```
Here it is advertising the type `BLE_HCI_ADV_TYPE_ADV_SCAN_IND` (Scannable Undirected advertising).
```
I NimBLEScan: "New advertiser: 34:f1:50:a2:6f:bd"
I NimBLEScan: "Won't report callback, event if device was added to list!"
I NimBLEScan: "getAdversitingType() = 0"
I NimBLEScan: "event->disc.event_type = 0"
I NimBLEScan: "Updated advertiser: 34:f1:50:a2:6f:bd"
I NimBLEScan: "Wait for scan response result!"
[Info] Advertised device found!
[Info] Device address: 34:f1:50:a2:6f:bd
```
Here we can see that the scan response either did not arrive or was missed for some other reason. So it is advertising itself as having a scan response and therefore we do not invoke the callback until we get the response so the data will be complete for the application. If the data does not get received before the scan ends it will be in the list still as we are waiting for the data.

wesee-dev, 2021-05-01:
Thanks for investigating the issue. 
I implemented some extra filtering after the scan completes. 
Since the device never sends the scan response, its name is always empty, so it's easy to filter this results.

closed 2021-05-03 by wesee-dev