h2zero/NimBLE-Arduino #228: Connecting client without scanning

Murtaza7745, 2021-05-01:
Hello there,
I am new to BLE and using Arduino IDE and would like some help connecting ESP32 as a BLE client to a known BLE server address without scanning.
I am stuck and need some help.
For now, I am using the NIMBLE_CLIENT example, which too much time to find the server if surrounded by many other BLE servers.

PLEASE HELP.

h2zero, 2021-05-01:
You can supply an address to the create client method like this: `NimBLEDevice::createClient("AA:BB:CC:DD:EE:FF");` then you can call pClient->connect();

Murtaza7745, 2021-05-02:
I am getting an error when I use the above method
"conversion from 'const char [18]' to 'NimBLEAddress' is ambiguous"

Murtaza7745, 2021-05-02:
#include <NimBLEDevice.h>

static NimBLEClient* pClient;

void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);
  Serial.println("Starting NimBLE Client");
  NimBLEDevice::init("");
  NimBLEDevice::setPower(ESP_PWR_LVL_P9);
  
  pClient = NimBLEDevice::createClient("6c:36:3e:48:71:fb");
  Serial.println("New client created");
  pClient->connect();
  Serial.print("Connected to: ");
  Serial.println(pClient->getPeerAddress().toString().c_str());
  Serial.print("RSSI: ");
  Serial.println(pClient->getRssi());
}

void loop() {
  // put your main code here, to run repeatedly:

}

h2zero, 2021-05-02:
You've connected to it, now you need to do something with that connection. Have a look at the examples, if you know the services/characteristics of the device you can read/write/subscribed to notifications on them now.

Murtaza7745, 2021-05-02:
I have not connected to it, I have created a server and advertised it in nrfconnect app from my mobile phone.
It doesn't show connected.

What is the issue here, can you run the code mentioned above and tell me what is happening?

Thank you:)

h2zero, 2021-05-02:
Please try adding this to your loop:
```
if (!pClient->isConnected()) {
    pClient->connect();
} else {
   Serial.println("Connected");
}
delay(1000);
```
Let me know what it does, also you could enable logging to see what's happening.

Murtaza7745, 2021-05-03:
I tried using the above method.

**This is the code:**

> 

> include <NimBLEDevice.h>
> static NimBLEClient* pClient;
> 
> void setup() {
>   // put your setup code here, to run once:
>   Serial.begin(115200);
>   Serial.println("Starting NimBLE Client");
>   NimBLEDevice::init("");
>   NimBLEDevice::setPower(ESP_PWR_LVL_P9);
>   
>   pClient = NimBLEDevice::createClient(NimBLEAddress("53:6e:03:7c:c4:54"));
>   Serial.println("New client created");
>   pClient->connect();
>   Serial.print("Connected to: ");
>   Serial.println(pClient->getPeerAddress().toString().c_str());
>   Serial.print("RSSI: ");
>   Serial.println(pClient->getRssi());
> }
> 
> void loop() {
>   // put your main code here, to run repeatedly:
>   if (!pClient->isConnected()) {
>     pClient->connect();
>   } else {
>    Serial.println("Connected");
>   }
>   delay(1000);
> }

**And this is the output from serial in Core debug level: verbose** 

> Starting NimBLE Client
> I NimBLEDevice: "BLE Host Task Started"
> I NimBLEDevice: "NimBle host synced."
> D NimBLEDevice: ">> setPower: 7 (type: 11)"
> D NimBLEDevice: "<< setPower"
> New client created
> D NimBLEClient: ">> connect(53:6e:03:7c:c4:54)"
> D NimBLEClient: "Got Client event "
> E NimBLEClient: "Connection failed; status=13 "
> Connected to: 53:6e:03:7c:c4:54
> RSSI: D NimBLEClient: ">> getRssi()"
> E NimBLEClient: "<< getRssi(): Not connected"
> 0
> D NimBLEClient: ">> connect(53:6e:03:7c:c4:54)"
> D NimBLEClient: "Got Client event "
> E NimBLEClient: "Connection failed; status=13 "
> D NimBLEClient: ">> connect(53:6e:03:7c:c4:54)"
> D NimBLEClient: "Got Client event "
> E NimBLEClient: "Connection failed; status=13 "
> D NimBLEClient: ">> connect(53:6e:03:7c:c4:54)"
> D NimBLEClient: "Got Client event "
> E NimBLEClient: "Connection failed; status=13 "
> D NimBLEClient: ">> connect(53:6e:03:7c:c4:54)"
> D NimBLEClient: "Got Client event "
> E NimBLEClient: "Connection failed; status=13 "
> D NimBLEClient: ">> connect(53:6e:03:7c:c4:54)"
> D NimBLEClient: "Got Client event "
> E NimBLEClient: "Connection failed; status=13 "
> D NimBLEClient: ">> connect(53:6e:03:7c:c4:54)"
> D NimBLEClient: "Got Client event "
> E NimBLEClient: "Connection failed; status=13 "
> D NimBLEClient: ">> connect(53:6e:03:7c:c4:54)"
> D NimBLEClient: "Got Client event "
> E NimBLEClient: "Connection failed; status=13 "
> D NimBLEClient: ">> connect(53:6e:03:7c:c4:54)"

**What should I do next?**
PLEASE HELP.

h2zero, 2021-05-03:
Thank you for providing the debug log. As you can see the connection failed with the code `13`. 
That code mean this:

0x0d | BLE_HS_ETIMEOUT | Operation timed out.
-- | -- | --

In this case I suspect that either the peer is not responding or the address is incorrect. You will not be able to connect to a device that is not advertising, that is why scanning first is the best approach.

Murtaza7745, 2021-05-03:
Both the peer and the address are correct, I have checked it very carefully.
It is also advertising since when I use the client example, it works perfectly.
But when I run this code, it's not working.

Please look into it

h2zero, 2021-05-03:
>It is also advertising since when I use the client example, it works perfectly.

That is interesting, there is one other parameter to the address that needs to be considered in this case. If you use the client example and  in the scan callback add this line `Serial.printf("address type: %u\n", advertisedDevice->getAddress()->getType());` it will print a valuable piece of information.

The default address type is 0, if the peer you're connecting to does not have that type you will need to specify it in the address parameters when creating the client. If this is not a random static address then connecting without scanning will not work.

Murtaza7745, 2021-05-05:
The address type i am getting is 1.
What should I do now?

h2zero, 2021-05-05:
You need to specify the address type in the NimBLEAddress constructor, the default is 0 and your device is using type 1 (random static).

```pClient = NimBLEDevice::createClient(NimBLEAddress("53:6e:03:7c:c4:54"));
```
Should be changed to :
```
pClient = NimBLEDevice::createClient(NimBLEAddress("53:6e:03:7c:c4:54", 1));
```

h2zero, 2021-05-17:
@Murtaza7745 any updates on this? Did the suggested change work for you?

jtwaleson, 2021-07-19:
@h2zero I was struggling with the same issue, can confirm that specifying `1` in the constructor works! Thank you!

h2zero, 2021-07-19:
@jtwaleson Great, thank you for the feedback :)

closed 2022-01-19 by h2zero