h2zero/NimBLE-Arduino #229: Suggestions needed for next release!

h2zero, 2021-05-05:
Hi all, I'm looking for some suggestions to include with the next release. This is predicted to be the last release of 1.x.x that will maintain the current compatibility with the Arduino core BLE library, in release 2.x.x all the deprecated classes/functions will be removed/replaced.

Currently on my todo list is:
- [Server] Add indexing/iterating of connected clients to get their information.
- [Server] Statically create characteristics/descriptors and expose addCharacteristic/addDescriptor methods.
- [Client] Improve descriptor retrieval by UUID reliability.
- [Scan] Invoke callback for any advertised devices that have not invoked it when scan is stopped manually.
- [general] Update core to NimBLE 1.3.0 / IDF master + cleanup logging

I'd like to put more on this list before I break stuff, please post your suggestions/desires/comments in this issue.

thorrak, 2021-05-05:
Are there already `#warning`'s for the deprecated items that are going away? 

I don't recall seeing anything in _my_ use of the library, but it might be that I'm not using enough of the old functionality...

h2zero, 2021-05-05:
Yes, all the affected items have been marked deprecated for quite a while now. If you've been compiling with -Wall you would have seen them. 

To be clear, nothing major in respect to compatibility is planned to change beyond what has been deprecated already but there may be some minor tweaks to the API as testing requires.

h2zero, 2021-05-11:
Adding to my list:
Whitelist scan/advertising/connection support.

closed 2021-08-03 by h2zero