h2zero/NimBLE-Arduino #230: Return type of `gap_event_handler`

ketan, 2021-05-06:
Could someone clarify what is the significance of the return value of this typedef:

`typedef int (*gap_event_handler)(ble_gap_event *event, void *arg);`. This is used by `NimBLEDevice::setCustomGapHandler`

I'm trying to see if there's a callback I can use to determine the MTU supported by various clients for my ESP32 to determine what is the longest string I can "notify". Is there a better way to do this?

chegewara, 2021-05-06:
I dont know if it will works with nimble, but in BLE library you need this event:

https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/bluetooth/esp_gattc.html#_CPPv4N24esp_ble_gattc_cb_param_t7cfg_mtuE
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/bluetooth/esp_gattc.html#_CPPv4N24esp_ble_gattc_cb_param_t23gattc_cfg_mtu_evt_paramE

It will return real MTU in current connection. It is GATTC now GAP.

h2zero, 2021-05-06:
Hi @ketan that is the signature of a callback function you can register for receiving GAP events. You can see how it is used in the server gap event handler [here](https://github.com/h2zero/NimBLE-Arduino/blob/d1d7db73534e1d39e45ffeb66139a325ba7a8045/src/NimBLEServer.cpp#L353). NimBLE uses GAP events for everything, not separated into GATTC/GATTS/GAP the way bluedroid does, so the MTU event is available through this callback.

Note that I am currently working on an API to more easily access the connected peer information including MTU, I expect it to be available shortly.

h2zero, 2021-05-07:
Just pushed a commit that addresses this; 206cb3d

h2zero, 2021-05-23:
@ketan were you able to resolve this using the new API?

ketan, 2021-05-24:
Yes this is resolved. But I was really hoping there was an
events/notifications api. I'm currently having to calculate the mtu before
sending each packet over BLE, and having a notification api would be much
nicer.

h2zero, 2021-05-24:
Sorry to hear that it wasn't the solution you were looking for. Have you tried to use the callback?

closed 2022-01-19 by h2zero