h2zero/NimBLE-Arduino #231: Unknown characteristic

podaen, 2021-05-07:
Hardware:
Board: ESP-WROOM-32
Core Installation/update date: 01/MAY/2021
IDE name: Arduino IDE

I am able to get a discription as text with 2901 and a characteristics as text on the client by calling the format in the 2904 but can I set the name of the characteristic to "my characteristic" in stead of "unknown characteristic" by the server to the client?

podaen, 2021-05-07:
I found this...

0x1800 OurService name
0x2A00 Devicename
0x2A01 characteristic holds the appearance value
0x2A04 parameters used to establish a connection

Correct me if I am wrong, the service name can be set by creating a characteristic with uuid 0x1800 and the 0x2A01 will give the characteristic name.

chegewara, 2021-05-07:
Its not like that. Characteristics and service name is defined in nRF connect app. If UUID is recognized then you see some name otherwise you see `Unknown`. If you have your own app then you can assign name.
For example you can see names for UART service/characteristics, because it is nordic semiconductor app and UART is also defined by nordic and used as standard.

podaen, 2021-05-07:
Thats true. If you press long enough on the characteristic and edit it, your able to set the name and format. What I m trying too do is let my program decide the name. I think I have chance on the 0x2803. But assigning a value to it, I am not capable to store my value to it I think.

The issue is when other apps are using the same predefined uuid the will command my program when the esp is bounded, unless I define attrubutes too it.

podaen, 2021-05-08:
I followd this tutorial

[Link](https://devzone.nordicsemi.com/nordic/short-range-guides/b/bluetooth-low-energy/posts/ble-characteristics-a-beginners-tutorial)

But I it diden't work out for me to set the name of the service. But's not so importented too me, it is just for debugging.

I did read in order too connect to an ios device there are requerements:

> 
> When you design a BLE device communicating with Apple products such as iPhone, iPad, and Mac, you must follow Apple’s design guidelines. To establish and maintain a successful connection, you must follow both the specifications of Bluetooth SIG and Apple’s guidelines. These guidelines define the role, advertising PDU, advertising interval, connection parameters, data packet length extension, privacy, permissions, pairing, MTU size, services, and GATT Server for the BLE device. A BLE implementation on an iOS device differs from that on an Android device in parameters such as advertising PDU, advertising data, advertising interval, connection parameters and privacy.
> 
> 1. Advertising PDU: the iOS device does not support the ADV_DIRECT_IND advertising PDU. Use only one of the three advertising PDUs: ADV_IND, ADV_NOCONN_IND, and ADV_SCAN_IND.
> 2. Advertising Data: The advertising data should contain at least the TX power level, local name, and services flags.
> 3. Advertising Interval: The recommended advertising pattern and advertising intervals are as follows:
> 
> First, advertise at 20-ms intervals for at least 30 seconds. If not discovered after 30 seconds, you may change to one of the following longer intervals: 152.5 ms, 211.25 ms, 318.75 ms, 417.5 ms, 546.25 ms, 760 ms, 852.5 ms, 1022.5 ms, 1285 ms. These exact specific intervals must be used. Even the slightest deviation may significantly increase the time to discovery.
> 
> 4. Connection Parameters: The parameters must follow certain rules and formulas. Some iOS devices scale the interval to 30 ms to balance power and performance constraints if you set Interval Min == Interval Max == 15 ms. The connection interval down to 11.25 ms may be accepted if the connected services include the HID service. The iOS device will not read or use the parameters in the Peripheral Preferred Connection Parameters characteristic, so the peripheral should send an L2CAP Connection Parameter Update Request if you need to update the connection parameters. The rules of the connection parameters are as follows:

[Source: Making a BLE Device Discoverable on iOS Devices](https://community.cypress.com/t5/Knowledge-Base-Articles/Making-a-BLE-Device-Discoverable-on-iOS-Devices-KBA223312/ta-p/250101)

Does anybody has a simple way to do that? This is what I have for now.

```
0x1800 Generic Access  // This is a service that's allready in nfr connect.
Contains:
	2A00 = Devicename (is set)
	2A01 = CharacteristicAppearance (is set, but unknowed)
	2A04 = Characteristic: Peripheral Preferred Connection Parameters (not in it)
	        define MIN_CONN_INTERVAL , MAX_CONN_INTERVAL, SLAVE_LATENCY, CONN_SUP_TIMEOUT.

1801 Generic Attribute // This is a service that's allready in nfr connect.
	0x2A05 = Characteristic: Service Changed. optional.
```

```
	/** Alternative onConnect() method to extract details of the connection.
	 *  See: src/ble_gap.h for the details of the ble_gap_conn_desc struct.
	 */
	 void onConnect(NimBLEServer* pServer, ble_gap_conn_desc* desc) {
		 Serial.print("Client address: ");
		 Serial.println(NimBLEAddress(desc->peer_ota_addr).toString().c_str());
		 /** We can use the connection handle here to ask for different connection parameters.
		  *  Args: connection handle, min connection interval, max connection interval
		  *  latency, supervision timeout.
		  *  Units; Min/Max Intervals: 1.25 millisecond increments.
		  *  Latency: number of intervals allowed to skip.
		  *  Timeout: 10 millisecond increments, try for 5x interval time for best results.
		  */
		  pServer->updateConnParams(desc->conn_handle, 24, 48, 0, 60);
	  };
```

h2zero, 2021-05-08:
> First, advertise at 20-ms intervals for at least 30 seconds. If not discovered after 30 seconds, you may change to one of the following longer intervals: 152.5 ms, 211.25 ms, 318.75 ms, 417.5 ms, [...]

You can set the advertising interval using `NimBLEAdvertising::setMinInterval` and `NimBLEAdvertising::setMaxInterval` the parameter is in 0.625ms units, so 152.5ms / 0.625ms = 244 would be the value for the parameter.

You should simply remove this `pServer->updateConnParams(desc->conn_handle, 24, 48, 0, 60);` 
The phone will set the parameters for you.

closed 2022-01-19 by h2zero

podaen, 2022-01-19:
Oeps, forgot to respond. I diden't continue on creating an app, because the app need different premission for ios bluetooth, than an android device.

h2zero, 2022-01-20:
No worries 😃, just cleaning up a bit. Feel free to re-open this if you need.