h2zero/NimBLE-Arduino #232: Pin security not working on android

schlaegerz, 2021-05-07:
I have pin security set up

```
 NimBLEDevice::setSecurityIOCap(BLE_HS_IO_DISPLAY_ONLY); // set display output capability
 NimBLEDevice::setSecurityAuth(true, true, true);
  _pin = esp_random() % 9000 + 1000
  
  
  
uint32_t BluetoothAccessPoint::onPassKeyRequest()
{
    ESP_LOGI(TAG, "Server PassKeyRequest");
    
    show_pin_to_ui(_pin);
    return _pin;
};

void BluetoothAccessPoint::onAuthenticationComplete(ble_gap_conn_desc *desc)
{
    ESP_LOGI(TAG, "On auth complete enc: %d aut: %d bond: %d key: %d", desc->sec_state.encrypted,
             desc->sec_state.authenticated,
             desc->sec_state.bonded,
             desc->sec_state.key_size);
    if (!desc->sec_state.encrypted)
    {
        NimBLEDevice::getServer()->disconnect(desc->conn_handle);
        ESP_LOGI(TAG, "Encrypt connection failed - disconnecting client");
        return;
    }
    ESP_LOGI(TAG, "Starting BLE work!");
}

```

And on android I do correctly get the pin pop-up to show up, but when I put in the pin it always fails and desc->sec_state.authenticated is set to 0.

Oddly on ios is works just fine, so I'm not sure what could be going wrong specifically on android.

I'm using a Galaxy S9 on Android 9

 Are there any ideas?

schlaegerz, 2021-05-07:
This is with the basic Android Bluetooth settings. Just when I try to connect to the device.

Also happens with my own app and BLE scanner app

h2zero, 2021-05-08:
I just noticed why you are having this issue. Your pin code is only 4 digit, you need 6 in BLE. iOS probably prepends 2 zeros in front of the passkey.

schlaegerz, 2021-05-10:
Oh that was it! Making me go crazy, no idea why I thought it should be 4 pins.

Thanks so much!

closed 2021-05-10 by schlaegerz