h2zero/NimBLE-Arduino #235: is NimBLE compatible with 2.0.0-alpha1 core?

societyofrobots, 2021-05-12:
summary: Using same exact code, NimBLE works for me using 1.0.6 core, but not working on alpha1.

[1.0.6 has a major security bug](https://github.com/espressif/arduino-esp32/issues/5038) which I'm not sure will be fixed, so I'm forced to move to alpha1.

NimBLE compiles using alpha, but it's not connecting. I tried both 1.2 and the latest NimBLE code updated a few days ago.

Does it work on alpha1 for anyone else? Or am I doing something dumb on my end?

This is the output I get, however my Android doesn't connect:
```
BLE device connected. (this line happens before I even enter my PIN)
Server PassKeyRequest (now it asks for PIN)
BLE not connected...now advertising. (sometimes this shows before I finish entering PIN, sometimes after)
```

My Android phone says can't connect `"...because of an incorrect PIN or passkey."`

code, where BLE_loop() is run indefinitely:
```
#include <NimBLEDevice.h>
#include <NimBLEHIDDevice.h>//for HID showing in iOS

BLEServer* pServer = NULL;
BLECharacteristic* pCharacteristic = NULL;
bool deviceConnected = false;
bool oldDeviceConnected = false;
uint32_t value = 0;

#define SERVICE_UUID           "6E400001-B5A3-F393-E0A9-E50E24DCCA9E" // UART service UUID
#define CHARACTERISTIC_UUID_RX "6E400002-B5A3-F393-E0A9-E50E24DCCA9E"
#define CHARACTERISTIC_UUID_TX "6E400003-B5A3-F393-E0A9-E50E24DCCA9E"

/**  None of these are required as they will be handled by the library with defaults. **
 **                       Remove as you see fit for your needs                        */  
class MyServerCallbacks: public BLEServerCallbacks {
    void onConnect(BLEServer* pServer) {
      deviceConnected = true;
      BLEDevice::startAdvertising();//adding this line allows for multiple simultaneous BLE connections
    };

    void onDisconnect(BLEServer* pServer) {
      deviceConnected = false;
    }
  /***************** New - Security handled here ********************
  ****** Note: these are the same return values as defaults ********/
    uint32_t onPassKeyRequest(){
      Serial.println("Server PassKeyRequest");
      return passkey; 
    }
    
    void onPassKeyNotify(uint32_t passkey) {
      Serial.println("Passkey Notify");
    }

    bool onConfirmPIN(uint32_t passkey){
      Serial.print("The passkey YES/NO number: ");Serial.println(passkey);
      return true; 
    }

    void onAuthenticationComplete(ble_gap_conn_desc* desc) {
     if(desc->sec_state.encrypted){
        if(!desc->sec_state.encrypted) {
            Serial.println("Encrypt connection failed - disconnecting");
            /** Find the client with the connection handle provided in desc */
            NimBLEDevice::getClientByID(desc->conn_handle)->disconnect();
            vTaskDelay(5000);//bruteforce protection, 5s
            return;
          }
       }
    }
  /*******************************************************************/
};

class MyCallbacks: public BLECharacteristicCallbacks {
  
    void onWrite(BLECharacteristic *pCharacteristic) {
      std::string rxValue = pCharacteristic->getValue();
        
    //if data recieved
    if (rxValue.length() > 0)// && authenticated)
      {
        Serial.print("\nReceived Value: ");

        //get it's length
        for (int i = 0; i < rxValue.length(); i++) {
          Serial.print(rxValue[i]);
        }

        Serial.print(", Length: ");
        Serial.print(rxValue.length());

        // Do stuff based on command received by bluetooth

        //if it's a one letter command
        if (rxValue.length()<=4)
          {
          if (rxValue.find("F") != -1) {
            MOS1=1;
          }
          else if (rxValue.find("G") != -1) {
            MOS1=0;
            M1_off();
          }
          }
      }
    }
};

void bluetooth_setup() {

  //read BLE passkey from EEPROM and use
  passkey = atoi(bleParamValue);//v2.0 atoi converts array to int

  NimBLEDevice::init(iotWebConf.getThingName());

  //"minimum power level is ESP_PWR_LVL_N0, and maximum power level is ESP_PWR_LVL_P9"
  NimBLEDevice::setPower(ESP_PWR_LVL_P9);
  NimBLEDevice::setSecurityAuth(true, true, true);
  NimBLEDevice::setSecurityPasskey(passkey);
  NimBLEDevice::setSecurityIOCap(BLE_HS_IO_DISPLAY_ONLY);
  
  // Create the BLE Server
  pServer = BLEDevice::createServer();
  pServer->setCallbacks(new MyServerCallbacks());

  // Create the BLE Service
  BLEService *pService = pServer->createService(SERVICE_UUID);

  // Create a BLE Characteristic
  pCharacteristic = pService->createCharacteristic(
                      CHARACTERISTIC_UUID_TX,
                      NIMBLE_PROPERTY::NOTIFY
                    );

  BLECharacteristic *pCharacteristic = pService->createCharacteristic(
                                         CHARACTERISTIC_UUID_RX,
                                          NIMBLE_PROPERTY::WRITE |
                                          NIMBLE_PROPERTY::WRITE_ENC
                                       );
  
  pCharacteristic->setCallbacks(new MyCallbacks());
  
  // Start the service
  pService->start();

  // Start advertising
  BLEAdvertising *pAdvertising = BLEDevice::getAdvertising();
  pAdvertising->setAppearance(384);

  pAdvertising->addServiceUUID((uint16_t)0x1101);//serial port; this works with iOS, but not Galaxy
  pAdvertising->setManufacturerData("SA Tech");
  pAdvertising->setScanResponse(true); // must be true or else BLE name gets truncated
  /** Note, this could be left out as that is the default value */
  pAdvertising->setMinPreferred(0x0);  // set value to 0x00 to not advertise this parameter

  BLEDevice::startAdvertising();
  Serial.println("Waiting for a client connection to notify...");
}

void BLE_loop() {
    // disconnecting
    if (!deviceConnected && oldDeviceConnected) {
        delay(500); // give the bluetooth stack the chance to get things ready
        pServer->startAdvertising(); // restart advertising
        Serial.println("start advertising");
        oldDeviceConnected = deviceConnected;
    }
    // connecting
    if (deviceConnected && !oldDeviceConnected) {
        // do stuff here on connecting
        Serial.println("BLE device connected.");
        oldDeviceConnected = deviceConnected;
    }
}
```

h2zero, 2021-05-12:
I have not done any testing with the latest master of arduino but I suspect the fix may already be in the nimble-1.3.0 branch. Please try that and let me know, I will also do some testing when I get some time.

societyofrobots, 2021-05-12:
I downloaded the [1.3 branch](https://github.com/h2zero/NimBLE-Arduino/tree/nimble/1.3.0) and overwrote the 1.2 folder files.

Gives a ton of compile errors. =(

```
In file included from C:\Users\X\Documents\Arduino\libraries\WebSockets\src\WebSockets.cpp:45:
C:\Users\X\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.0-alpha1/tools/sdk/esp32/include/mbedtls/port/include/esp32/sha.h:20:2: warning: #warning esp32/sha.h is deprecated, please use sha_parallel_engine.h instead [-Wcpp]
 #warning esp32/sha.h is deprecated, please use sha_parallel_engine.h instead
  ^~~~~~~
C:\Users\X\Documents\Arduino\libraries\NimBLE-Arduino\src\nimble\host\src\ble_hs_dbg.c: In function 'ble_hs_dbg_le_event_disp':
C:\Users\X\Documents\Arduino\libraries\NimBLE-Arduino\src\nimble\host\src\ble_hs_dbg.c:131:29: error: 'BLE_HCI_RD_LOC_SUPP_FEAT_RSPLEN' undeclared (first use in this function); did you mean 'BLE_HCI_OCF_LE_RD_LOC_SUPP_FEAT'?
             for (i = 0; i < BLE_HCI_RD_LOC_SUPP_FEAT_RSPLEN; ++i) {
                             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
                             BLE_HCI_OCF_LE_RD_LOC_SUPP_FEAT
C:\Users\X\Documents\Arduino\libraries\NimBLE-Arduino\src\nimble\host\src\ble_hs_dbg.c:131:29: note: each undeclared identifier is reported only once for each function it appears in
C:\Users\X\Documents\Arduino\libraries\NimBLE-Arduino\src\nimble\host\src\ble_hs_dbg.c:155:64: error: dereferencing pointer to incomplete type 'struct hci_le_subev_direct_adv_rpt'
         struct hci_le_subev_direct_adv_rpt_param *params = data->params;
                                                                ^~
C:\Users\X\Documents\Arduino\libraries\NimBLE-Arduino\src\nimble\host\src\ble_hs_dbg.c:158:66: error: dereferencing pointer to incomplete type 'struct hci_le_subev_direct_adv_rpt_param'
                 len < sizeof(*data) + data->num_reports * sizeof(*params)) {
                                                                  ^~~~~~~
C:\Users\X\Documents\Arduino\libraries\NimBLE-Arduino\src\nimble\host\src\ble_hs_dbg.c:169:20: error: invalid use of undefined type 'struct hci_le_subev_direct_adv_rpt_param'
             params += 1;
                    ^~
C:\Users\X\Documents\Arduino\libraries\NimBLE-Arduino\src\nimble\host\src\ble_hs_dbg.c:178:27: error: dereferencing pointer to incomplete type 'struct hci_le_subev_rd_loc_p256_pubkey'
         if (len != sizeof(*data)) {
                           ^~~~~
C:\Users\X\Documents\Arduino\libraries\NimBLE-Arduino\src\nimble\host\src\ble_hs_dbg.c:193:27: error: dereferencing pointer to incomplete type 'struct hci_le_subev_gen_dhkey_complete'
         if (len != sizeof(*data)) {
                           ^~~~~
C:\Users\X\Documents\Arduino\libraries\NimBLE-Arduino\src\nimble\host\src\ble_hs_dbg.c:324:27: error: dereferencing pointer to incomplete type 'struct hci_le_subev_chan_sel_alg'
         if (len != sizeof(*data)) {
                           ^~~~~
C:\Users\X\Documents\Arduino\libraries\NimBLE-Arduino\src\nimble\host\src\ble_hs_dbg.c: In function 'ble_hs_dbg_event_disp':
C:\Users\X\Documents\Arduino\libraries\NimBLE-Arduino\src\nimble\host\src\ble_hs_dbg.c:655:22: error: 'BLE_HCI_EVENT_HDR_LEN' undeclared (first use in this function); did you mean 'BLE_HCI_CREATE_CONN_LEN'?
     evdata = evbuf + BLE_HCI_EVENT_HDR_LEN;
                      ^~~~~~~~~~~~~~~~~~~~~
                      BLE_HCI_CREATE_CONN_LEN
exit status 1
Error compiling for board Adafruit ESP32 Feather.
```

h2zero, 2021-05-12:
Thanks for reporting, I guess a bit of work needs to be done with arduino 2.0. Looks like a few things missing from the config file at the very least.

societyofrobots, 2021-05-12:
Is this something the 2.0 guys need to fix? If so, would it be a long wait?

h2zero, 2021-05-12:
Nope, it's something in NimBLE that needs fixed which I will take care of shortly.

h2zero, 2021-05-12:
Just looked at the files in the log again and it looks like you have the wrong files somehow.
This is the branch you need https://github.com/h2zero/NimBLE-Arduino/tree/nimble/1.3.0

h2zero, 2021-05-13:
Just did some testing with the branch linked above against the arduino master and it appears to work fine for me. I'll do my best to get that branch into master soon.

societyofrobots, 2021-05-13:
I definitely used the right files. This time instead of over-writing them, I fully deleted 1.2 entirely and then added the 1.3 files. The compile errors then went away.

However, I'm still having the same problem, it's not connecting unless I use 1.0.6 core.

h2zero, 2021-05-13:
Glad to hear it, if you weren't using git to checkout the branch that would certainly cause the issue.

closed 2021-05-13 by societyofrobots