h2zero/NimBLE-Arduino #236: Missing "Remove characteristic"  functionality

Bascy, 2021-05-13:
We've switched recently from arduino framework BLE implementation to NimBLE and we really like the 400k it saves us!

But I am missing the possiblity to remove a characteristic from a service.

Is there a specific reason for not enabling those functions?

h2zero, 2021-05-13:
If you are asking about removing remote characteristics on the client that functionality was renamed.
https://github.com/h2zero/NimBLE-Arduino/blob/f970a1eac81c9f1bdfdb4d93038d8500b716ef20/src/NimBLERemoteService.h#L45

Bascy, 2021-05-13:
No I meant removing a characteristic from a NimBLEService

h2zero, 2021-05-13:
That server side functionality has not been brought up before and I did not see it in the original library so it wasn't ported to this.

It could be added in a similar manner as removeService, PR's are welcome 😃.

h2zero, 2021-07-18:
@Bascy I just opened a PR for this #264, if you'd like to test it I'd appreciate any feedback.

closed 2022-01-19 by h2zero