h2zero/NimBLE-Arduino #238: Calling characteristic.indicate() inside server callback makes the callback never return.

doudar, 2021-05-15:
After a day of troubleshooting some code, if you do something like:

`` void MyCallbacks::onWrite(BLECharacteristic *pCharacteristic) { ``
``std::string rxValue = pCharacteristic->getValue();``
``uint8_t returnValue[3] = {0x80, (uint8_t)rxValue[0], 0x02};``
``pCharacteristic->setValue(returnValue, 3);``
``pCharacteristic->indicate(); ``
``}``

pCharacteristic->indicate() never returns and the callback is never invoked again.

h2zero, 2021-05-15:
Yep, I found this issue a little while ago also. I have the fix prepared for it in the 2.0 branch but has to be backported and I haven't felt like dealing with the conflicts yet 😄. Sorry about that, I didn't think it would be an issue too soon since few people use indications.

doudar, 2021-05-15:
That's okay...Only a day of my life :)

h2zero, 2021-05-15:
Good thing those are cheap.

h2zero, 2021-05-16:
If it's any consolation the backport for this fix it going to take all of my Saturday night. It might be my fault technically but I'm still blaming you for using indications 😄. 

Kidding of course, it's a real pain when you get ahead of yourself in git though. PR for the fix should be up soon.

doudar, 2021-05-16:
Wow! That was quick!

Might be a bit before I can test this. In school all week (recurrent training) but I'll try to fit it in sometime.

h2zero, 2021-05-16:
Lol, code was already done, just needed to refactor a few things. 

Not a big rush, no other issues involving this yet. It's somewhat tested and works fine in the other branch but this is not so there may be a bug somewhere.

closed 2021-05-17 by h2zero