h2zero/NimBLE-Arduino #242: unable to bond after reinitializing the BLE

JaswanthBravo, 2021-05-19:
Hi,
When I trying to connect to the BLE after setting up, my  mobile will connect the esp32. After reinitializing the BLE when I try to connect to the Esp32 it shows error in security callbacks at authentication complete, like **Encrypt connection failed - disconnecting client**.

Here is my code

#include <NimBLEDevice.h>

// Core definitions (assuming you have dual-core ESP32)
static const BaseType_t pro_cpu = 0;
static const BaseType_t app_cpu = 1;

NimBLEServer *pServer = NULL;
NimBLECharacteristic * pTxCharacteristic;
NimBLECharacteristic * pRxCharacteristic;
uint8_t txValue = 0;

// UART service UUID
#define SERVICE_UUID           "6E400001-B5A3-F393-E0A9-E50E24DCCA9E"
#define CHARACTERISTIC_UUID_RX "6E400002-B5A3-F393-E0A9-E50E24DCCA9E"
#define CHARACTERISTIC_UUID_TX "6E400003-B5A3-F393-E0A9-E50E24DCCA9E"

//Variables
uint8_t n = 0;

class ServerCallbacks: public NimBLEServerCallbacks {
    void onConnect(NimBLEServer* pServer) {
      Serial.println("Device Connected");
    };

    void onDisconnect(NimBLEServer* pServer) {
      Serial.println("Device Disconnected");
      pServer->getAdvertising()->start();
    }

    bool onConfirmPIN(uint32_t pass_key) {
      return false;
    }

    uint32_t onPassKeyRequest() {
      return 000000;
    }

    void onAuthenticationComplete(ble_gap_conn_desc* desc) {
      /** Check that encryption was successful, if not we disconnect the client */
      if (!desc->sec_state.encrypted) {
        NimBLEDevice::getServer()->disconnect(desc->conn_handle);
        Serial.println("Encrypt connection failed - disconnecting client");
        //vTaskDelay(5000);//bruteforce protection, 5s
        return;
      } else {
        NimBLEDevice::startSecurity(desc->conn_handle);
      }
    };

};

class CharacteristicCallbacks: public NimBLECharacteristicCallbacks {
    void onWrite(NimBLECharacteristic *pCharacteristic) {
      std::string rxValue = pCharacteristic->getValue();

      if (rxValue.length() > 0) {
        Serial.println("*********");
        Serial.print("Received Value: ");
        for (int i = 0; i < rxValue.length(); i++)
          Serial.print(rxValue[i]);

        Serial.println();
        Serial.println("*********");
      }
    }
};

CharacteristicCallbacks* my_callback = new CharacteristicCallbacks();

void setup() {
  Serial.begin(115200);
  Serial.print("Heap before setup ");
  Serial.println(xPortGetFreeHeapSize());

  xTaskCreatePinnedToCore(bleLoop,
                          "BLE Loop",
                          1024,
                          NULL,
                          1,
                          NULL,
                          app_cpu);

  vTaskDelete(NULL);
}

void loop() {
  Serial.println("Main Loop is Active");
}

void bleLoop(void *parameters) {
  Serial.print("High water mark (words): ");
  Serial.println(uxTaskGetStackHighWaterMark(NULL));
  Serial.println("task running on core");
  Serial.println(xPortGetCoreID());
  Serial.println("With priority");
  Serial.println(uxTaskPriorityGet(NULL));

  while (1) {
    Serial.print("Heap in loop ");
    Serial.println(xPortGetFreeHeapSize());
    Serial.println(n);
    Serial.println(NimBLEDevice::getInitialized());

    if (NimBLEDevice::getInitialized() == 0) {
      xTaskCreatePinnedToCore(bleSetup,
                              "Setup BLE",
                              2300,
                              NULL,
                              1,
                              NULL,
                              pro_cpu);
      Serial.println("Ble is activating");
    }

    if (++n == 15)  {
      n = 0;
      NimBLEDevice::deinit(true);
    }

    vTaskDelay(3000 / portTICK_PERIOD_MS);
  }
}

void bleSetup(void *parameters) {

  Serial.println("Starting NimBLE Server");
  // Create the BLE Device
  NimBLEDevice::init("HomeControl");
  NimBLEDevice::setPower(ESP_PWR_LVL_P9);

  //Create the BLE Security
  NimBLEDevice::setSecurityIOCap(BLE_HS_IO_DISPLAY_ONLY);
  NimBLEDevice::setSecurityPasskey(785214);
  NimBLEDevice::setSecurityAuth(true, true, true);
  NimBLEDevice::setSecurityInitKey(BLE_SM_PAIR_KEY_DIST_ENC | BLE_SM_PAIR_KEY_DIST_ID);
  NimBLEDevice::setSecurityRespKey(BLE_SM_PAIR_KEY_DIST_ENC | BLE_SM_PAIR_KEY_DIST_ID);

  // Create the BLE Server
  pServer = NimBLEDevice::createServer();
  pServer->setCallbacks(new ServerCallbacks(), true);

  // Create the BLE Service
  NimBLEService *pService = pServer->createService(SERVICE_UUID);

  //Create a BLE Characteristic
  pTxCharacteristic = pService->createCharacteristic(
                        CHARACTERISTIC_UUID_TX,
                        NIMBLE_PROPERTY::NOTIFY );
  pTxCharacteristic->setValue("Home Control Device is Connected");

  pRxCharacteristic = pService->createCharacteristic(
                        CHARACTERISTIC_UUID_RX,
                        NIMBLE_PROPERTY::WRITE |
                        NIMBLE_PROPERTY::WRITE_ENC |
                        NIMBLE_PROPERTY::WRITE_AUTHEN );
  pRxCharacteristic->setCallbacks(my_callback);

  // Start the service
  pService->start();

  // Start advertising
  pServer->getAdvertising()->start();
  Serial.println("Waiting a client connection to notify...");

  // Print out remaining stack memory (words)
  Serial.print("High water mark (words): ");
  Serial.println(uxTaskGetStackHighWaterMark(NULL));
  Serial.println("task running on core");
  Serial.println(xPortGetCoreID());
  Serial.println("With priority");
  Serial.println(uxTaskPriorityGet(NULL));

  vTaskDelete(NULL);
}

and the Serial output is.
18:29:02.073 -> Heap before setup 238964
18:29:02.073 -> High water mark (words): 660
18:29:02.073 -> task running on core
18:29:02.073 -> 1
18:29:02.073 -> With priority
18:29:02.073 -> 1
18:29:02.073 -> Heap in loop 237552
18:29:02.073 -> 0
18:29:02.073 -> 0
18:29:02.073 -> Ble is activating
18:29:02.073 -> Starting NimBLE Server
18:29:02.354 -> Waiting a client connection to notify...
18:29:02.354 -> High water mark (words): 676
18:29:02.354 -> task running on core
18:29:02.354 -> 0
18:29:02.354 -> With priority
18:29:02.354 -> 1
18:29:05.088 -> Heap in loop 229032
18:29:05.088 -> 1
18:29:05.088 -> 1
18:29:06.963 -> Device Connected
18:29:08.088 -> Heap in loop 229016
18:29:08.088 -> 2
18:29:08.088 -> 1
18:29:11.056 -> Heap in loop 229016
18:29:11.056 -> 3
18:29:11.056 -> 1
18:29:14.055 -> Heap in loop 229016
18:29:14.055 -> 4
18:29:14.055 -> 1
18:29:17.085 -> Heap in loop 229016
18:29:17.085 -> 5
18:29:17.085 -> 1
18:29:20.054 -> Heap in loop 229016
18:29:20.054 -> 6
18:29:20.054 -> 1
18:29:21.694 -> Device Disconnected
18:29:23.069 -> Heap in loop 229016
18:29:23.069 -> 7
18:29:23.069 -> 1
18:29:26.084 -> Heap in loop 229016
18:29:26.084 -> 8
18:29:26.084 -> 1
18:29:29.083 -> Heap in loop 229016
18:29:29.083 -> 9
18:29:29.083 -> 1
18:29:32.071 -> Heap in loop 229016
18:29:32.071 -> 10
18:29:32.071 -> 1
18:29:35.071 -> Heap in loop 229016
18:29:35.071 -> 11
18:29:35.071 -> 1
18:29:38.070 -> Heap in loop 229016
18:29:38.070 -> 12
18:29:38.070 -> 1
18:29:41.069 -> Heap in loop 229016
18:29:41.069 -> 13
18:29:41.069 -> 1
18:29:44.068 -> Heap in loop 229016
18:29:44.068 -> 14
18:29:44.068 -> 1
18:29:47.083 -> Heap in loop 260936
18:29:47.083 -> 0
18:29:47.083 -> 0
18:29:47.083 -> Ble is activating
18:29:47.083 -> Starting NimBLE Server
18:29:47.270 -> failed to configure restored IRK
18:29:47.270 -> Waiting a client connection to notify...
18:29:47.270 -> High water mark (words): 556
18:29:47.270 -> task running on core
18:29:47.318 -> 0
18:29:47.318 -> With priority
18:29:47.318 -> 1
18:29:50.067 -> Heap in loop 228684
18:29:50.067 -> 1
18:29:50.067 -> 1
18:29:53.066 -> Heap in loop 228684
18:29:53.066 -> 2
18:29:53.066 -> 1
18:29:53.394 -> Device Connected
18:29:56.066 -> Heap in loop 228664
18:29:56.066 -> 3
18:29:56.066 -> 1
18:29:59.065 -> Heap in loop 228664
18:29:59.065 -> 4
18:29:59.065 -> 1
18:30:02.064 -> Heap in loop 228664
18:30:02.064 -> 5
18:30:02.064 -> 1
18:30:03.376 -> Encrypt connection failed - disconnecting client
18:30:03.439 -> Device Disconnected
18:30:05.094 -> Heap in loop 228664
18:30:05.094 -> 6
18:30:05.094 -> 1
18:30:08.078 -> Heap in loop 228664
18:30:08.078 -> 7
18:30:08.078 -> 1
18:30:11.077 -> Heap in loop 228664
18:30:11.077 -> 8
18:30:11.077 -> 1

please help me how to resolve this issue.

h2zero, 2021-05-19:
@JaswanthBravo Thanks for reporting this. I believe the problem could be caused by an issue upstream, I will investigate.

h2zero, 2021-05-23:
@JaswanthBravo I am unable to reproduce this problem with the current master of this repo. Could you try updating if not already and let me know if you still have this issue?

JaswanthBravo, 2021-05-24:
I am using the latest repo still having the issues. This issue was occurred while bonding to the esp32 after BLE is reinitialized.
18:30:03.376 -> Encrypt connection failed - disconnecting client
18:30:03.439 -> Device Disconnected.

But if the device is bonded previously it will connect to the esp32 without error.

h2zero, 2021-05-24:
Might it be an issue that you've bonded more than the max number of devices (default is 3)? When that occurs the oldest bond is deleted and the pairing might fail at that point. If that's not the case I will keep looking.

JaswanthBravo, 2021-05-24:
At first BLE initialized I bonded my android phone and unbonded that device. after reinitialized the BLE, I try to bond the same mobile device but it is not bonding and through this error.
18:30:03.376 -> Encrypt connection failed - disconnecting client
18:30:03.439 -> Device Disconnected.

I suspect may there is a issue with the IRK after BLE is reinitialized, because it show error like. 
18:29:47.083 -> Starting NimBLE Server
18:29:47.270 -> failed to configure restored IRK

h2zero, 2021-05-24:
Thank you for the further details, this is an upstream bug that's been reported a while ago and still is not resolved. If I can reproduce I will see if I can get it fixed.

h2zero, 2021-05-24:
Update: I am able to reproduce. Hopefully I can get it fixed today.

h2zero, 2021-05-25:
@JaswanthBravo I just pushed commit e9bda73 that should resolve this issue. Please test when you can and report back.

JaswanthBravo, 2021-05-25:
After your update now the device connects after BLE is reinitialized. Thank you

h2zero, 2021-05-25:
Great! Thanks for reporting.

closed 2021-05-25 by JaswanthBravo