h2zero/NimBLE-Arduino #244: Issue with NimBLERemoteCharacteristic::writeValue (for std::string) - strings with nulls get truncated

boozer2, 2021-05-21:
In the overload for NimBLERemoteCharacteristic::writeValue that takes a std::string as an argument, if a string is passed that contains nulls, it gets truncated. The reason is it's using strlen(newValue.c_str()) to determine the length, but that will stop at the first null. A safer way to do this (and I tested) is to instead use newValue.length() which is made for std::string and uses the actual string length instead of looking for a null termination in a C string.

From NimBLERemoteCharacteristic.cpp:
Original:
```
bool NimBLERemoteCharacteristic::writeValue(const std::string &newValue, bool response) {
    return writeValue((uint8_t*)newValue.c_str(), strlen(newValue.c_str()), response);
} // writeValue
```

Corrected:
```
bool NimBLERemoteCharacteristic::writeValue(const std::string &newValue, bool response) {
    return writeValue((uint8_t*)newValue.c_str(), newValue.length(), response);
} // writeValue

```

Here's a quck test I did to demonstrate the issue:
```
const std::string testStr = { 0x41, 0x42, 0x43, 0x00, 0x44, 0x45 };
const byte testStrB[6] = { 0x41, 0x42, 0x43, 0x00, 0x44, 0x45 };

void hexPrint(const uint8_t* s, size_t len) {
  for (int n = 0; n < len; n++) {
    Serial.printf("%02X ", s[n]);
  }
  Serial.println();
}

void hexPrint2(const std::string &s) {
  hexPrint((uint8_t*)s.c_str(), strlen(s.c_str()));
}

void hexPrint3(const std::string &s) {
  hexPrint((uint8_t*)s.c_str(), s.length());
}

void setup() {
  Serial.begin(115200);
  delay(500);
  Serial.print("uint8_t passed directly, with length specified: "); //No issue
  hexPrint(testStrB, 6);
  Serial.print("string converted to c_str cast to uint8_t*, with length specified: "); //No issue
  hexPrint((uint8_t*)testStr.c_str(), 6);
  Serial.print("string sent, converted in helper function using c_str and cast, and using strlen on c_str: "); //Truncated
  hexPrint2(testStr);
  Serial.print("string sent, converted in helper function using c_str and cast, and using string::length: "); //No issue
  hexPrint3(testStr);
}

void loop() {
}
```

h2zero, 2021-05-21:
Thanks, no idea how that went unnoticed for this long, good catch. Would you like to PR that change?

boozer2, 2021-05-21:
Sure - I've never done one before but doesn't seem too hard - I'll do that tonight. Thanks!

chegewara, 2021-05-21:
> Thanks, no idea how that went unnoticed for this long, good catch. Would you like to PR that change?

Just guessing, but i think that most people is using std::string as null terminated strings, just like C strings and uint8_t* function in case when you are sending bytes.

boozer2, 2021-05-21:
I submitted a PR - thanks!

h2zero, 2021-05-21:
> Just guessing, but i think that most people is using std::string as null terminated strings, just like C strings and uint8_t* function in case when you are sending bytes.

Probably, but there have been a few places where this behavior was noted and never even thought about that code haha.

closed 2021-05-22 by boozer2