h2zero/NimBLE-Arduino #246: bonding fails between ESP32 and OMRON Blood-Pressure-Device

Guesel, 2021-05-22:
Hallo,
I try to connect an ESP32 with the OMRON EVOLV device. The EVOLV is discoverable, all services and characteristics can be read.
But for getting the measurement results it needs the devices to bond. The security params at EVOLV are:
IO Capability:  NoInput, NoOutput; OOB = no; Encryption-Key-Size = 16; IRK- and LTK-Key = true.
"JustWorks" should be run as defined for default. 
(see attached wireshark-protocoll)

That's why my code in setup looks like:

void setup (){

    Serial.begin(-------
    NimBLEDevice::init("");

    /** Set the IO capabilities of the device, each option will trigger a different pairing method.
     *  BLE_HS_IO_KEYBOARD_ONLY    - Passkey pairing
     *  BLE_HS_IO_DISPLAY_YESNO   - Numeric comparison pairing
     *  BLE_HS_IO_NO_INPUT_OUTPUT - DEFAULT setting - just works pairing
     */
 
    NimBLEDevice::setSecurityIOCap(BLE_HS_IO_NO_INPUT_OUTPUT);
    NimBLEDevice::setSecurityAuth(true, false, true);
    NimBLEDevice::setSecurityInitKey(BLE_SM_PAIR_KEY_DIST_ENC| BLE_SM_PAIR_KEY_DIST_ID);
    NimBLEDevice::setSecurityRespKey(BLE_SM_PAIR_KEY_DIST_ENC| BLE_SM_PAIR_KEY_DIST_ID);
-----
-----

From my understanding the devices should bond automatically with these parameters or send a pairing-request-message.
Nothing happend.

The attached wireshark shows the bonding between nRF-Connect (Android) and the OMRON-Device.  As you can see the devices bonding and then starts the discovery starts .

My questions:
- is the setup-code right?
- is anything function missing in nimble?

May i ask for help?

Thx. 
Güsel

[image]

h2zero, 2021-05-22:
Hi @Guesel, could you provide a debug log preferably with this https://github.com/h2zero/NimBLE-Arduino/blob/40c98b25835c61c3ea615b130468b7c9fc5289ee/src/nimconfig.h#L22 enabled?

If the device has a secured characteristic and you attempt to read from it the pairing should trigger automatically, without a log there is no way to know if this is happening.

Guesel, 2021-05-26:
Thanks for the answer. The problem could be solved with a general reset (removing the batteries). The coupling process then starts reliably and automatically in the background. The issue could be closed. Thank you for your work.

closed 2021-05-26 by Guesel

Darshansgit, 2023-07-01:
@Guesel hehllo, I have a similar personal project at hand where im trying to connect a esp32 device with a omron silver bp cuff, were you able to do it and fetch data successfully? If yes could you provide me with some directives, thank you

Guesel, 2023-07-02:
Hello there, I own a working solution for an OMRON Evolv intelli sense blood pressure monitor. If you send me your private e-mail address, you will receive the source code as a zip file. The code is not optimized, but runs fine on an ESP32. The function is: reading the OMRON measurement value memory for up to 100 measurements including error information and sending the values to a database via WLAN.
Greetings Güsel

Darshansgit, 2023-07-02:
Hello @Guesel thank you for your response, my private email is darshansubedi31@gmail.com,I have a omron silver bp cuff, I hope ill be able to implement your solution, Thank you for your time.

juanimoni, 2023-07-08:
Hello Guesel, would you mind sending me also your code ? I have an Omron X7 and would also like to read its data.
(will be for sure easier to learn from your code than doing it from scratch)
juan.ignacio.cid@gmail.com
Thank you in advance and regards

Darshansgit, 2023-07-12:
@Guesel could you provide me some help please? Really need it

Jawadhur23, 2023-11-17:
@Guesel I am also working in similar project I have been working to transfer the OMRON BP7120T to ESP32 can you provide me a code for that I would like to check 
Thanks in advance

gabrielolimpio, 2024-10-16:
Could you send it to me too? @Guesel