h2zero/NimBLE-Arduino #247: Timeout Connecting to BLEServer on 2.0.0-alpha1

SlightlyStoopid, 2021-05-23:
Hi h2zero,

Environment:
esp32: 2.0.0-alpha1
board: Huzzah32 Feather

I'm was running this with Xtensa @ 2.5.0 and Arduino-Espressif32 @ 1.0.6 and it works great (many thanks btw),

My .ini for updating
```
platform_packages =
	toolchain-xtensa32@~2.80400.0
        framework-arduinoespressif32@https://github.com/espressif/arduino-esp32.git#2.0.0-alpha1
```

This appears to make connecting to the BLEServer impossible, with every attempt resulting in :
:warning:D NimBLEServer: "BLE_GAP_EVENT_DISCONNECT 520":warning:
 
Timeout debug output logs using the : NimBLE_Server.ino example
```
I NimBLEDevice: "BLE Host Task Started"
ble_hs_hci_cmd_send: ogf=0x03 ocf=0x0003 len=0
0x03 0x0c 0x00 
Command complete: cmd_pkts=5 ogf=0x3 ocf=0x3 status=0 
ble_hs_hci_cmd_send: ogf=0x04 ocf=0x0001 len=0
0x01 0x10 0x00 
Command complete: cmd_pkts=5 ogf=0x4 ocf=0x1 status=0 hci_ver=8 hci_rev=782 lmp_ver=8 mfrg=96 lmp_subver=782
ble_hs_hci_cmd_send: ogf=0x04 ocf=0x0003 len=0
0x03 0x10 0x00 
Command complete: cmd_pkts=5 ogf=0x4 ocf=0x3 status=0 supp_feat=0x877bffdbfecdeebf
ble_hs_hci_cmd_send: ogf=0x03 ocf=0x0001 len=8
0x01 0x0c 0x08 0x90 0x80 0x00 0x02 0x00 0x80 0x00 0x20 
Command complete: cmd_pkts=5 ogf=0x3 ocf=0x1 status=0 
ble_hs_hci_cmd_send: ogf=0x03 ocf=0x0063 len=8
0x63 0x0c 0x08 0x00 0x00 0x80 0x00 0x00 0x00 0x00 0x00 
Command complete: cmd_pkts=5 ogf=0x3 ocf=0x63 status=0 
ble_hs_hci_cmd_send: ogf=0x08 ocf=0x0001 len=8
0x01 0x20 0x08 0x7f 0x06 0x00 0x00 0x00 0x00 0x00 0x00 
Command complete: cmd_pkts=5 ogf=0x8 ocf=0x1 status=0 
ble_hs_hci_cmd_send: ogf=0x08 ocf=0x0002 len=0
0x02 0x20 0x00 
Command complete: cmd_pkts=5 ogf=0x8 ocf=0x2 status=0 
ble_hs_hci_cmd_send: ogf=0x08 ocf=0x0003 len=0
0x03 0x20 0x00 
Command complete: cmd_pkts=5 ogf=0x8 ocf=0x3 status=0 
ble_hs_hci_cmd_send: ogf=0x04 ocf=0x0009 len=0
0x09 0x10 0x00 
Command complete: cmd_pkts=5 ogf=0x4 ocf=0x9 status=0 bd_addr=a4:cf:12:32:e3:aa
ble_hs_hci_cmd_send: ogf=0x08 ocf=0x0018 len=0
0x18 0x20 0x00 
Command complete: cmd_pkts=5 ogf=0x8 ocf=0x18 status=0 rand=0x4275d98fd5889c70
ble_hs_hci_cmd_send: ogf=0x08 ocf=0x0018 len=0
0x18 0x20 0x00 
Command complete: cmd_pkts=5 ogf=0x8 ocf=0x18 status=0 rand=0x69d69514cca00a77
Device added to RL, Resolving list count = 1
ble_hs_hci_cmd_send: ogf=0x03 ocf=0x0031 len=1
0x31 0x0c 0x01 0x01 
Command complete: cmd_pkts=5 ogf=0x3 ocf=0x31 status=0 
ble_hs_hci_cmd_send: ogf=0x03 ocf=0x0033 len=7
0x33 0x0c 0x07 0xff 0x00 0x00 0x0c 0x00 0x00 0x00 
Command complete: cmd_pkts=5 ogf=0x3 ocf=0x33 status=0 
looking up peer sec; 
I NimBLEDevice: "NimBle host synced."
D NimBLEDevice: ">> setPower: 7 (type: 11)"
D NimBLEDevice: "<< setPower"
D NimBLEDevice: "Setting bonding: 0, mitm: 0, sc: 1"
D NimBLEServer: ">> createService - 0xdead"
D NimBLEServer: "<< createService"
D NimBLECharacteristic: ">> setValue: length=7, data=42757267657200, characteristic UUID=0xbeef"
D NimBLECharacteristic: "<< setValue"
[... 210 lines cut ...]
Number of Completed Packets: num_handles=1
handle:0 pkts:1
ble_hs_hci_evt_acl_process(): conn_handle=0 pb=2 len=9 data=0x05 0x00 0x04 0x00 0x04 0x13 0x00 0xff 0xff 
rxed att command: find info req; conn=0 start_handle=0x0013 end_handle=0xffff
txed att command: error rsp; conn=0 req_op=4 handle=0x0013 error_code=10
host tx hci data; handle=0 length=9
ble_hs_hci_acl_tx(): 0x00 0x00 0x09 0x00 0x05 0x00 0x04 0x00 0x01 0x04 0x13 0x00 0x0a 
Number of Completed Packets: num_handles=1
handle:0 pkts:1
D NimBLEUUID: "Comparing UUIDs; type 16 to 16; UUID 0xdead to 0xbaad"
D NimBLEUUID: "Comparing UUIDs; type 16 to 16; UUID 0xbaad to 0xbaad"
D NimBLEUUID: "Comparing UUIDs; type 16 to 16; UUID 0xf00d to 0xf00d"
D NimBLECharacteristic: ">> notify: length: 6"
D NimBLECharacteristic: "<< notify: No clients subscribed."
Disconnection Complete: status=0 handle=0 reason=8
D NimBLEServer: ">> handleGapEvent: BLE_GAP_EVENT_SUBSCRIBE"
I NimBLEServer: "subscribe event; attr_handle=8, subscribed: false"
D NimBLEServer: ">> handleGapEvent: BLE_GAP_EVENT_DISCONNECT"
Client disconnected - start advertising
D NimBLEAdvertising: ">> Advertising start: customAdvData: 0, customScanResponseData: 0"
W NimBLEAdvertising: "Advertising already active"
D NimBLEServerCallbacks: "onDisconnect(): Default"
D NimBLEAdvertising: ">> Advertising start: customAdvData: 0, customScanResponseData: 0"
W NimBLEAdvertising: "Advertising already active"
```

h2zero, 2021-05-23:
Hi @SlightlyStoopid, could you please try it with the master branch of NimBLE-Arduino. Some recent updates were made to work with the arduino core 2.0.0

SlightlyStoopid, 2021-05-23:
Hi h2zero, thanks for the quick reply, looks like resolved my connection issue thanks again.

closed 2021-05-23 by SlightlyStoopid