h2zero/NimBLE-Arduino #248: Add data pointer exposure in NimBLECharacteristic

tibotix, 2021-05-23:
Hey, 
I don't know if I understood this correctly, but would be better in aspects of performance to give the ability to get the underlying data pointer of `m_value`, instead of copying the whole string like the `getValue()` function does?
I am working with BLExUART and dealing with thousands of bytes in the `m_value` buffer, so I think it would be a significant difference when copying just the pointer instead of the whole string.

This is what I thought about in specific:
```
uint8_t*          getValueRaw(){
        portENTER_CRITICAL(&m_valMux);
        uint8_t* d = (uint8_t*)m_value.data();
        portEXIT_CRITICAL(&m_valMux);
        return d;
    }
```

Such a function also exists in the original BLE Arduino implementation: https://github.com/nkolban/ESP32_BLE_Arduino/blob/master/src/BLEValue.cpp

Thanks for your help in advance.

h2zero, 2021-05-23:
It was decided a while back to remove that function because providing a pointer to internal volatile data would be problematic if the app was reading it and at the same time an event triggers and it gets updated. In your example that would only provide safety while copying the pointer, you would need to also read the data from the pointer inside a critical section in your app code.

tibotix, 2021-05-25:
Hey @h2zero, thanks for your reply. Your point makes sense. I didn't think about that.

closed 2021-05-25 by tibotix

h2zero, 2021-05-26:
No worries 😃. 

For what it's worth I should also say that the data can only be sent in 512 bytes max per connection interval. At the lowest interval of 7.5ms I don't believe the data copy will slow anything down anyway.