h2zero/NimBLE-Arduino #251: Maximum 3 clients

johntaves, 2021-06-07:
I commented in
#define CONFIG_BT_NIMBLE_MAX_CONNECTIONS 8
in nimconfig.h, then cleaned my project and built, but no change in behavior for the 4th attempted connection. It looked like many NimBLE*.o files were created on that build.

I reviewed issue #99. It seems that it should work with just the nimconfig.h change, but maybe there's an issue with platform io.

I am using platform io, and I apologize for having no clue how platform io manages the various included libraries.

h2zero, 2021-06-07:
Try adding -D CONFIG_BT_NIMBLE_MAX_CONNECTIONS 8 to your PlatformIO.ini and delete the .pio folder.

johntaves, 2021-06-08:
build_flags = -DCONFIG_BT_NIMBLE_MAX_CONNECTIONS=8

I added this to the platformio.ini file, deleted the .pio folder and it works. I am not sure, but maybe clean and build without deleting the .pio folder would also work.

Thanks for your help.

closed 2022-01-19 by h2zero