h2zero/NimBLE-Arduino #252: possible memory leak issue

societyofrobots, 2021-06-09:
I'm trying locate the source of a bug (which may or may not be caused by NimBLE).

I have my code running for about a week, accessing it occasionally via BLE with my phone, and then one day it just says "unable to connect." I can still see the BLE in my list of available BLE, but phone refuses to connect. So I power reset, and after a week the problem happens again.

It sounds like a memory leak maybe. I'm going to set up a few units to run with a USB datalogger on it.

What recommendation would you have that would help me debug this?

h2zero, 2021-06-09:
Depending on the arduino core version you're using there is a bug that can cause this. The controller thinks you're already connected and refuses the new connection. Beyond that I cannot speculate without a log of the issue.

societyofrobots, 2021-06-10:
Interesting, that could just be it.

Do you have a link to that bug? I can probably design a workaround.

It'll probably take a week to experience the bug again, but I'll get a log of it asap.

h2zero, 2021-06-10:
>Do you have a link to that bug? I can probably design a workaround.

I don't have any specifics for this, just observations and conversations in other issue that resulted in the modifications to the client code that added a timer to the disconnect call.

I look forward to seeing any logs you have.

closed 2022-01-19 by h2zero