h2zero/NimBLE-Arduino #254: What is the correct way to stop BLE?

sivar2311, 2021-06-15:
I have created a BLE server with some characteristics and the corresponding descriptors. After I have received the data I need, I want to stop the whole BLE server. 
What is the correct way to do this? 
Is a simple call to `NimBLE::deinit()` sufficient?
I always get the error message 
"E NimBLEAdvertising: "Error enabling advertising; rc=30, "
What am I doing wrong, or do I have to do something else?

Thanks in advance!

h2zero, 2021-06-15:
`NimBLE::deinit()` is correct. You will need to call `NimBLE::init()` again later if you want to restart it. That error message can probably be ignored as it's likely called by the disconnection event trying to restart advertising. You can also get rid of it by calling `NimBLEServer::advertiseOnDisconnect(false); before deinit.

sivar2311, 2021-06-15:
>You will need to call `NimBLE::init()` again later if you want to restart it.

That's the way i want to have it :)

>You can also get rid of it by calling `NimBLEServer::advertiseOnDisconnect(false);` before deinit.

Tested and it works ! Perfect! 
Thank you very much!

closed 2021-06-15 by sivar2311