h2zero/NimBLE-Arduino #255: Why include FreeRTOS source code copies in NimBLE library

Bascy, 2021-06-21:
I just noticed that NimBLE has its own copied of FreeRTOS.h and .cpp included, whcih are not in sync with the version of FreeRTOS we used before including NimBLE into our project. 

It also changes handling of logging of standard FreeRTOS functionality to use the NIMBLE_LOG methods.

Is there a specific reason for copying this code into the library, besides changing the logging structure?

Is there an easy way to exclude these files from being compiled when someone prefers using the original and updated ones?

h2zero, 2021-06-21:
Hi @Bascy it's just been here historically since this started because they were used by this library originally. 

That is no longer the case but I have left them in up to now to not break anything for anyone that was using them.

These will simply be removed in the future. You can just delete the files for your use most likely, though it may cause conflict if the original BLE library gets linked in.

Bascy, 2021-06-21:
Removing them is indeed a working work-around .. but that won't work on my CI/CD pipeline as it will always retrieve the latest version of NimBLE from git. Also when I refresh libraries locally, I have to remember to manually delete the files ....

I'll have to figure out a way to remove the files on every build ...

Just out of curiousity .. as this lib is specifically for the esp32 architecture .. aren't the FreeRTOS files always there already?

h2zero, 2021-06-21:
For now I would suggest forking this repo and pointing your sources to your fork with the files removed. 

Originally I was going to modify those to get rid of much of the excess in them but it wasn't a pressing concern. In the next release I will consider their removal.

h2zero, 2021-06-21:
Just in case you're interested the 2.0.0 branch has a lighter weight version of those files, also renamed the header.

Bascy, 2021-06-22:
How stable is the 2.0.0 branch?

h2zero, 2021-06-22:
It's quite stable in my testing so far, but you may need to make some code changes in your application. I still need to do the documentation on it and more commits are coming.

sivar2311, 2021-06-22:
Sorry for stealing this issues but you made me curious :)
Can you already name some changes in the 2.0.0? And what code changes are needed for example?

h2zero, 2021-06-22:
I will have to go through my notes on it but there isn't much that I can think of right now. I've been in testing mode on it for a while and don't remember all of it. Most of the changes will show up on compilation if it's used by your app code. I have breaking changes in the works for future commits on it though.

sivar2311, 2021-06-22:
Thanks for the info. Looks like I'll have to be patient. :grin:
The idea with just testing and looking for compilation errors was of course "too obvious" as I would have thought of it... :joy:

Bascy, 2021-06-22:
I'l go for the fork solution :)

closed 2021-06-22 by Bascy