h2zero/NimBLE-Arduino #256: The old problems are back

Guesel, 2021-06-27:
regarding to the closed issue #246 : bonding fails between ESP32 and OMRON Blood-Pressure-Device

After updating the Nimble-lib to the last version via zip-file the devices does not bonding anymore. The trick with removing the battery no longer works either (general reset).  I've written a little test program. It has no error handling. It establishes a connection between the devices and reads in the blood pressure service. First it outputs the characteristic for the measurement features. Then the CCD is set for the indication process and an indication is released. The parameters for bonding are set in the setup.
The bonding process should start automatically with the attempt to write to the CCD. The attempt to write is unsuccessful because device security has not been activated. Maybe you can see a reason from the debug info. A device error cannot exist because nrfConnect and the native manufacturer application work (see the pictures in issue #246). 

My questions are:

- is the setup ok?
- is a function missing in the code or in the wrong place?

For any hint would be appreciated!

Güsel

The debug-output:
[problem-debug.txt](https://github.com/h2zero/NimBLE-Arduino/files/6721681/problem-debug.txt)

The test-source:
[ESP-Fix.txt](https://github.com/h2zero/NimBLE-Arduino/files/6721683/ESP-Fix.txt)

h2zero, 2021-06-27:
@Guesel I do not see anywhere in the logs where the server is requiring security to read/write to it's characteristics. If the device doesn't request it then it will not automatically happen. What you can do to force it is call NimBLEDevice::startSecurity() after connecting. 

I also noticed this:
```
if(pChr->canIndicate()) {
    
    uint8_t s[2] = {0x02,0x00};
    uint8_t *wert = &s[0];
    pDsc->writeValue(wert, 2, false);							// Set CCD = 0x0200 for INDICATION

    Serial.println("\n-> ESP-Fix: Set CCD to 0x0200 for INDICATION");
    if(!pChr->subscribe(true, notifyCB)) {pClient->disconnect();}	// subscribe
}
```
I'm not sure why you are doing this but I don't think it will do what you expect. The first part will set indications on the server but the subscribe call immediately overrides that by subscribing to notifications instead. You could remove all of that and just call subscribe() with `false` as the first parameter and it will set indications on the server instead of notifications. https://h2zero.github.io/esp-nimble-cpp/class_nim_b_l_e_remote_characteristic.html#aa08b2f62376568e1fc833e4ff91e8aa7

Guesel, 2021-06-28:
@h2zero  thank you for the time to analyze my problem! 
Apparently the decive does not request the connection security when a ccd is read or written. Surprise!  NimBLE Device::start Security() did the trick!
The second part was my fault, the parameter for "notify" is of course wrong. Sorry!
The device works with indication. I use NimBLERemoteCharacteristic :: subscribe with response = true and everything seems to work. No event is released without the response parameter. Is there an explanation? 
The background to the program is that blood readings are automatically transferred just in time to the ESP and to an SQL database. This should be done without launching an app on Android. It all has to be very reliable! 
All problems are solved for the moment! Let's see what's coming tomorrow!

h2zero, 2021-06-28:
Glad it's working now 😀 

> I use NimBLERemoteCharacteristic :: subscribe with response = true and everything seems to work. No event is released without the response parameter. Is there an explanation?

That is actually by the spec, the parameter will be removed from the call in the future as it should always write with response.

h2zero, 2022-04-05:
Closing as resolved.

closed 2022-04-05 by h2zero

juanimoni, 2023-08-12:
Hi @h2zero, I´m using a similar Omron device as Guesel.
Using the same "improvements" ("NimBLEDevice::start Security()" & "subscribe(false, notifyCB, true") I can see I have subscribed to Indications (reading indicator value I get 0x0200) BUT I´m not getting any data through the notifyCB Handler, as the CB Handler is never called. 
Any suggestions on how to proceed ?

The Characteristic who´s value I´m trying to read has ONLY the INDICATION property (no read or write or notify).
trying to read another Characteristic which has the READ property works fine.

Thank you in advance for any help and regards !

h2zero, 2023-08-12:
If you use nRFConnect on your phone do you receive indications? Might be the device is just not sending any data.

juanimoni, 2023-08-12:
If I use nRFConect, I have to press twice on the characteristic. (I see to the right an icon with one up and one down arrows disabled, which gets enabled after pressing once  and shows the expected value after the second press).
I see :
Data Written to descr. 2902-0000-1000-8000-00805f9b34fb, value: (0x) 00-00
"Notifications and indications disabled" sent
Data Written to descr. 2902-0000-1000-8000-00805f9b34fb, value: (0x) 02-00
"Indications enabled" sent
... two notifications... and 
Indication received from 2a35-0000-1000-8000-00805f9b34fb, value: (0x) 1E-88-00-49-00-5E-00_00-45-00-01-00-00 which is the value I was expecting...

how does NimBLE "emulate" this doble click ? The first one through subscribing to INDICATION, but the second ?
Thanks a lot for your support !

h2zero, 2023-08-12:
There should be no need to do the double action. Please enable debug logs and post them here.

juanimoni, 2023-08-12:
3. Advertised Device found :: Name: , Address: 00:5f:bf:0b:cd:19, manufacturer data: 4c0002152c948f480615449890fff8e2ea45ac7800010001c4
D NimBLEScan: >> stop()
4. Scan Ended
D NimBLEScan: << stop()
5. New client created
D NimBLEClient: >> connect(00:5f:bf:0b:cd:19)
D NimBLEClient: Got Client event 
I NimBLEClient: Connected event
D NimBLEClient: Got Client event 
I NimBLEClient: mtu update event; conn_handle=0 mtu=160
I NimBLEClient: Connection established
D NimBLEClient: >> deleteServices
D NimBLEClient: << deleteServices
E NimBLEDevice: ble_gap_security_initiate: rc=7 
D NimBLEClient: << connect()
6. Connected to :: 00:5f:bf:0b:cd:19
7. RSSI :: D NimBLEClient: >> getRssi()
-63
D NimBLEClient: >> getService: uuid: 0000180a-0000-1000-8000-00805f9b34fb
D NimBLEClient: >> retrieveServices
D NimBLEClient: Service Discovered >> status: 14 handle: -1
D NimBLEClient: << Service Discovered
D NimBLEClient: << retrieveServices
D NimBLEClient: >> retrieveServices
D NimBLEClient: Service Discovered >> status: 0 handle: 512
D NimBLERemoteService: >> NimBLERemoteService()
D NimBLERemoteService: << NimBLERemoteService(): 0x180a
D NimBLEClient: Service Discovered >> status: 14 handle: -1
D NimBLEClient: << Service Discovered
D NimBLEClient: << retrieveServices
8.0 Device Information service found :: 0x180a
D NimBLERemoteService: >> getCharacteristic: uuid: 00002a24-0000-1000-8000-00805f9b34fb
D NimBLERemoteService: >> retrieveCharacteristics() for service: 0x180a
D NimBLERemoteService: Characteristic Discovered >> status: 14 handle: -1
D NimBLERemoteService: << Characteristic Discovered
D NimBLERemoteService: << retrieveCharacteristics()
D NimBLERemoteService: >> retrieveCharacteristics() for service: 0x180a
D NimBLERemoteService: Characteristic Discovered >> status: 0 handle: 545
D NimBLERemoteCharacteristic: >> NimBLERemoteCharacteristic()
D NimBLERemoteCharacteristic: << NimBLERemoteCharacteristic(): 0x2a24
D NimBLERemoteService: Characteristic Discovered >> status: 14 handle: -1
D NimBLERemoteService: << Characteristic Discovered
D NimBLERemoteService: << retrieveCharacteristics()
8.1 Model Number Characteristic :: 0x2a24
8.1.1 Model Number can read... 
8.1.2 Model Number Value :: D NimBLERemoteCharacteristic: >> readValue(): uuid: 0x2a24, handle: 545 0x221
I NimBLERemoteCharacteristic: Read complete; status=0 conn_handle=0
D NimBLERemoteCharacteristic: Got 8 bytes
I NimBLERemoteCharacteristic: Read complete; status=14 conn_handle=0
D NimBLERemoteCharacteristic: << readValue length: 8 rc=0
X7 Smart
D NimBLEClient: >> getService: uuid: 00001810-0000-1000-8000-00805f9b34fb
D NimBLEClient: >> retrieveServices
D NimBLEClient: Service Discovered >> status: 14 handle: -1
D NimBLEClient: << Service Discovered
D NimBLEClient: << retrieveServices
D NimBLEClient: >> retrieveServices
D NimBLEClient: Service Discovered >> status: 0 handle: 2048
D NimBLERemoteService: >> NimBLERemoteService()
D NimBLERemoteService: << NimBLERemoteService(): 0x1810
D NimBLEClient: Service Discovered >> status: 14 handle: -1
D NimBLEClient: << Service Discovered
D NimBLEClient: << retrieveServices
8.0 Blood Service :: 0x1810
D NimBLERemoteService: >> getCharacteristic: uuid: 00002a35-0000-1000-8000-00805f9b34fb
D NimBLERemoteService: >> retrieveCharacteristics() for service: 0x1810
D NimBLERemoteService: Characteristic Discovered >> status: 14 handle: -1
D NimBLERemoteService: << Characteristic Discovered
D NimBLERemoteService: << retrieveCharacteristics()
D NimBLERemoteService: >> retrieveCharacteristics() for service: 0x1810
[... 33 lines cut ...]
D NimBLERemoteDescriptor: << Descriptor readValue(): length: 2 rc=0
0x00-0x00-
D NimBLERemoteCharacteristic: >> setNotify(): Characteristic: uuid: 0x2a35, handle: 2065 0x0811, props:  0x20
Descriptor: uuid: 0x2902, handle: 2066, 02
D NimBLERemoteCharacteristic: >> getDescriptor: uuid: 0x2902
D NimBLERemoteCharacteristic: << getDescriptor: found the descriptor with uuid: 0x2902
D NimBLERemoteCharacteristic: << setNotify()
D NimBLERemoteDescriptor: >> Descriptor writeValue: Descriptor: uuid: 0x2902, handle: 2066
I NimBLERemoteDescriptor: Write complete; status=0 conn_handle=0
D NimBLERemoteDescriptor: << Descriptor writeValue, rc: 0
8.1.6 Blood HAS Subscribed for INDICATE...
8.1.4 Blood Descriptor value : 
D NimBLERemoteDescriptor: >> Descriptor readValue: Descriptor: uuid: 0x2902, handle: 2066
D NimBLEClient: Got Client event 
D NimBLERemoteDescriptor: Read complete; status=0 conn_handle=0
D NimBLERemoteDescriptor: Got 2 bytes
D NimBLERemoteDescriptor: Read complete; status=14 conn_handle=0
D NimBLERemoteDescriptor: << Descriptor readValue(): length: 2 rc=0
0x02-0x00-
8.1.7 Blood Pressure Feature Characteristic value : 
D NimBLERemoteCharacteristic: >> readValue(): uuid: 0x2a35, handle: 2065 0x811
I NimBLERemoteCharacteristic: Read complete; status=258 conn_handle=0
E NimBLERemoteCharacteristic: << readValue rc=258
0x00-0x00-0x00-0x00-0x00-0x00-0x00-0x00-0x00-0x00-
10. Done with this device!

juanimoni, 2023-08-12:
I have put some prints in order to check if the Descriptor was changing from 0x0000 to 0x0200...
... and tried to read the characteristic directly, getting the 0's in the last line

h2zero, 2023-08-12:
Thanks, nothing unusual in the logs. If you watch the logs for a while does it ever say notification/indication received?

juanimoni, 2023-08-12:
Thanks to you for spending your time in helping me.

No, I don´t get either Notifications nor Indications in the logs.

should it work automagically by simply stating "if (!pChr->subscribe(false, notifyCB, true))...." ?
should I be able to read also the characteristic ? (even though its property is only Indication ?)

h2zero, 2023-08-12:
You're welcome 😄 

> should it work automagically by simply stating "if (!pChr->subscribe(false, notifyCB, true))...." ?

Yes, it should just work.

> should I be able to read also the characteristic ? (even though its property is only Indication ?)

No, it does not have read capability.

Guesel, 2023-08-12:
I recommend to install the Omron App on an Android Device. Then use the Bluetooth Trace function as developer.
The protocol you get is  readable with Wireshark. Omron device need somes specialized steps establish the communication. The mußte be paired, bonded, synched und getting a GUID. Then you have to Set and unset any notification, specialized command are needed to getting the interesting data.

juanimoni, 2023-08-12:
Thank you Guesel and H2zero. 
I was getting mad and as Guesel states, reading the interesting data from Omron seems not to be automagically...
... what a pitty !!
I will try to further investigate :) 
thank you again for the hints and support.

Guesel, 2023-08-13:
Hi there! For starting you project can i deliver some , hopefully, information. Attached you find a pdf with the first steps you should do.
[OMRON-Procedure-Step1.pdf](https://github.com/h2zero/NimBLE-Arduino/files/12329006/OMRON-Procedure-Step1.pdf)
Good luck!

juanimoni, 2023-08-13:
Hi Guesel, thanks a lot !

Yesterday after getting your message I already put my tablet into developer mode and started sniffing around at the Omron software, but I really was too tired to start analyzing the information I got into wireshark.
At least I know now how to proceed.

Today I have been programming the ESP32 to send mqtt messages... this will be the way to send the Omron info to my server.

I still have 2 days before my kids come back from vacation. 
After that, my "programming-Fest" is over.... so tomorrow I will analyze your info and hopefully I get a step forward.
regards !

juanimoni, 2023-08-14:
Hi again Guesel, I could check your pdf 
I´m getting pretty much the same you are showing in the pdf... 👍 (thanks for the congrats !)

two questions arrise now :
1. I´m using an Omron X7 Smart Device, probably that´s why your pdf is not matching 100% my data...
 (as an example, in line 43 you are using by handle 321 : 080100026026004d and I find in my code 08010000102c0035 instead...)

2. how do I proceed now ?  the interesting part of your pdf are of course lines 41-43... 
I get as an answer from Handle 321 in line 43 :
  361 : 2681c000541e00400100008000809400
  371 : 00000100000000010100440000000000
  381 :  0000460000fb00000000000000000000

(you are showing in the pdf a value of 2e81000260268000800080008000ffffff1c748bffffffffffff5040010f00013f001fe0ffffffffffffffff00280000)
what do those values mean ? (LGQCCCA2A1LDF2EZ NN GMGJTBGTG-C8 fkflMMJJHHTTSSMMC-C8 CS)
and how do I integrate this with NimBle ?

thank you for your support once again !
and best regards.

juanimoni, 2023-09-11:
got it working !
and my son has done a nice interface in home assistant
thanks to both

Guesel, 2023-09-11:
Congratulations, I am very happy about your success! All the best to you and your family!