h2zero/NimBLE-Arduino #258: Notify on connect

podaen, 2021-07-03:
Hardware:
Board: ESP-WROOM-32
Core Installation/update date: 01/MAY/2021
IDE name: Arduino IDE

I am able to send notifications in a **loop** to an app like this and it **works**

```
void SendBLESettings() {
	if (deviceConnected == true) {
		if (BlnSettingsSended == false) {
			delay(10000);
			Serial.println("send ble settings");
			MyPowerCharacteristic->setValue(byte(IntBluetoothPower));
			MyPowerCharacteristic->notify();//power
			Serial.println(IntBluetoothPower);
                 }
         }
}
```

If I try to call this **in connect**. I see in my serial log that it runs it but I do **not recieved it** on the other end.

```
class MyServerCallbacks : public NimBLEServerCallbacks {
	void onConnect(BLEServer* pServer, ble_gap_conn_desc* desc) {
		deviceConnected = true;
		Serial.println("Bluetooth Device Connected.");
		SendBLESettings();
	}
	void onDisconnect(NimBLEServer* pServer) {
		deviceConnected = false;
		//NimBLEDevice::startAdvertising();
		Serial.println("Disconnected");
		BlnSettingsSended = false;
	}
	uint32_t onPassKeyRequest() {
		Serial.println("Server PassKeyRequest");
		return PASSKEY;
		//return 000000;
	}

	bool onConfirmPIN(uint32_t pass_key) {
		Serial.print("The passkey YES/NO number: "); Serial.println(pass_key);
		return true;
	}

	void onAuthenticationComplete(ble_gap_conn_desc* desc) {
		/** Check that encryption was successful, if not we disconnect the client */
		if (!desc->sec_state.encrypted) {
			NimBLEDevice::getServer()->disconnect(desc->conn_handle);
			Serial.println("Encrypt connection failed - disconnecting client");
			return;
		}
		Serial.println("Starting BLE work!");
		
	}
	/*******************************************************************/
};
```
Any Idea's why that could be? I only runs this "sendsettings" once and it works in the loop. Why not in the connect? I set a delay up to 10seconds and had long before achieved a connection in the app. note: The passkey is also given in the android.

h2zero, 2021-07-03:
If you put a delay in the `sendsettings` you are delaying the BLE stack. The reason for the notification not being sent at that time is because the client has not yet subscribed to notifications for the characteristic. I suggest you set a callback for the characteristic and send the notifications from the `onSubscribe` callback.

podaen, 2021-07-03:
That could be a solution. I didn't see that. I don't have any problems when I am connected and recieve and directly send notifications soo I think it will work. I can leave the delay out and I m not loading my running code with an extra task. Thanks you verry much.

podaen, 2021-07-04:
I tried this but I am not recieving anything on the other hand

```
class MyPowerCharacteristicCallbacks : public NimBLECharacteristicCallbacks {
	void onWrite(NimBLECharacteristic* pCharacteristic) {
		std::string value = pCharacteristic->getValue();
		String StrTemp = value.c_str();
	}
	//*@brief Callback function called when a client changes subscription status.
	//	* @param[in] pCharacteristic The characteristic that is the source of the event.
	//	* @param[in] desc The connection description struct that is associated with the client.
	//	* @param[in] subValue The subscription status :
	//	**0 = Un - Subscribed
	//		* *1 = Notifications
	//		* *2 = Indications
	//		* *3 = Notifications and Indications
	//		* /
	void onSubscribe(NimBLECharacteristic* pCharacteristic,
		ble_gap_conn_desc* desc,
		uint16_t subValue) {
		Serial.println("subscription power");
		pCharacteristic->setValue(byte(IntBluetoothPower));
		pCharacteristic->notify();//power
		Serial.println(IntBluetoothPower);
	}
};
```
Serial monitor show me this...

> Bluetooth Device Connected.
> Starting BLE work!
> subscription power
> 1
> Disconnected

h2zero, 2021-07-04:
Could you set the log level to debug and paste the log?

podaen, 2021-07-04:
From the build you mean?

h2zero, 2021-07-04:
I mean run the code with debug log level and paste the serial output so I can see the log messages from the library.

podaen, 2021-07-04:
hmm... Interesting.

> 
> I NimBLEDevice: "BLE Host Task Started"
> I NimBLEDevice: "NimBle host synced."
> D NimBLEDevice: ">> setLocalMTU: 255"
> D NimBLEDevice: "<< setLocalMTU"
> D NimBLEDevice: "Setting bonding: 1, mitm: 1, sc: 1"
> D NimBLEServer: ">> createService - 35c8c410-8e8d-43e9-9786-5d48fc352903"
> D NimBLEServer: "<< createService"
> D NimBLEUUID: "Comparing UUIDs; type 16 to 16; UUID 0x2904 to 0x2902"
> D NimBLEUUID: "Comparing UUIDs; type 16 to 16; UUID 0x2904 to 0x2904"
> D NimBLEUUID: "Comparing UUIDs; type 16 to 16; UUID 0x2901 to 0x2902"
> D NimBLEUUID: "Comparing UUIDs; type 16 to 16; UUID 0x2901 to 0x2904"
> D NimBLEUUID: "Comparing UUIDs; type 128 to 128; UUID 3ff8e456-17a8-488b-95ad-650496f52903 to 8b024ca6-cd53-452e-8968-50e0570e2903"
> D NimBLEUUID: "Comparing UUIDs; type 16 to 16; UUID 0x2904 to 0x2902"
> D NimBLEUUID: "Comparing UUIDs; type 16 to 16; UUID 0x2904 to 0x2904"
> D NimBLEUUID: "Comparing UUIDs; type 16 to 16; UUID 0x2901 to 0x2902"
> D NimBLEUUID: "Comparing UUIDs; type 16 to 16; UUID 0x2901 to 0x2904"
> D NimBLEUUID: "Comparing UUIDs; type 128 to 128; UUID 3ff8e456-17a8-488b-95ad-650496f52903 to 86216ac6-b935-11eb-8529-0242ac132903"
> D NimBLEUUID: "Comparing UUIDs; type 128 to 128; UUID 8b024ca6-cd53-452e-8968-50e0570e2903 to 86216ac6-b935-11eb-8529-0242ac132903"
> D NimBLEUUID: "Comparing UUIDs; type 16 to 16; UUID 0x2904 to 0x2902"
> D NimBLEUUID: "Comparing UUIDs; type 16 to 16; UUID 0x2904 to 0x2904"
> D NimBLEUUID: "Comparing UUIDs; type 16 to 16; UUID 0x2901 to 0x2902"
> D NimBLEUUID: "Comparing UUIDs; type 16 to 16; UUID 0x2901 to 0x2904"
> D NimBLEUUID: "Comparing UUIDs; type 128 to 128; UUID 3ff8e456-17a8-488b-95ad-650496f52903 to 15f6c0d3-5d12-4b50-a3b3-fc8b41092903"
> D NimBLEUUID: "Comparing UUIDs; type 128 to 128; UUID 8b024ca6-cd53-452e-8968-50e0570e2903 to 15f6c0d3-5d12-4b50-a3b3-fc8b41092903"
> D NimBLEUUID: "Comparing UUIDs; type 128 to 128; UUID 86216ac6-b935-11eb-8529-0242ac132903 to 15f6c0d3-5d12-4b50-a3b3-fc8b41092903"
> D NimBLEUUID: "Comparing UUIDs; type 16 to 16; UUID 0x2904 to 0x2902"
> D NimBLEUUID: "Comparing UUIDs; type 16 to 16; UUID 0x2904 to 0x2904"
> D NimBLEUUID: "Comparing UUIDs; type 16 to 16; UUID 0x2901 to 0x2902"
> D NimBLEUUID: "Comparing UUIDs; type 16 to 16; UUID 0x2901 to 0x2904"
> D NimBLEUUID: "Comparing UUIDs; type 128 to 128; UUID 3ff8e456-17a8-488b-95ad-650496f52903 to b3fb75eb-f7c8-4b6a-bf33-9620b86a2903"
> D NimBLEUUID: "Comparing UUIDs; type 128 to 128; UUID 8b024ca6-cd53-452e-8968-50e0570e2903 to b3fb75eb-f7c8-4b6a-bf33-9620b86a2903"
> D NimBLEUUID: "Comparing UUIDs; type 128 to 128; UUID 86216ac6-b935-11eb-8529-0242ac132903 to b3fb75eb-f7c8-4b6a-bf33-9620b86a2903"
> D NimBLEUUID: "Comparing UUIDs; type 128 to 128; UUID 15f6c0d3-5d12-4b50-a3b3-fc8b41092903 to b3fb75eb-f7c8-4b6a-bf33-9620b86a2903"
> D NimBLEUUID: "Comparing UUIDs; type 16 to 16; UUID 0x2904 to 0x2902"
> D NimBLEUUID: "Comparing UUIDs; type 16 to 16; UUID 0x2904 to 0x2904"
> D NimBLEUUID: "Comparing UUIDs; type 16 to 16; UUID 0x2901 to 0x2902"
> D NimBLEUUID: "Comparing UUIDs; type 16 to 16; UUID 0x2901 to 0x2904"
> D NimBLEUUID: "Comparing UUIDs; type 128 to 128; UUID 3ff8e456-17a8-488b-95ad-650496f52903 to 4fa6c4d7-25c6-4806-810d-4111e3c42903"
> D NimBLEUUID: "Comparing UUIDs; type 128 to 128; UUID 8b024ca6-cd53-452e-8968-50e0570e2903 to 4fa6c4d7-25c6-4806-810d-4111e3c42903"
> D NimBLEUUID: "Comparing UUIDs; type 128 to 128; UUID 86216ac6-b935-11eb-8529-0242ac132903 to 4fa6c4d7-25c6-4806-810d-4111e3c42903"
> D NimBLEUUID: "Comparing UUIDs; type 128 to 128; UUID 15f6c0d3-5d12-4b50-a3b3-fc8b41092903 to 4fa6c4d7-25c6-4806-810d-4111e3c42903"
> D NimBLEUUID: "Comparing UUIDs; type 128 to 128; UUID b3fb75eb-f7c8-4b6a-bf33-9620b86a2903 to 4fa6c4d7-25c6-4806-810d-4111e3c42903"
> D NimBLEUUID: "Comparing UUIDs; type 16 to 16; UUID 0x2904 to 0x2902"
> D NimBLEUUID: "Comparing UUIDs; type 16 to 16; UUID 0x2904 to 0x2904"
> D NimBLEUUID: "Comparing UUIDs; type 16 to 16; UUID 0x2901 to 0x2902"
> D NimBLEUUID: "Comparing UUIDs; type 16 to 16; UUID 0x2901 to 0x2904"
> D NimBLEService: ">> start(): Starting service: UUID: 35c8c410-8e8d-43e9-9786-5d48fc352903, handle: 0xffff"
> D NimBLEService: "Adding 6 characteristics for service UUID: 35c8c410-8e8d-43e9-9786-5d48fc352903, handle: 0xffff"
> D NimBLEService: "<< start()"
> D NimBLEAdvertising: ">> Advertising start: customAdvData: 0, customScanResponseData: 0"
> primary service
>            uuid 0x1800
>          handle 1
>      end_handle 5
> characteristic
>            uuid 0x2a00
>      def_handle 2
>      val_handle 3
>    min_key_size 0
>           flags [READ]
> characteristic
>            uuid 0x2a01
>      def_handle 4
>      val_handle 5
>    min_key_size 0
>           flags [READ]
> primary service
[... 169 lines cut ...]

This is the characteristic where I'm talking about.

```
void CharacteristicPower() {
	MyPowerCharacteristic = MyService->createCharacteristic(
		CharUUID_Power,
		NIMBLE_PROPERTY::READ |//voor het verzenden van een notificatie
		NIMBLE_PROPERTY::WRITE |//voor het ontvagen van boodschap (moet er in staan ook bij pinrequest)
		NIMBLE_PROPERTY::READ_ENC |//voor pin beveiliging
		NIMBLE_PROPERTY::NOTIFY
		//NIMBLE_PROPERTY::WRITE_ENC  //voor pin beveiliging
	);
	MyPowerCharacteristic->setCallbacks(&MyPowerchrCallbacks);
	NimBLE2904* MyPower2904 = (NimBLE2904*)MyPowerCharacteristic->createDescriptor("2904");
	MyPower2904->setFormat(NimBLE2904::FORMAT_UTF8);//This can recieve 20 characters

	NimBLEDescriptor* MyPowerDescriptor = MyPowerCharacteristic->createDescriptor(
		"2901",
		NIMBLE_PROPERTY::READ,
		20
	);
	MyPowerDescriptor->setValue("Power");
}
```

h2zero, 2021-07-04:
Thanks, that is quite odd that it would report no clients subscribed when clearly there is. I'll need to investigate.

podaen, 2021-07-04:
It seems some libary files are not found in the project like "ble_att.c".  Can't find any #include of that file....

also have some not fatal errors in the entire solution...

> BLE.h line 684 device is undifined
> BLE.h 685 expected a ')'
> ble_uuid.h 95 expected a ')'
> ble_uuid.h 107 expected a ')'

h2zero, 2021-07-04:
You need to delete your NimBLE-Arduino folder and then copy the new master to your libraries.

podaen, 2021-07-04:
Connection is faster than every before. But it not I still don't recieve anything. Here is my log

> 
> I NimBLEDevice: "BLE Host Task Started"
> ble_hs_hci_cmd_send: ogf=0x03 ocf=0x0003 len=0
> 0x03 0x0c 0x00 
> ble_hs_hci_cmd_send: ogf=0x04 ocf=0x0001 len=0
> 0x01 0x10 0x00 
> ble_hs_hci_cmd_send: ogf=0x04 ocf=0x0003 len=0
> 0x03 0x10 0x00 
> ble_hs_hci_cmd_send: ogf=0x03 ocf=0x0001 len=8
> 0x01 0x0c 0x08 0x90 0x80 0x00 0x02 0x00 0x80 0x00 0x20 
> ble_hs_hci_cmd_send: ogf=0x03 ocf=0x0063 len=8
> 0x63 0x0c 0x08 0x00 0x00 0x80 0x00 0x00 0x00 0x00 0x00 
> ble_hs_hci_cmd_send: ogf=0x08 ocf=0x0001 len=8
> 0x01 0x20 0x08 0x7f 0x06 0x00 0x00 0x00 0x00 0x00 0x00 
> ble_hs_hci_cmd_send: ogf=0x08 ocf=0x0002 len=0
> 0x02 0x20 0x00 
> ble_hs_hci_cmd_send: ogf=0x08 ocf=0x0003 len=0
> 0x03 0x20 0x00 
> ble_hs_hci_cmd_send: ogf=0x04 ocf=0x0009 len=0
> 0x09 0x10 0x00 
> ble_hs_hci_cmd_send: ogf=0x08 ocf=0x0018 len=0
> 0x18 0x20 0x00 
> ble_hs_hci_cmd_send: ogf=0x08 ocf=0x0018 len=0
> 0x18 0x20 0x00 
> Device added to RL, Resolving list count = 1
> ble_hs_hci_cmd_send: ogf=0x03 ocf=0x0031 len=1
> 0x31 0x0c 0x01 0x01 
> ble_hs_hci_cmd_send: ogf=0x03 ocf=0x0033 len=7
> 0x33 0x0c 0x07 0xff 0x00 0x00 0x0c 0x00 0x00 0x00 
> ble_hs_hci_cmd_send: ogf=0x03 ocf=0x0031 len=1
> 0x31 0x0c 0x01 0x00 
> looking up peer sec; 
> ble_hs_hci_cmd_send: ogf=0x08 ocf=0x0018 len=0
> 0x18 0x20 0x00 
> ble_hs_hci_cmd_send: ogf=0x08 ocf=0x0018 len=0
> 0x18 0x20 0x00 
> Device added to RL, Resolving list count = 2
> looking up peer sec; 
> I NimBLEDevice: "NimBle host synced."
> D NimBLEDevice: ">> setLocalMTU: 255"
> D NimBLEDevice: "<< setLocalMTU"
> D NimBLEDevice: "Setting bonding: 1, mitm: 1, sc: 1"
> D NimBLEServer: ">> createService - 35c8c410-8e8d-43e9-9786-5d48fc352903"
> D NimBLEServer: "<< createService"
> D NimBLEService: ">> start(): Starting service: UUID: 35c8c410-8e8d-43e9-9786-5d48fc352903, handle: 0xffff"
> D NimBLEService: "Adding 6 characteristics for service UUID: 35c8c410-8e8d-43e9-9786-5d48fc352903, handle: 0xffff"
> D NimBLEService: "<< start()"
> D NimBLEAdvertising: ">> Advertising start: customAdvData: 0, customScanResponseData: 0"
> primary service
>            uuid 0x1800
>          handle 1
>      end_handle 5
> characteristic
>            uuid 0x2a00
>      def_handle 2
>      val_handle 3
>    min_key_size 0
>           flags [READ]
> characteristic
>            uuid 0x2a01
>      def_handle 4
>      val_handle 5
>    min_key_size 0
>           flags [READ]
> primary service
>            uuid 0x1801
>          handle 6
>      end_handle 9
[... 174 lines cut ...]
> I NimBLEServer: "subscribe event; attr_handle=22, subscribed: true"
> I NimBLECharacteristic: "New subscribe value for conn: 0 val: 1"
> D NimBLECharacteristicCallbacks: "onSubscribe: default"
> D NimBLEServer: ">> handleGapEvent: "
> I NimBLEServer: "subscribe event; attr_handle=35, subscribed: true"
> I NimBLECharacteristic: "New subscribe value for conn: 0 val: 1"
> D NimBLECharacteristicCallbacks: "onSubscribe: default"
> ble_hs_hci_evt_acl_process(): conn_handle=0 pb=2 len=7 data=0x03 0x00 0x04 0x00 0x02 0xff 0x00 
> host tx hci data; handle=0 length=7
> ble_hs_hci_acl_tx(): 0x00 0x00 0x07 0x00 0x03 0x00 0x04 0x00 0x03 0xff 0x00 
> D NimBLEServer: ">> handleGapEvent: "
> I NimBLEServer: "mtu update event; conn_handle=0 mtu=255"
> ble_hs_hci_evt_acl_process(): conn_handle=0 pb=2 len=9 data=0x05 0x00 0x04 0x00 0x12 0x0d 0x00 0x01 0x00 
> host tx hci data; handle=0 length=5
> ble_hs_hci_acl_tx(): 0x00 0x00 0x05 0x00 0x01 0x00 0x04 0x00 0x13 
> ble_hs_hci_evt_acl_process(): conn_handle=0 pb=2 len=9 data=0x05 0x00 0x04 0x00 0x12 0x12 0x00 0x01 0x00 
> host tx hci data; handle=0 length=5
> ble_hs_hci_acl_tx(): 0x00 0x00 0x05 0x00 0x01 0x00 0x04 0x00 0x13 
> ble_hs_hci_evt_acl_process(): conn_handle=0 pb=2 len=9 data=0x05 0x00 0x04 0x00 0x12 0x17 0x00 0x01 0x00 
> host tx hci data; handle=0 length=5
> ble_hs_hci_acl_tx(): 0x00 0x00 0x05 0x00 0x01 0x00 0x04 0x00 0x13 
> ble_hs_hci_evt_acl_process(): conn_handle=0 pb=2 len=9 data=0x05 0x00 0x04 0x00 0x12 0x24 0x00 0x01 0x00 
> host tx hci data; handle=0 length=5
> ble_hs_hci_acl_tx(): 0x00 0x00 0x05 0x00 0x01 0x00 0x04 0x00 0x13 
>

h2zero, 2021-07-04:
I don't know why you aren't receiving it but I can see it has been sent.
```
GATT procedure initiated: notify; att_handle=12
host tx hci data; handle=0 length=8
ble_hs_hci_acl_tx(): 0x00 0x00 0x08 0x00 0x04 0x00 0x04 0x00 0x1b 0x0c 0x00 0x01
``` 
So it's definitely doing what it's supposed to.

podaen, 2021-07-04:
without connection no?!

```
I NimBLECharacteristic: "New subscribe value for conn: 0 val: 1"
subscription power
```

podaen, 2021-07-04:
I have a messagebox that pops up whene I recieve it. and I have it recieved by trieing it a few times . When I delay it I recieved it evry time. Don't know what the minimum amount is yet.

```
	void onSubscribe(NimBLECharacteristic* pCharacteristic,
		ble_gap_conn_desc* desc,
		uint16_t subValue) {
		delay(5000);
		Serial.println("subscription power");
		pCharacteristic->setValue(byte(IntBluetoothPower));
		pCharacteristic->notify();//power
		Serial.println(IntBluetoothPower);
	}
```

podaen, 2021-07-04:
Now I think it is because the app is buzy by checking it is connected or not.

h2zero, 2021-07-04:
The logs look to be exchanging MTU after the notification is sent, it's quite possible the app/phone is ignoring the notification until that process is complete. I would suggest using indications instead of notifications, then the client will need to respond and should guarantee the message arrives.

podaen, 2021-07-05:
I know it runs on the same time but it's not the mtu request. It's somthing in the app. I tried an other project of me and it works. Soo I will close it.

podaen, 2021-07-05:
Is this engough to indicate?

```
void CharacteristicPower() {
	MyPowerCharacteristic = MyService->createCharacteristic(
		CharUUID_Power,
		NIMBLE_PROPERTY::READ |
		NIMBLE_PROPERTY::WRITE |
		NIMBLE_PROPERTY::READ_ENC |
		NIMBLE_PROPERTY::INDICATE
	);
	MyPowerCharacteristic->setCallbacks(&MyPowerchrCallbacks);

	NimBLE2904* MyPower2904 = (NimBLE2904*)MyPowerCharacteristic->createDescriptor("2904");
	MyPower2904->setFormat(NimBLE2904::FORMAT_UTF8);//This can recieve 20 characters

	NimBLEDescriptor* MyPowerDescriptor = MyPowerCharacteristic->createDescriptor(
		"2901",
		NIMBLE_PROPERTY::READ,
		20
	);
	MyPowerDescriptor->setValue("Power");
}
```

```
	void onSubscribe(NimBLECharacteristic* pCharacteristic,
		ble_gap_conn_desc* desc,
		uint16_t subValue) {
		//delay(5000);
		Serial.println("subscription power");
		pCharacteristic->setValue(byte(IntBluetoothPower));
		//pCharacteristic->notify();//power
		pCharacteristic->indicate();
		Serial.println(IntBluetoothPower);
	}
```

h2zero, 2021-07-05:
Yes, that is all you need.

podaen, 2021-07-05:
The problem is with notification that when I add two register for bytes after each other it doesn't recieve anything anymore. And when I change it llike above it doesn't recieve anything with register for bytes.

 I don't understand this. I am only sending a value of 0x01.  And this is sended:
host tx hci data; handle=0 length=8
ble_hs_hci_acl_tx(): 0x00 0x00 0x08 0x00 0x04 0x00 0x04 0x00 0x1b 0x0c 0x00 0x01

podaen, 2021-07-06:
I did some tests with other applications too and in what I can see is that it's sends notifications when it's not registerd yet. Soo onsubscribe is too fast. i removed the mtu request and check the device was not buzy with scanning and soo on. I can not faster register than I am doing now.

h2zero, 2021-07-06:
Speed is not likely the problem, the request to subscribe was received and notification/indication sent. These packets are guaranteed to be delivered. My guess is there is something wrong on the client side that is not reporting the packet to your app, you would need to use a sniffer to confirm.

podaen, 2021-07-06:
I have a lot of delays on the app. one for checking if connected, one for stop scanning and one now for register. 

Could It be that my server is not config well? 

```
void StartBLE() {
	iniBLE();
	pServer = NimBLEDevice::createServer();
	pServer->setCallbacks(new MyServerCallbacks());
	MyService = pServer->createService(NimBLEUUID(SERVICE_UUID), 30);//30 handles
	CharacteristicPower();
	CharacteristicBrightness();
	MyService->start();
	StartAdvertising();
	Serial.println("Start advertising");
}
void iniBLE() {
	NimBLEDevice::init("test");//devicename
	NimBLEDevice::setMTU(255);//128bytes
	NimBLEDevice::setSecurityAuth(true, true, true);
	NimBLEDevice::setSecurityPasskey(PASSKEY);
	NimBLEDevice::setSecurityIOCap(BLE_HS_IO_DISPLAY_ONLY);
}
void StartAdvertising() {
	MyAdvertising = NimBLEDevice::getAdvertising();
	MyAdvertising->addServiceUUID(SERVICE_UUID);
	MyAdvertising->setScanResponse(false);
        MyAdvertising->start();
}
```

h2zero, 2021-07-07:
I see no issues with your configuration, and from the logs posted we can see the esp32 side is working correctly. I am at a loss as to what could be happening on the client side app that would cause this. I'm curious, if using another esp32 as a client in a similar manner, if it would have the same result.

chegewara, 2021-07-07:
It is very long topic and i was some time on vacations, so i didnt read all posts.
From opening post i can tell you it will not work. Why:
- before you can receive notifications, you have too register for notifications in client app (no matter what kind of client it is),
- even if you ignore descriptor 2902 and just send notifications, client app has to get characteristic handle (most likely all characteristics), and i can assure you it takes some time and wont happen before you send first notification from `onConnect` callback

podaen, 2021-07-07:
@chegewara :100: Nice to back. I did the register for notifications immediately at the client after connecting and left the onConnect out on the server. Soo I don't think I can do more than that.

I don't have a sniffer yet, I am using nfr connect for now I can see the value is setted right. 

I can think one causes for now and that is that the end marker in the response of the server is not the same as the client and therefore runs longer than needed.

podaen, 2021-07-19:
I solved it by making a charachteristic that if all registration are done for reading, the app sends that the device is ready to recieve through that specific characteristic. Only after that comfirmation the notification are sended emidditly.

closed 2021-07-19 by podaen