h2zero/NimBLE-Arduino #259: How can I get BLE raw data?

Dec-Z, 2021-07-05:
Hi, I want my esp32 to scan ble device and send back a raw data.

[image]

[image]

I using example scan code. And this is scan response.
`14:12:45.055 -> Advertised Device: Name: OEMUTS, Address: 00:15:88:07:40:e4, serviceUUID: 0xfaf0, txPower: 8`
`14:12:45.055 -> Service Data:`                                                                                                                                 
`14:12:45.055 -> UUID: 0xfaf0, Data: ⸮@⸮ `
 Can anyone give me some pointers.
    :-)

h2zero, 2021-07-05:
You can use `NimBLEAdvertisedDevice::getServiceData()` to get the data. https://h2zero.github.io/esp-nimble-cpp/class_nim_b_l_e_advertised_device.html#acf4398ca81f800445092ccc77deaf00b

If you want to display it on the serial monitor you can use the `NimBLEUtils::buildHexData()` to convert it to an ascii string.
https://h2zero.github.io/esp-nimble-cpp/class_nim_b_l_e_utils.html#a6382edb7e8e4030b4fc65d463fa2a886

Dec-Z, 2021-07-07:
Sorry for the late reply sir. I still can't get raw data from any device. I saw the code you mentioned  already in the .cpp file, so I haven't edited it. If you don't mind Can I ask for more advice? 
Which part do I have to edit to get raw data from scanning?
```
std::string NimBLEAdvertisedDevice::getServiceData(const NimBLEUUID & uuid) {
    ble_hs_adv_field *field = nullptr;
    uint8_t bytes;
    uint8_t index = 0;
    uint8_t data_loc = findServiceData(index, &bytes);
    uint8_t uuidBytes = uuid.bitSize() / 8;
    uint8_t plSize = m_payload.size() - 2;

    while(data_loc < plSize) {
        field = (ble_hs_adv_field *)&m_payload[data_loc];
        if(bytes == uuidBytes && NimBLEUUID(field->value, bytes, false) == uuid) {
           return std::string((char*)(field->value + bytes), field->length - bytes - 1);
        }

        index++;
        data_loc = findServiceData(index, &bytes);
    }

    NIMBLE_LOGI(LOG_TAG, "No service data found");
    return "";
} //getServiceData
```

```
char* NimBLEUtils::buildHexData(uint8_t* target, const uint8_t* source, uint8_t length) {
    // Guard against too much data.
    if (length > 100) length = 100;

    if (target == nullptr) {
        target = (uint8_t*) malloc(length * 2 + 1);
        if (target == nullptr) {
            NIMBLE_LOGE(LOG_TAG, "buildHexData: malloc failed");
            return nullptr;
        }
    }
    char* startOfData = (char*) target;

    for (int i = 0; i < length; i++) {
        sprintf((char*) target, "%.2x", (char) *source);
        source++;
        target += 2;
    }
    // Handle the special case where there was no data.
    if (length == 0) {
        *startOfData = 0;
    }
    return startOfData;
} // buildHexData

```

h2zero, 2021-07-07:
You don't need to edit anything in the library code, it's all accessible from your app. Could you explain more about what data you want to retrieve, are you looking to get the entire advertisement packet? For that you can use `NimBLEAdvertisedDevice::getPayload()` and `NimBLEAdvertisedDevice::getPayloadLength()`

Dec-Z, 2021-07-21:
I used this code so it's work now.
```
uint8_t* p, i, array[30]; 
      p = advertisedDevice->getPayload();
      for (i=0; i<24 ; i++)   
      Serial.printf(" %02x" , *p++);
```
and got this at the responed.
`02 01 06 03 02 f0 fa 0e 16 f0 fa c9 2b 07 88 15 00 00 00 d2 06 4a 13 16`
 
My work wouldn't be finished without this library, Thank you very much.

closed 2021-07-21 by Dec-Z