h2zero/NimBLE-Arduino #260: `OnConfirmPIN` is not called even if `BLE_HS_IO_DISPLAY_YESNO` is specified

wakwak-koba, 2021-07-06:
When I uncommented line 160 of `example/NimBLE_Server/NimBLE_Server.ino`, enabled it, and compiled it, I thought that `ServerCallbacks::onConfirmPIN()` would be called, but it was never called and `ServerCallbacks::onPassKeyRequest()` would be called.

https://github.com/h2zero/NimBLE-Arduino/blob/e9bda73e88d42b5b18d2cf1c64e3c708e7ee17b0/examples/NimBLE_Server/NimBLE_Server.ino#L160

How can I get `ServerCallbacks::onPassKeyRequest()` to be called?

h2zero, 2021-07-07:
Did you also change `NimBLEDevice::setSecurityAuth(/*BLE_SM_PAIR_AUTHREQ_BOND | BLE_SM_PAIR_AUTHREQ_MITM |*/ BLE_SM_PAIR_AUTHREQ_SC);` to require `BLE_SM_PAIR_AUTHREQ_MITM `?

Such as: NimBLEDevice::setSecurityAuth(BLE_SM_PAIR_AUTHREQ_MITM | BLE_SM_PAIR_AUTHREQ_SC);`

wakwak-koba, 2021-07-07:
Even with `NimBLEDevice::setSecurityAuth(BLE_SM_PAIR_AUTHREQ_MITM | BLE_SM_PAIR_AUTHREQ_SC);`, `onPassKeyRequest()` will be called.

h2zero, 2021-07-07:
That is strange, I will test this and see what could be happening.

wakwak-koba, 2021-07-09:
I was testing with two Android Phones and started ordering for the following points:
https://github.com/h2zero/NimBLE-Arduino/blob/e9bda73e88d42b5b18d2cf1c64e3c708e7ee17b0/src/nimble/host/src/ble_sm_sc.c#L56-L79

I feel that `BLE_SM_IOACT_DISP` has been chosen as a compromise between the server and the client.

I'd like to check the values ​​of both `io_caps`, but how do I enable BLE_HS_LOG under Arduino?

h2zero, 2021-07-09:
It seems that the client device you are using does not support BLE secure connections (numeric comparison).

wakwak-koba, 2021-07-10:
I placed BLE_HS_LOG here and there and tested it.
As you can guess, the Android Phone I used for my test didn't support Secure-mode and apparently only connected with Legacy.

https://github.com/h2zero/NimBLE-Arduino/blob/e9bda73e88d42b5b18d2cf1c64e3c708e7ee17b0/src/nimble/host/src/ble_sm_lgcy.c#L38-L60

I thought it would be useful to have an easy way to check the IOCAPs and ble_sm_proc.flags of the other terminal.

This issue is resolved. 
I'm sorry for making a noise...

closed 2021-07-10 by wakwak-koba