h2zero/NimBLE-Arduino #261: BLE Server timeout when connecting from nRF Connect

pscs, 2021-07-10:
Hi, I'm trying to make a BLE server on an ESP32 WROOM. 

I was getting connection timeouts with my own code, so to see if I could get something working, I'm using the example code at https://github.com/h2zero/NimBLE-Arduino/tree/master/examples/NimBLE_Server

But, when I try to connect from nRF Connect on my Android phone, it's just the same as with my code - it connects, then says 'Error 8: GATT CONN TIMEOUT'.

I've removed the "pServer->updateConnParams(desc->conn_handle, 24, 48, 0, 60);" line from the example, because I've seen another post suggesting that may be the problem. But, that makes no noticeable difference.

I've attached the debug log (ble.log) from the ESP32 and a bt_hci.log from the phone (for Wireshark - bt.log) in the hope that they'll help.

I'm not a BLE expert, but from the logsit looks as if the phone is sending a 'Find Information Request' message for handle 0x000d, but never getting a response to it because the ESP32 has disconnected before that for some reason.

[ble.log](https://github.com/h2zero/NimBLE-Arduino/files/6795305/ble.log)
[bt.log](https://github.com/h2zero/NimBLE-Arduino/files/6795307/bt.log)

pscs, 2021-07-10:
Also, here's a Wireshark capture from my BLE sniffer running on my laptop, while my phone was trying to connect to the ESP32
[bt2.zip](https://github.com/h2zero/NimBLE-Arduino/files/6795349/bt2.zip)

h2zero, 2021-07-10:
Hi @pscs, the logs look strange indeed. I can see the request being responded to multiple times for the 0x000d handle, which is the 2904 descriptor. I'm unsure why this is happening but if you delete that descriptor do you still have the issue?

pscs, 2021-07-10:
I've deleted all references to 'pBeef2904', which is what I guess you meant.

The same thing happens with the connection timing out. The logs are obviously different (just asking for handles up to 0x000c). I've uploaded the ESP32 debug log and the sniffer capture log

[bt3.zip](https://github.com/h2zero/NimBLE-Arduino/files/6795566/bt3.zip)
[ble1.log](https://github.com/h2zero/NimBLE-Arduino/files/6795567/ble1.log)

> I can see the request being responded to multiple times for the 0x000d handle

I thought that was normal. I thought the controller said, for instance "tell me about handles 8-13" and the server says "handle 8 doesn't exist", so the controller says, "OK, so tell me about handles 9-13", and so on, so you see many requests referring to the last handle in a range. Or maybe I'm misunderstanding.

h2zero, 2021-07-10:
> I thought that was normal. I thought the controller said, for instance "tell me about handles 8-13" and the server says "handle 8 doesn't exist", so the controller says, "OK, so tell me about [...]

You're correct, sorry, I didn't look close enough. I thought these were the same:
```
ble_hs_hci_evt_acl_process(): conn_handle=0 pb=2 len=11 data=0x07 0x00 0x04 0x00 0x08 0x0a 0x00 0x0d 0x00 0x02 0x28 
rxed att command: read type req; conn=0 start_handle=0x000a end_handle=0x000d
txed att command: error rsp; conn=0 req_op=8 handle=0x000a error_code=10
host tx hci data; handle=0 length=9
ble_hs_hci_acl_tx(): 0x00 0x00 0x09 0x00 0x05 0x00 0x04 0x00 0x01 0x08 0x0a 0x00 0x0a 
Number of Completed Packets: num_handles=1
handle:0 pkts:1
ble_hs_hci_evt_acl_process(): conn_handle=0 pb=2 len=11 data=0x07 0x00 0x04 0x00 0x08 0x0a 0x00 0x0d 0x00 0x03 0x28 
rxed att command: read type req; conn=0 start_handle=0x000a end_handle=0x000d
```

When I have more time later today I will look more closely at the wireshark trace.

pscs, 2021-07-10:
OK - I've got it working, but I'm not entirely confident.

I found that the tests above were done with NimBLE 1.2.0 because that's the latest version that the Arduino IDE library manager knows about. I decided to delete that and manually download the 'master' branch - and that seems to work fine, and it connects and I can do reads & writes and receive notifications

I'm not entirely sure which branch I should be using... It looks as if 1.2.0 is the latest released version, so should be the one to use.

h2zero, 2021-07-11:
That is quite odd, the master branch has some updates to the nimble core that may have fixed the issue. It's fine to use the master branch as it's very stable at the moment, I just need to cleanup and document a few things and it will be ready for release as 1.3.0.

pscs, 2021-07-12:
Thanks. I'll use the 'master' branch for now.

FWIW, I did switch back & forth between the 1.2.0 and master branches downloading afresh and the results were repeatable. 'master' works fine, 1.2.0 doesn't. Very odd because I presume 1.2.0 works fine for other people, and I'm just using a stock example.

closed 2021-07-12 by pscs

h2zero, 2021-07-12:
>FWIW, I did switch back & forth between the 1.2.0 and master branches downloading afresh and the results were repeatable. 'master' works fine, 1.2.0 doesn't. Very odd because I presume 1.2.0 works fine for other people, and I'm just using a stock example.

Yes that is odd, 1.2.0 has been out for quite some time and I have not heard of any similar issues to what you have described. I am curious as to what arduino core version you are using?