h2zero/NimBLE-Arduino #262: no appearance at all

gromeck, 2021-07-12:
Hi,
can somebody give me a hint how to read the appearance from a device?
It looks like the advertised device should report it in the scan results, but so far, I had no device at all with a non-zero appearance.
Thx

h2zero, 2021-07-12:
You can call `advertisedDevice-> haveAppearance();` to check if the device is advertising one and `advertisedDevice-> getAppearance();` to get the value of it.

If it isn't being advertised then the value will be 0.

gromeck, 2021-07-13:
Yes, thanks, this is exactly what I did -- but none of ~50 responding devices around me report their appearance.
On the other hand, only 3 of these 50 use public addresses, I assume that there are mainly mobiles with rolling addresses -- but none reveals its appearance :-(

gromeck, 2021-07-14:
... or other way around: what are your experiences with the appearance?

h2zero, 2021-07-14:
It's rare to see a device advertising an appearance in my experience, usually only on HID devices.

davidchatting, 2021-08-18:
That's strange - is there another way of telling device type then? Because my impression is scans on my phone etc normally do a good job of categorising devices.

h2zero, 2021-08-18:
>That's strange - is there another way of telling device type then?

If the device doesn't advertise an appearance then you would need to investigate the advertised data to see what it's advertising, could have a name/service UUID/service data etc.. For instance, if it is advertising service UUID 0x181A, it's an environmental sensor. You can see all the assigned UUID's [here](https://btprodspecificationrefs.blob.core.windows.net/assigned-values/16-bit%20UUID%20Numbers%20Document.pdf).

>Because my impression is scans on my phone etc normally do a good job of categorising devices. 

The built in scanner on your phone only shows bluetooth classic devices, which almost always have appearances. BLE devices rarely do, usually only HID devices.

davidchatting, 2021-08-18:
Thank you that's very helpful.

h2zero, 2022-04-05:
Closing this as a non-issue. Please reopen as a discussion if necessary.

closed 2022-04-05 by h2zero