h2zero/NimBLE-Arduino #263: Specifying Client Address

miag676, 2021-07-14:
Hello, first of all thanks for making and maintaining the library. I would like to connect to a specific BLE beacon. Currently I am using this command and it's working perfectly:
`pClient = NimBLEDevice::createClient(NimBLEAddress("49:42:06:00:20:4e"));`
However, if I substitute it with a variable I get the error: no matching function for call to 'NimBLEAddress::NimBLEAddress(String)'.
This is the code:
`pClient = NimBLEDevice::createClient(NimBLEAddress(String(value)));`

Can the command accept such values or must it be specified like in the first case?

Any help would be appreciated!
Kind regards,
Mia

miag676, 2021-07-14:
I've managed to resolve the error by using an integer: `pClient = NimBLEDevice::createClient(NimBLEAddress(value.toInt()));`, however, it also truncates the address from 49:42:06:00:20:4e to 49. Do you know what could be the solution?

h2zero, 2021-07-14:
Hi @miag676, the NimBLEAddress constructor takes a std::string parameter. If you change the code in the first post to std::string(value) instead of String(value) it should work.

miag676, 2021-07-14:
Thanks for your response! I tried using it like this: `pClient = NimBLEDevice::createClient(NimBLEAddress(std::string(value)));`, but get the following error: no matching function for call to 'std::__cxx11::basic_string<char>::basic_string(String&)'. Did I write anything wrong in this line of code?

sivar2311, 2021-07-14:
Try this:
`NimBLEDevice::createClient(NimBLEAddress(std::string(value.c_str())));` 

A `std::string` needs a `char*` for initialization

miag676, 2021-07-14:
Thank you very much. This solved the issue!

closed 2024-06-04 by h2zero