h2zero/NimBLE-Arduino #267: Error when reading a client's service

miag676, 2021-07-19:
Hi, I have a problem with reading the characteristic of a connected device. Most of the time everything works perfectly, however, in a few occassions my client only connects to a BLE device and then doesn't read it's characterisitc. In my program I first read the addresses of the BLE devices I want to connect to (from a database using WiFi), then I connect to them and read the temperature measurement from the characteristic and push that measurement to a database. In the serial monitor it only writes: Connected to: 49:42:06:00:20:4e, and then starts connecting to another device. 

Is there a way to troubleshoot this by adding some code, in order to figure out is going wrong? 
Could it be that the WiFi connection is interfering with the BLE connection? Before establishing the connection I use WiFi.disconnect(), but is it possible that it's still causing some problems?

Thanks in advance!

Mia

h2zero, 2021-07-19:
Please try setting the debug log level to debug and post the output of the log here.

miag676, 2021-07-19:
Thanks for your reply. Here is the output of the log:
`D NimBLEClient: ">> disconnect()"
D NimBLEClient: "<< disconnect()"
KEY: -MexID00-MKZwjkyiYjM, KEY: serial, VALUE: 49:42:06:00:3b:29W NimBLEDevice: "Number of clients exceeds Max connections. Cur=17 Max=3"
New client created
D NimBLEClient: ">> connect(49:42:06:00:3b:29)"
D NimBLEClient: "Got Client event "
I NimBLEClient: "disconnect; reason=534, "
D NimBLEClientCallbacks: "onDisconnect: default"
D NimBLEClient: "Got Client event "
I NimBLEClient: "Connected event"
D NimBLEClient: "Got Client event "
E NimBLEClient: "Connection failed; status=574 "
Connected to: 49:42:06:00:3b:29
D NimBLEClient: ">> getService: uuid: 0xfff0"
D NimBLEClient: ">> retrieveServices"
E NimBLEClient: "Disconnected, could not retrieve services -aborting"
D NimBLEClient: "<< getService: not found"
`
I noticed it says service not found, but why does this happen only on every few loops? Most of the time it works fine.

h2zero, 2021-07-19:
Thanks for the log. The 2 error codes here are these:

0x0216 | 0x16 | BLE_ERR_CONN_TERM_LOCAL | Connection Terminated By Local Host
-- | -- | -- | --

0x023e | 0x3e | BLE_ERR_CONN_ESTABLISHMENT | Connection Failed to be Established
-- | -- | -- | --

What this means is that the controller was unable to complete the connection. This is normal in a busy environment in the 2.4ghz spectrum from packets loss or corruption. Not much that you can do other than try to connect again.

miag676, 2021-07-29:
Thank you for your help!

closed 2021-07-29 by miag676