h2zero/NimBLE-Arduino #269: Conflicting with Firebase RTDB

capedra, 2021-07-22:
Hello. Whenever I try to begin a stream on Firebase RTDB, it simply shows "stream error, not connected", and whenever I comment out the part of the code that scans and connects to the BLE device, Firebase works perfectly. It's important to note that I'm not even calling the function that is responsible for scanning and connecting to the BLE device: I'm just implementing it. I don't know how to solve this.

h2zero, 2021-08-03:
Sorry for the delay, I have not used Firebase before so I'm not sure what could be happening here. If you want to share some minimal code to reproduce the issue I'd be happy to test it.

h2zero, 2022-04-04:
Closing as no response was provided. Please reopen if the problem persists.

closed 2022-04-04 by h2zero