h2zero/NimBLE-Arduino #270: BLE RSSI of connected device

allenddld1, 2021-07-23:
Hi,
I'm trying to get the RSSI of client connected to the esp32 BLE server in the serial monitor using the NIMclient and NIMserver example. I can get the RSSI value when client connected server but I can't figure out a way to obtain the RSSI when the client keeps connecting with server. Is it possible? I attached the code to getrssi in the client below

  void onResult(BLEAdvertisedDevice* advertisedDevice) {
    Serial.print("BLE Advertised Device found: ");
    Serial.println(advertisedDevice->toString().c_str());
  if (advertisedDevice->getName() == "esp32"){
      BLEDevice::getScan()->stop();

      myDevice = advertisedDevice; 
      doConnect = true;
      doScan = true;
      Serial.print("BLE RSSI: ");
      
      int ble_rssi = advertisedDevice->getRSSI();
      Serial.println("insideadvertisedRSSI: " + String(ble_rssi));}}}

I have tried add  int rssi = pClient->getRssi();   Serial.println(rssi); in the loop. But it crash the program.

h2zero, 2021-07-24:
> I have tried add int rssi = pClient->getRssi(); Serial.println(rssi); in the loop. But it crash the program.

This should be working fine, can you provide a decoded backtrace for this crash?

allenddld, 2021-07-24:
Thank you for the response h2zero. I attach my code and error below:
```

#include <Arduino.h>

/*******original********
#include "BLEDevice.h"
***********************/
#include "NimBLEDevice.h"
#include "WiFi.h"
#include <esp_wifi.h>
#include "driver/adc.h"
// The remote service we wish to connect to.
static BLEUUID serviceUUID("4fafc201-1fb5-459e-8fcc-c5c9c331914b");
// The characteristic of the remote service we are interested in.
static BLEUUID    charTXUUID("beb5483e-36e1-4688-b7f5-ea07361b26a8");//
static BLEUUID    charRXUUID("6E400003-B5A3-F393-E0A9-E50E24DCCA9E");

static boolean doConnect = false;
static boolean connected = false;
static boolean doScan = false;
static BLERemoteCharacteristic* pTXCharacteristic;
static BLERemoteCharacteristic* pRXCharacteristic;
static BLEAdvertisedDevice* myDevice;
static BLEClient*  pClient;
static boolean rssiinrange = false; 
#define LED_BUILTIN 2
int ble_rssi = 0;
const int buzzer = 17;
const int freq = 5000;
const int ledChannel = 0;
const int resolution = 8;
void rssi_check(void);
void set_as_observer(void);
int scanTime = 5; //In seconds
int a =0; 

static void notifyCallback(
  BLERemoteCharacteristic* pBLERemoteCharacteristic,
  uint8_t* pData,
  size_t length,
  bool isNotify) {
    Serial.print("Notify callback for characteristic ");
    Serial.print(pBLERemoteCharacteristic->getUUID().toString().c_str());
    Serial.print(" of data length ");
    Serial.println(length);
    Serial.print("data: ");
    Serial.println((char*)pData);
}

/**  None of these are required as they will be handled by the library with defaults. **
 **                       Remove as you see fit for your needs                        */  
class MyClientCallback : public BLEClientCallbacks {
  void onConnect(BLEClient* pclient) {
  }

  void onDisconnect(BLEClient* pclient) {
    connected = false;
    Serial.println("onDisconnect");
  }
/***************** New - Security handled here ********************
****** Note: these are the same return values as defaults ********/
  uint32_t onPassKeyRequest(){
    Serial.println("Client PassKeyRequest");
    return 123456; 
  }
  bool onConfirmPIN(uint32_t pass_key){
    Serial.print("The passkey YES/NO number: ");Serial.println(pass_key);
    return true; 
  }
[... 204 lines cut ...]
loop
Forming a connection to 3c:61:05:2f:a8:c2
 - Created client
 - Connected to server
 - Found our service
 - Found our characteristic
 - Found our characteristic
We are now connected to the BLE Server.
Setting new characteristic value to "Time since boot: 7"
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400d32c7  PS      : 0x00060330  A0      : 0x800d1ba6  A1      : 0x3ffc6450
A2      : 0x00000000  A3      : 0x00000038  A4      : 0x3ffc9410  A5      : 0x3ffc355c  
A6      : 0x3f4002a6  A7      : 0x3ffc355c  A8      : 0x800e6348  A9      : 0x3ffc6440
A10     : 0x3ffc64b0  A11     : 0x400e5e80  A12     : 0x00000002  A13     : 0x00000038  
A14     : 0x3ffc9440  A15     : 0xff000000  SAR     : 0x00000018  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000008  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff  

ELF file SHA256: 0000000000000000

Backtrace: 0x400d32c7:0x3ffc6450 0x400d1ba3:0x3ffc6480 0x400e7211:0x3ffc64e0 0x4008fea2:0x3ffc6500

Rebooting...
ets Jun  8 2016 00:22:57
```

h2zero, 2021-07-24:
That is a nullptr exception. The reason is because you have not set the global pointer of `pClient` to anything.

In your `connectToServer` function please try changing:
```
BLEClient*  pClient  = BLEDevice::createClient();
```
to:
```
pClient  = BLEDevice::createClient();
```

This should correct the error.

allenddld, 2021-07-24:
Thank you @h2zero , it works great.

h2zero, 2021-07-24:
Great 😃, could you close this if there are no further related issues please?

allenddld, 2021-07-25:
close

commit 0d7592a7 (2024-12-20): Fix crash that could occur when server reconnects
* Update to not change services if they have not actually changed when starting the server
Closes #270
commit 85e500d8 (2024-12-22): Fix crash that could occur when server reconnects
* Update to not change services if they have not actually changed when starting the server
Closes #270
closed 2021-07-25 by allenddld1