h2zero/NimBLE-Arduino #271: [question] How to send string?

qlalfdu, 2021-07-28:
I Use the example NimBLE_Server.ino
In this example, resending the string sent by the smartphone, 
how can I send another string from esp32 to the smartphone terminal?
For example millis()

Thank you

h2zero, 2021-08-03:
Sorry, I forgot to reply to this. 

You would need to change the value of the characteristic to whatever value you want to send and trigger the notification.
An example of how to do this can be seen here https://github.com/h2zero/NimBLE-Arduino/blob/master/examples/Refactored_original_examples/BLE_notify/BLE_notify.ino

qlalfdu, 2021-08-03:
> 죄송합니다. 답변하는 것을 잊었습니다.
> 
> 특성 값을 전송하고 알림을 트리거하려는 값으로 변경해야 합니다.
> 이 작업을 수행하는 방법의 예는 [https://github.com/h2zero/NimBLE-Arduino/blob/master/examples/Refactored_original_examples/BLE_notify/BLE_notify.ino에서](https://github.com/h2zero/NimBLE-Arduino/blob/master/examples/Refactored_original_examples/BLE_notify/BLE_notify.ino) 볼 수 있습니다.[](https://github.com/h2zero/NimBLE-Arduino/blob/master/examples/Refactored_original_examples/BLE_notify/BLE_notify.ino)

qlalfdu, 2021-08-03:
Thank you very much.

closed 2021-08-03 by qlalfdu