h2zero/NimBLE-Arduino #272: Failed to compile with esp32-c3 development board

Danjuanlab, 2021-07-31:
I used the ble keyboard sample code based on the NimBLE library to write code to the esp32 c3 development board, but the compilation error occurred. From the code point of view, the C3 config file lacks some variable definitions and type methods. I don't know if this is a problem with the NimBLE library or the lack of information in the ESP32 c3 library.

ERROR-LOG:

C:\Users\DJlab_Effect\Documents\Arduino\libraries\NimBLE-Arduino\src\NimBLEDevice.cpp:65:65: error: 'CONFIG_BTDM_SCAN_DUPL_CACHE_SIZE' was not declared in this scope
 uint16_t                    NimBLEDevice::m_scanDuplicateSize = CONFIG_BTDM_SCAN_DUPL_CACHE_SIZE;
                                                                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
C:\Users\DJlab_Effect\Documents\Arduino\libraries\NimBLE-Arduino\src\NimBLEDevice.cpp:65:65: note: suggested alternative: 'CONFIG_BT_CTRL_SCAN_DUPL_CACHE_SIZE'
 uint16_t                    NimBLEDevice::m_scanDuplicateSize = CONFIG_BTDM_SCAN_DUPL_CACHE_SIZE;
                                                                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
                                                                 CONFIG_BT_CTRL_SCAN_DUPL_CACHE_SIZE
C:\Users\DJlab_Effect\Documents\Arduino\libraries\NimBLE-Arduino\src\NimBLEDevice.cpp:66:62: error: 'CONFIG_BTDM_SCAN_DUPL_TYPE' was not declared in this scope
 uint8_t                     NimBLEDevice::m_scanFilterMode = CONFIG_BTDM_SCAN_DUPL_TYPE;
                                                              ^~~~~~~~~~~~~~~~~~~~~~~~~~
C:\Users\DJlab_Effect\Documents\Arduino\libraries\NimBLE-Arduino\src\NimBLEDevice.cpp:66:62: note: suggested alternative: 'CONFIG_BT_CTRL_SCAN_DUPL_TYPE'
 uint8_t                     NimBLEDevice::m_scanFilterMode = CONFIG_BTDM_SCAN_DUPL_TYPE;
                                                              ^~~~~~~~~~~~~~~~~~~~~~~~~~
                                                              CONFIG_BT_CTRL_SCAN_DUPL_TYPE
C:\Users\DJlab_Effect\Documents\Arduino\libraries\NimBLE-Arduino\src\NimBLEDevice.cpp: In static member function 'static void NimBLEDevice::init(const string&)':
C:\Users\DJlab_Effect\Documents\Arduino\libraries\NimBLE-Arduino\src\NimBLEDevice.cpp:554:16: error: 'struct esp_bt_controller_config_t' has no member named 'mode'
         bt_cfg.mode = ESP_BT_MODE_BLE;
                ^~~~
C:\Users\DJlab_Effect\Documents\Arduino\libraries\NimBLE-Arduino\src\NimBLEDevice.cpp:555:16: error: 'struct esp_bt_controller_config_t' has no member named 'ble_max_conn'; did you mean 'ble_max_act'?
         bt_cfg.ble_max_conn = CONFIG_BT_NIMBLE_MAX_CONNECTIONS;
                ^~~~~~~~~~~~
                ble_max_act

h2zero, 2021-07-31:
If you are not using the current master branch code please try updating it and let me know. There was a recent commit to support the C3.

Danjuanlab, 2021-07-31:
@h2zero  Using the new C3 branch, the code can be compiled correctly, thanks.

closed 2021-07-31 by Danjuanlab