h2zero/NimBLE-Arduino #273: Multiple definitions of `btInUse`

ketan, 2021-08-03:
I just upgraded to 1.3.0 and see this error:

```
Linking .pio/build/lolin32_mpu9250_ms5611/firmware.elf
.pio/build/lolin32_mpu9250_ms5611/lib532/libNimBLE-Arduino.a(NimBLEDevice.cpp.o): In function `btInUse':
/Users/ketanpadegaonkar/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/ext/new_allocator.h:110: multiple definition of `btInUse'
.pio/build/lolin32_mpu9250_ms5611/libFrameworkArduino.a(esp32-hal-bt.c.o):/Users/ketanpadegaonkar/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-bt.c:19: first defined here
collect2: error: ld returned 1 exit status
*** [.pio/build/lolin32_mpu9250_ms5611/firmware.elf] Error 1
```

I can see that `btInUse` is defined [here](https://github.com/espressif/arduino-esp32/blob/5bb8177aa1feec665938be97607b526a4f0ec0bb/cores/esp32/esp32-hal-bt.c#L19) and [here](https://github.com/h2zero/NimBLE-Arduino/blob/1.3.0/src/NimBLEDevice.cpp#L34-L35), is this expected, is there any workaround?

h2zero, 2021-08-03:
I'm not having any issues compiling with it. Are you calling any functions from the esp32-hal-bt file? That will likely cause it.

ketan, 2021-08-03:
I have a program that includes `BluetoothSerial.h`, this is to allow the program to use BT classic or BLE (depending on user configuration).

It is possible that `esp32-hal-bt.c` is being included by `BluetoothSerial.h` indirectly, but I'm certainly not including it.

h2zero, 2021-08-03:
Yes, it looks like bluetoothserial calls those functions [here](https://github.com/espressif/arduino-esp32/blob/5bb8177aa1feec665938be97607b526a4f0ec0bb/libraries/BluetoothSerial/src/BluetoothSerial.cpp#L581)

How does this work in your application? Is this a compile option or a startup option?

ketan, 2021-08-03:
Basically my code looks something like this:

```
void setup() {
  if (user setting in eeprom == BLE) {
    NimBLEDevice::init(...);
    ...
  } else {
    BluetoothSerial bluetoothSerial;
    bluetoothSerial.begin();
    ...
  }
}
```

There is obviously a way for users to change the user setting in eeprom via a web interface — so the choice of BLE or BT classic cannot be done statically at compile time.

h2zero, 2021-08-03:
Ok, that explains the error then. Simply having that function in your code will cause the linker error. I will need to revert the change to avoid this.

h2zero, 2021-08-03:
Just pushed the change, please let me know if it works for you. I will issue a new release once confirmed.

h2zero, 2022-01-15:
Resolved

closed 2022-01-15 by h2zero